espressif/esp-idf #14799: RMT byte encoder less than 1 byte

gamadan, 2024-10-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I only need the byte encoder to dynamically encode four bits.  Not a full byte.  
```
uint8_t data = 0b1111 | *((uint8_t*)primary_data);
encoded_symbols += bytes_encoder->encode(bytes_encoder, channel, &data, sizeof(data), &session_state);
```

The code above encodes 8 bits, 4 (leading) zeros, and 4 bits from primary_data.

My protocol has 20 address bits, 4 data bits, and 2 stop bits.  Do I really need to dynamically encode the last 4 address bits with the 4 data bits each time?

suda-morris, 2025-01-06:
> Do I really need to dynamically encode the last 4 address bits with the 4 data bits each time?

I think this is the best approach for now.

gamadan, 2025-01-17:
Doesn't seem there will be a lot of interest.  Plus I've seen a issue closed out due to overwhelming issue backlog.  So no reason to keep this one open.  I was able to solve the issue by dynamically encoding the upper nibble along with the 4 bits of dynamic data. 
 So, as long as the payload is 8 or more bits, there is a work around.

Copied straight from the RMT API guide, "due to the flexibility of its data format, RMT can be extended to a versatile and general-purpose transceiver, transmitting or receiving many other types of signals"  Just seems unfortunate to make something flexible, extendable, and versatile, but then only granular down to the byte, not bit.

closed 2026-09-16 by espressif-bot