espressif/esp-idf #14800: Cannot open UART1,open: No such file or directory

lujiex, 2024-10-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

Open the UART1 normally

### What is the actual behavior?

E (00:04:36.496) uartManager: Cannot open UART1
open: No such file or directory

### Steps to reproduce.

[image]
Set "Channel for console output" to "USB Serial/JTAG Controller" and UART1 will fail to open with the error "Cannot open UART1
open: No such file or directory"

[image]
Set Channel for console output to Default: UART0 ", UART1 can be successfully opened, but my serial port console program will be abnormal, "Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.”

[image]

The source code in question is as follows：
[image]

### Debug Logs.

```plain
UART1 is open normally, but the console LOG is abnormal：
--- esp-idf-monitor 1.4.0 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (133) esp_image: segment 0: paddr=00010020 vaddr=3c130020 size=92d98h (601496) map
I (250) esp_image: segment 1: paddr=000a2dc0 vaddr=3fc98600 size=04e10h ( 19984) load
I (255) esp_image: segment 2: paddr=000a7bd8 vaddr=40374000 size=08440h ( 33856) load
I (264) esp_image: segment 3: paddr=000b0020 vaddr=42000020 size=1219dch (1186268) map
I (477) esp_image: segment 4: paddr=001d1a04 vaddr=4037c440 size=0c0f4h ( 49396) load
I (499) boot: Loaded app from partition at offset 0x10000
I (499) boot: Disabling RNG early entropy source...
[... 315 lines cut ...]

I (962) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (962) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (1052) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (00:00:45.570) wifiManager: got ip:192.168.8.139
I (00:00:45.571) esp_netif_handlers: sta ip: 192.168.8.139, mask: 255.255.255.0, gw: 192.168.8.1
I (00:00:45.571) wifiManager: apsta mode,set NAPT
W (00:00:45.574) esp_netif_lwip: napt disabled on esp_netif:0x3fcd4910
I (00:00:45.586) uartManager: Uart start...
E (00:00:45.588) uartManager: Cannot open UART1
open: No such file or directory
I (00:00:45.589) http_server: HTTP start...
I (00:00:45.592) http_server: Registering URI handlers
I (00:00:45.597) esp_https_server: Starting server
I (00:00:45.599) esp_https_server: Server listening on port 443
I (00:00:45.601) http_server: Registering URI handlers
yfbao> 
yfbao> version
IDF Version:v5.3-dirty
Chip info:
        model:ESP32-S3
        cores:2
        feature:/802.11bgn/BLE/External-Flash:16 MB
        revision number:2
```

igrr, 2024-10-29:
You probably need to register UART VFS first. If UART is used as console output this is done automatically, but otherwise the driver isn't registered to save resources.

https://github.com/espressif/esp-idf/blob/a0f798cfc4bbd624aab52b2c194d219e242d80c1/components/esp_driver_uart/include/driver/uart_vfs.h#L20

lujiex, 2024-10-29:
> You probably need to register UART VFS first. If UART is used as console output this is done automatically, but otherwise the VFS driver isn't registered to save resources.
> 
> https://github.com/espressif/esp-idf/blob/a0f798cfc4bbd624aab52b2c194d219e242d80c1/components/esp_driver_uart/include/driver/uart_vfs.h#L20

Thanks, you solved my problem. I called "uart_vfs_dev_register();" before initializing the serial port. After that, it can open normally.
The following code is running normally：
[image]

haberturdeur, 2024-11-14:
Hi @lujiex, if the issue has been resolved, could you please close it as done?
Thank you.

closed 2024-11-14 by lujiex