espressif/esp-idf #14801: [WIFI][ESP32S3][v5.3.1] WIFI provisioning failing to scan and connect to mobile hotspot multiple times

KonssnoK, 2024-10-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

on top of a0f798cfc4bbd624aab52b2c194d219e242d80c1
@zhangyanjiaoesp 
we are in the middle of moving to v5.3 from v4.4.
We are seeing issues in the wifi that did not occur in v4.4

Specifically,
we are moving the provisioning part of our devices.
Apart from having to change BLE stack ( https://github.com/espressif/esp-idf/issues/14790#issuecomment-2443402129 )

the device is now not apparently able to connect to phone hotspots anymore. Phone is a pixel 8.

I think this might be related to nimble taking the antenna for too long.
Could you point me to the right options to fix this?

```
I (194914) wifi:new:<7,0>, old:<13,0>, ap:<255,255>, sta:<7,0>, prof:1, snd_ch_cfg:0x0
I (194916) wifi:state: init -> auth (0xb0)
I (195693) mesh: mesh is not inited
I (196693) mesh: mesh is not inited
I (197693) mesh: mesh is not inited
I (198693) mesh: mesh is not inited
I (199207) wifi:state: auth -> init (0x200)
I (199220) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

I (199221) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1, snd_ch_cfg:0x0
E (01:09:12.802) wifi_prov_mgr: STA Disconnected
E (01:09:12.805) wifi_prov_mgr: Disconnect reason : 2
I (01:09:12.811) prov_BLE: Disconnected. Connecting to the AP again...
E (199243) wifi:sta is connecting, return error
I (199693) mesh: mesh is not inited
I (200693) mesh: mesh is not inited
E (01:09:14.278) now_packet: Receive ESPNOW data too short, len:3
E (01:09:14.298) now_packet: Invalid CRC 274C 445
E (01:09:14.410) now_packet: Invalid CRC 75D8 445
E (01:09:14.465) now_packet: Invalid CRC DA09 4BF
I (201693) mesh: mesh is not inited
I (202693) mesh: mesh is not inited
I (203693) mesh: mesh is not inited
I (204055) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

E (01:09:17.633) wifi_prov_mgr: STA Disconnected
E (01:09:17.634) wifi_prov_mgr: Disconnect reason : 205
I (01:09:17.638) prov_BLE: Disconnected. Connecting to the AP again...
E (204070) wifi:sta is connecting, return error
I (204693) mesh: mesh is not inited
E (01:09:19.175) now_packet: Invalid CRC 7F5D 4C0
E (01:09:19.218) now_packet: Invalid CRC 4E37 445
E (01:09:19.222) now_packet: Invalid CRC 5FB3 4C0
I (205693) mesh: mesh is not inited
I (206693) mesh: mesh is not inited
I (207693) mesh: mesh is not inited
I (208693) mesh: mesh is not inited
I (208882) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1, snd_ch_cfg:0x0
I (208883) wifi:state: init -> auth (0xb0)
I (209595) wifi:state: auth -> assoc (0x0)
I (209693) mesh: mesh is not inited
I (01:09:23.600) stats: iteration_time_us=999976 Name=last% IDLE1=60 IDLE0=98 wifi=38 
I (210597) wifi:state: assoc -> init (0x2700)
I (210610) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

I (210611) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1, snd_ch_cfg:0x0
E (01:09:24.192) wifi_prov_mgr: STA Disconnected
E (01:09:24.195) wifi_prov_mgr: Disconnect reason : 39
I (01:09:24.201) prov_BLE: Disconnected. Connecting to the AP again...
E (210633) wifi:sta is connecting, return error
I (210693) mesh: mesh is not inited
I (211693) mesh: mesh is not inited
E (01:09:25.695) now_packet: Invalid CRC B1CA 445
E (01:09:25.729) now_packet: Invalid CRC E3D1 4C0
I (212693) mesh: mesh is not inited
I (213693) mesh: mesh is not inited
I (214693) mesh: mesh is not inited
I (215445) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

E (01:09:29.024) wifi_prov_mgr: STA Disconnected
E (01:09:29.024) wifi_prov_mgr: Disconnect reason : 201
E (01:09:29.028) wifi_prov_mgr: STA AP Not found
I (01:09:29.034) prov_BLE: Disconnected. Connecting to the AP again...
E (01:09:29.043) prov_BLE: Provisioning failed! Reason : Wi-Fi access-point not found
I (215693) mesh: mesh is not inited
I (216693) mesh: mesh is not inited
E (01:09:30.517) now_packet: Invalid CRC E0BD 4E0
E (01:09:30.645) now_packet: Invalid CRC 7937 4C0
W (01:09:30.648) network: LOST IP
D (01:09:30.648) modem_comm: IP event 1
I (217693) mesh: mesh is not inited
I (218693) mesh: mesh is not inited
I (219065) mesh: mesh is not started
W (01:09:32.641) mesh_main: <MESH_MSG_SEND_INTERNAL>send err:16388
I (01:09:32.642) tmpr_propagation: Propagated local temperature
I (219693) mesh: mesh is not inited
I (220285) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1, snd_ch_cfg:0x0
I (220285) wifi:state: init -> auth (0xb0)
I (220859) mesh: mesh is not inited
I (220905) wifi:state: auth -> assoc (0x0)
I (220911) wifi:state: assoc -> run (0x10)
I (01:09:34.600) stats: iteration_time_us=1000036 Name=last% IDLE1=50 IDLE0=98 wifi=49 
I (221049) wifi:state: run -> init (0x7a0)
I (221062) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

I (221063) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1, snd_ch_cfg:0x0
E (01:09:34.644) wifi_prov_mgr: STA Disconnected
E (01:09:34.647) wifi_prov_mgr: Disconnect reason : 7
I (01:09:34.653) prov_BLE: Disconnected. Connecting to the AP again...
E (221084) wifi:sta is connecting, return error
I (221860) mesh: mesh is not inited
E (01:09:36.174) now_packet: Invalid CRC C084 4C1
E (01:09:36.187) now_packet: Invalid CRC 4409 4E0
E (01:09:36.292) now_packet: Invalid CRC 4543 4C1
E (01:09:36.302) now_packet: Invalid CRC C845 4E0
I (222860) mesh: mesh is not inited
I (223860) mesh: mesh is not inited
I (01:09:37.643) offline: object saved to PSRAM, free 5518 type 10 idx 5
I (224860) mesh: mesh is not inited
I (225860) mesh: mesh is not inited
I (225896) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

E (01:09:39.475) wifi_prov_mgr: STA Disconnected
E (01:09:39.475) wifi_prov_mgr: Disconnect reason : 201
E (01:09:39.479) wifi_prov_mgr: STA AP Not found
I (01:09:39.485) prov_BLE: Disconnected. Connecting to the AP again...
E (01:09:39.494) prov_BLE: Provisioning failed! Reason : Wi-Fi access-point not found
I (226860) mesh: mesh is not inited
E (01:09:41.061) now_packet: Invalid CRC 5E8C 4E0
E (01:09:41.089) now_packet: Invalid CRC 1543 4C1
I (227860) mesh: mesh is not inited
I (228860) mesh: mesh is not inited
I (229860) mesh: mesh is not inited
I (230736) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

E (01:09:44.314) wifi_prov_mgr: STA Disconnected
E (01:09:44.315) wifi_prov_mgr: Disconnect reason : 201
E (01:09:44.319) wifi_prov_mgr: STA AP Not found
I (01:09:44.325) prov_BLE: Disconnected. Connecting to the AP again...
I (01:09:44.340) prov_BLE: Failed to connect with provisioned AP, reseting provisioned credentials
I (230862) mesh: mesh is not inited
I (231862) mesh: mesh is not inited
```

KonssnoK, 2024-10-29:
here logs with verbose wifi.
The important ones are the "failed"

[24102901_wifi_succeeded_scan.txt](https://github.com/user-attachments/files/17555408/24102901_wifi_succeeded_scan.txt)
[24102901_wifi_failed_scan.txt](https://github.com/user-attachments/files/17555409/24102901_wifi_failed_scan.txt)
[24102901_wifi_success.txt](https://github.com/user-attachments/files/17555410/24102901_wifi_success.txt)
[24102901_wifi_fail.txt](https://github.com/user-attachments/files/17555411/24102901_wifi_fail.txt)

zhangyanjiaoesp, 2024-10-30:
@KonssnoK 
[image]

[image]

**These disconnections mean the internal timeout.** 

[image]

**This disconnection means the STA receives a disassociation frame from the AP, reason code is 7.**

zhangyanjiaoesp, 2024-10-30:
@KonssnoK
Have you enabled the LR mode? Because I saw this information in the log: `scan specific ssid=KI via LR`

KonssnoK, 2024-10-30:
@zhangyanjiaoesp we do have enabled LR mode while the device is provisioning because we use esp-now for propagating credentials to other devices in the same installation.
Once the credentials are propagated and the installation is finished, we remove the LR flag.

KonssnoK, 2024-10-30:
> @KonssnoK [...]

what i don't understand is why the final error on the connection try is 205 AP not found. we were able to scan for it, why would it fail saying it's not able to see it anymore?

zhangyanjiaoesp, 2024-10-30:
@KonssnoK If the connection fails, the STA will add the AP to a blacklist. During the next scan, if the STA finds that the AP it wants to connect is in the blacklist, it will return a 205 error and then remove the AP from the blacklist. When the next time it attempts to reconnect，the STA enter the scanning and connecting state again. You can consider the 205 as an intermediate state.

zhangyanjiaoesp, 2024-10-30:
@KonssnoK Can you provide a demo for us to test? Since Wi-Fi provisioning involves multiple modules, if we can reproduce it locally, it will speed up our debugging process.

KonssnoK, 2024-10-30:
@zhangyanjiaoesp let me see what i can do

KonssnoK, 2024-10-30:
so i made a change to the provisioning example on my branch
https://github.com/KonssnoK/esp-idf/tree/lc/provisioning

C:\src\esp-idf\examples\provisioning\wifi_prov_mgr

but of course i do not see the logs that i-ve sent from our device that was migrated from 4.4.

Could it be something related to the wifi calibration done in 4.4? Ot that gets reset automatically when we update to 5.3?

I will try to check for more differences

KonssnoK, 2024-10-31:
@zhangyanjiaoesp i am really disturbed by the fact that now i cannot reproduce the behavior i logged above with both devkits and our dedicated hardware.
Is there any internal setting that needs to be updated when moving from 4.4 to 5.3?
Is there something that could justify having such a behavior for some period after an update?

zhangyanjiaoesp, 2024-10-31:
@KonssnoK Do you mean that you can't reproduce the scan fail and connect fail issue?

KonssnoK, 2024-10-31:
no i just reproduced it again on our hardware. I do not understand how to trigger it on a devkit tho.
[24103101_wifi_fail.txt](https://github.com/user-attachments/files/17585235/24103101_wifi_fail.txt)

zhangyanjiaoesp, 2024-10-31:
> no i just reproduced it again on our hardware. I do not understand how to trigger it on a devkit tho. [24103101_wifi_fail.txt](https://github.com/user-attachments/files/17585235/24103101_wifi_fail.txt)

1. This log shows the AP always send disassociation to the ESP device, can you capture a packet for this? 
2. How about you change another hotspot?

KonssnoK, 2024-10-31:
> 1. This log shows the AP always send disassociation to the ESP device, can you capture a packet for this?
>
>     2. How about you change another hotspot?

@zhangyanjiaoesp my problem is that our device is supposed to be compatible with multiple phones, used during installation. 
If I find a phone where it doesn't work, sadly I need to understand why..
We did try with other routers without seeing such a strange behavior. The main issue is that my phone (Pixel8) did not experience the issue before using v4.4.

zhangyanjiaoesp, 2024-11-01:
@KonssnoK
 The fixes in v4.4 have been integrated into v5.3

KonssnoK, 2024-11-04:
so by looking at logs it seems that the issue might be related to moving from one channel to another before connecting.

[image]

I'm still working on tryin to replicate on a devkit

zhangyanjiaoesp, 2024-11-04:
Are you saying that every time the current channel is different from the AP channel, there will be a connection failure?

KonssnoK, 2024-11-04:
@zhangyanjiaoesp no sorry i was wrong in the end,
I updated the channel also in the devkit example, and this is a comparison between a failure (left) and success (right)

I see multiple 

`D (594710) wifi:Drop duplicate SAE auth: seq=1`

and then disconnection with reason 7.

[image]

zhangyanjiaoesp, 2024-11-04:
`D (594710) wifi:Drop duplicate SAE auth: seq=1`  This log indicates that STA has received the auth packet and will discard the retransmitted auth packet.
Can you capture packets for the device and the hotspot? Both success and failure are necessary. It is best to display the time in hours/minutes/seconds in the log to align with the packet capture time

KonssnoK, 2024-11-05:
I'm trying to capture packets but most of the time it cannot even scan the access point... 
I should have something soon

KonssnoK, 2024-11-05:
@zhangyanjiaoesp 

here is an example of a very strange behavior from the device:
- scan ok 
- provisioning
- rescan for connecting
- failure: AP not found

Note this works in v4.4

- device logs with debug level on wifi
[241105_wifi_scanok_connect_not_found.txt](https://github.com/user-attachments/files/17631016/241105_wifi_scanok_connect_not_found.txt)

File .pcapng
[241105_wifi_scanok_fail_notfound.txt](https://github.com/user-attachments/files/17631043/241105_wifi_scanok_fail_notfound.txt)

KonssnoK, 2024-11-05:
@zhangyanjiaoesp in this other case

- scan ok
- connection fails
- AP not found

logs
[24110503_wifi_scanok_connect_not_found.txt](https://github.com/user-attachments/files/17632978/24110503_wifi_scanok_connect_not_found.txt)

File .pcapng
[24110503_wifi_scanok_not_found.txt](https://github.com/user-attachments/files/17632983/24110503_wifi_scanok_not_found.txt)

KonssnoK, 2024-11-05:
more random errors generated by the device

log(partial)
[24110504_wifi_scanok_errors.txt](https://github.com/user-attachments/files/17633253/24110504_wifi_scanok_errors.txt)

pcapng
[24110504_wifi_scanok_errors.txt](https://github.com/user-attachments/files/17633258/24110504_wifi_scanok_errors.txt)

zhangyanjiaoesp, 2024-11-06:
> @zhangyanjiaoesp
> 
> here is an example of a very strange behavior from the device:
> 
>     * scan ok
> 
>     * provisioning
> 
>     * rescan for connecting
> 
>     * failure: AP not found
> 
> 
> Note this works in v4.4
> 
> device logs with debug level on wifi [241105_wifi_scanok_connect_not_found.txt](https://github.com/user-attachments/files/17631016/241105_wifi_scanok_connect_not_found.txt)
> 
> File .pcapng [241105_wifi_scanok_fail_notfound.txt](https://github.com/user-attachments/files/17631043/241105_wifi_scanok_fail_notfound.txt)
> 
> MAC AP: d2:0f:7d;8a:8b:29 MAC DEV: AC:17:54:00:0A:B0

In this log, the scan is fail, it didn't find the specific AP, if founds, the log will show `D (398716) wifi:ap found, mac=d2:0f:7d:8a:8b:29` . The log `wifi:connect status 0 -> 1`  is only used to represent some internal status and is not a sign of starting a connection. The sign of starting a connection is: `wifi:state: init -> auth (0xb0)`

zhangyanjiaoesp, 2024-11-06:
For the other three logs and packet captures, I will check and reply to you ASAP.

KonssnoK, 2024-11-06:
> In this log, the scan is fail, it didn't find the specific AP, if founds, the log will show `D (398716) wifi:ap found, mac=d2:0f:7d:8a:8b:29` . The log `wifi:connect status 0 -> 1` is only used to [...]

I understand it's not finding the AP, but it found it 5 seconds before during the scan, how can the device stop seeing an AP less than 1 meter away in 2 consecutive scans?

zhangyanjiaoesp, 2024-11-06:
@KonssnoK 

1. According to the packets, the `wifi:state: run -> init (0x7a0)` issue seems related to the NULL data sent by STA. The following wifi lib contains a bug fix, please test again.
[wifi_lib_1106_s3.zip](https://github.com/user-attachments/files/17644375/wifi_lib_1106_s3.zip)
wifi firmware version: 3eceb8154

[image]

2. The `wifi:state: assoc -> init (0x3510)`  indicates that STA has received an associate response with a status code of invalid PMKID. 
If the STA successfully connects to the AP, it will cache the PMK for this time. When reconnecting next time, if it still connects to the exact same AP, it will directly use the cached PMK, which can reduce the computation time of the STA's PMK and speed up the connection. But if the AP finds that the PMK has expired, it will return such an error.

[image]

KonssnoK, 2024-11-06:
@zhangyanjiaoesp do you know when the nulls are sent? because i don't always manage to reproduce the issue (i want to reproduce it before testing your library)

KonssnoK, 2024-11-06:
@zhangyanjiaoesp with the new library for now i'm not able to replicate the issue, but again, it doesn't always happens.. i'll try some more

zhangyanjiaoesp, 2024-11-07:
> @zhangyanjiaoesp do you know when the nulls are sent? because i don't always manage to reproduce the issue (i want to reproduce it before testing your library)

When the STA receives the association response, the hardware responds with an ACK to the AP and simultaneously begins sending NULL data. This is a feature added to address the issue where some APs may not be able to send EAPOL packets in a timely manner. However, if the hotspot does not receive the STA's ACK in time, it will continuously retransmit the association response. When the hotspot receives the NULL data, it will send a disassociation (disassoc) message to the STA, as described in the scenario above. If the hotspot is able to quickly receive the STA's ACK response, the NULL data sent by the STA will not affect the Wi-Fi connection, as shown in the diagram below:
[image]

KonssnoK, 2024-11-07:
@zhangyanjiaoesp ok so the way to reproduce is most probably a busy channel.

The fix that you sent me is already final? Or it just temporarely removed the NULL data?
Do you need more logs?

I'm not sure this covers also the AP not found error seen some of the times.

KonssnoK, 2024-11-07:
> Are you saying that after using the new wifi lib, the AP not found issue has been resolved?

I have no proof for that, I will try some more in the next days when i can find a busy network in the office

zhangyanjiaoesp, 2024-11-07:
I think the new wifi lib will not affect the scan results, as it only modifies the connection process code. Scanning occurs before the connection process.

zhangyanjiaoesp, 2024-11-28:
@KonssnoK 
Hello, apologies for the inconvenience. After our internal discussion, we believe the original behavior is reasonable. When the device receives the association response, it replies with a hardware ACK and sends a NULL data frame. The issue lies in the fact that the mobile hotspot did not receive the hardware ACK and kept retransmitting the association response, even though it was able to receive the NULL data frame from the device and send the disassociation frame. Clearly, there seems to be a performance issue with the hotspot. If I delay sending the NULL data, it might make the device's behavior appear unusual, and if the device does not send the NULL data, the original issue cannot be resolved. 

> When the STA receives the association response, the hardware responds with an ACK to the AP and simultaneously begins sending NULL data. This is a feature added to address the issue where some APs may not be able to send EAPOL packets in a timely manner.

Would it be possible for you to try a different mobile hotspot?

KonssnoK, 2024-12-17:
hi @zhangyanjiaoesp ,
i did the following:
- Update to 5.3.2, latest commit 083aad99cfc1a7981009ac7f18e29824c47ffba2
- Retried with the pixel8: Still failing, attaching a log
[24121701_wifi_fail.txt](https://github.com/user-attachments/files/18166484/24121701_wifi_fail.txt)
- Retried with a samsun S21+: seemed to work ok

zhangyanjiaoesp, 2024-12-18:
@KonssnoK 
Hi, have you seen my [comment above](https://github.com/espressif/esp-idf/issues/14801#issuecomment-2505617723)? The previous fix was not merged into the IDF because we determined that the issue was related to the mobile hotspot. We recommend that you try using a different phone.

I see that you have switched to a new phone, and it is working properly.

KonssnoK, 2024-12-18:
Hi @zhangyanjiaoesp,
yes i read your previous comment.
Once again, our issue is that we cannot force installers to use specific phones when using our devices.
If this issue happens with the pixel phone it could happen with other phones.

Broad compatibility is what we are looking for.

zhangyanjiaoesp, 2024-12-18:
We understand that you want the product to have better compatibility, but it is not feasible for us to ensure compatibility with all devices. This is because some hotspots or routers exhibit behavior that is either unreasonable or even non-compliant with protocol specifications. What we can do is improve our handling when our response to reasonable behavior on the other end is inadequate. However, if the issue clearly lies with the other device, making adjustments to accommodate such devices seems unreasonable. Moreover, attempting to ensure compatibility with such problematic devices could potentially introduce new issues, leading to incompatibility with other devices.

zhangyanjiaoesp, 2024-12-19:
@KonssnoK 
So, do you agree to switch to a different phone? If yes, can we close this issue?

AxelLin, 2024-12-19:
> @KonssnoK Hi, have you seen my [comment above](https://github.com/espressif/esp-idf/issues/14801#issuecomment-2505617723)? The previous fix was not merged into the IDF because we determined that the [...]

@zhangyanjiaoesp Why v4.4 does not has such issue?

KonssnoK, 2024-12-20:
> @KonssnoK So, do you agree to switch to a different phone? If yes, can we close this issue?

well i'll continue to use both phones.

Could you give me a small summary of the problem so that i can open an issue to google?

Thanks

AxelLin, 2024-12-20:
@KonssnoK 

Just curious if you can connect to pixel 8 using examples/wifi/getting_started/station example on esp-idf-5.3.1?
(This test excluding BT/wifi:Coexist issue.)

zhangyanjiaoesp, 2024-12-23:
@KonssnoK 
The summary is here:
[image]

This log indicates that the STA disconnected from the AP due to receiving a disassociation packet sent by the AP. The disassociation packet contains a reason code of 7 (Class 3 frame received from a non-associated STA). The Wireshark packet capture is as follows:
[image]

When the STA receives the association response, it replies with a hardware ACK and sends a NULL data frame. The issue lies in the fact that the mobile hotspot did not receive the hardware ACK and kept retransmitting the association response, however it was able to receive the NULL data frame from the STA and send the disassociation frame. 

The normal packet capture is as follows: 
[image]

The hotspot successfully receives the ACK, and the two devices communicate normally.

(By the way, The screenshots above are all from your previous packet captures, and the captures do not include MAC ACK packets. If you can perform a new capture and include the MAC ACK packets, it would likely be more convincing. The MAC ACK packet is similar to the packet captures with a blue background shown below.)
[image]

KonssnoK, 2025-05-08:
@zhangyanjiaoesp @Alvin1Zhang 
for reference after updating to latest version of 5.3.3 the issue stopped happening, 
probably related to 
6e443922a5579ca22e7520ae48d2e084df05ed05 ?

zhangyanjiaoesp, 2025-05-08:
@KonssnoK 
I can't find the `6e443922a5579ca22e7520ae48d2e084df05ed05` commit on v5.3.3, can you give me a link? 
However, we did fix the issue of sending NULL data immediately after association. The fix can be found here: bccec2b6c1bdfd34d3a9024879e73ef974c917c0. It should be on the release/v5.3 branch, but it is not included in v5.3.3.

KonssnoK, 2025-05-08:
sorry i wrote 5.3.3 but i meant the 5.3 dev branch