espressif/esp-idf #14803: StoreProhibited exception when using i2c_slave_receive

danielcolchete, 2024-10-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevkitC-V4

### Power Supply used.

External 5V

### What is the expected behavior?

Calling i2c_slave_receive() and then xQueueReceive() with a properly initialized i2c_slave and a properly allocated unit8_t *data should not cause an exception.

### What is the actual behavior?

I'm getting an StoreProhibited exception during runtime:

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c2e4  PS      : 0x00060b33  A0      : 0x800844cc  A1      : 0x3ffb0f30
0x4000c2e4: memcpy in ROM

A2      : 0x90909090  A3      : 0x3ffaff50  A4      : 0x48484848  A5      : 0x90909090
A6      : 0x6c6c6548  A7      : 0x694c206f  A8      : 0x00000000  A9      : 0x3ffb0e90
A10     : 0x00000000  A11     : 0x3ffb0eb1  A12     : 0x3ffb0e5f  A13     : 0x00000030
A14     : 0x00000000  A15     : 0x3ffb0e69  SAR     : 0x00000005  EXCCAUSE: 0x0000001d
EXCVADDR: 0x90909090  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x04848483
0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x4000c2e1:0x3ffb0f30 0x400844c9:0x3ffb0f40 0x4008460d:0x3ffb0f80 0x40084661:0x3ffb0fa0 0x4008243d:0x3ffb0fd0 0x40082ef5:0x3ffb0ff0 0x400847e3:0x3ffb47a0 0x400d2373:0x3ffb47c0 0x40087efe:0x3ffb47e0 0x4008710d:0x3ffb4800
0x4000c2e1: memcpy in ROM
0x400844c9: s_i2c_handle_complete at /home/vham/esp/esp-idf/components/esp_driver_i2c/i2c_slave.c:80
0x4008460d: s_slave_fifo_isr_handler at /home/vham/esp/esp-idf/components/esp_driver_i2c/i2c_slave.c:132
0x40084661: s_slave_isr_handle_default at /home/vham/esp/esp-idf/components/esp_driver_i2c/i2c_slave.c:178
0x4008243d: shared_intr_isr at /home/vham/esp/esp-idf/components/esp_hw_support/intr_alloc.c:445
0x40082ef5: _xt_lowint1 at /home/vham/esp/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x400847e3: xt_utils_wait_for_intr at /home/vham/esp/esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at /home/vham/esp/esp-idf/components/esp_hw_support/cpu.c:55
0x400d2373: esp_vApplicationIdleHook at /home/vham/esp/esp-idf/components/esp_system/freertos_hooks.c:58
0x40087efe: prvIdleTask at /home/vham/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4344 (discriminator 1)
0x4008710d: vPortTaskWrapper at /home/vham/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

### Steps to reproduce.

1. Get 2 ESP32-DevkitC-V4, connect GPIOs 21 and 22 between them, with a pull-up resistor on both. Use the same power supply with the same 5V and GND for both (no USB power supply) .
2. The ESP32-DevkitC-V4 being the I2C Master should be i2c_master_transmit()'ing to the slave regularly (every 5 seconds in my code)
4. Use the same example code from the I2C driver documentation to read from the slave device on the other side. The slave will occasionally crash.
...

### More Information.

After a lot of troubleshooting I believe I found the root cause. I'm documenting the issue here to track the pull request for the fix.

danielcolchete, 2024-10-30:
The root cause is here: https://github.com/espressif/esp-idf/blob/9106c43accd9f5e75379f62f12597677213f5023/components/esp_driver_i2c/i2c_slave.c#L76

i2c_slave->data_buf allocated to SOC_I2C_FIFO_LEN bytes. t->rcv_fifo_cnt is about a different buffer. The code below will corrupt the memory for the i2c_slave:
i2c_ll_read_rxfifo(hal->dev, i2c_slave->data_buf, t->rcv_fifo_cnt);

danielcolchete, 2024-10-30:
My pull request will likely not be approved because I can't agree with the CLA. We'll likely need someone else to implement the fix.

Alvin1Zhang, 2024-10-30:
Thanks for reporting.

mythbuster5, 2024-11-14:
Reply related to i2c slave: Hi, we recognize this version of i2c slave lack ease of use, and it really has some known problems. So we are coming with the second version of i2c slave driver and usage example, and the second version will be our focus to maintain，which is already online on master branch and will be online on v5.4 release version. Wish this can solve your problem. Thanks for your comprehension.

I will close this issue now.

closed 2024-11-14 by mythbuster5