espressif/esp-idf #14807: deleting last char from the filepath

sova32, 2024-10-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Waveshare esp32-S3-Pico

### Power Supply used.

USB

### What is the expected behavior?

vorcked example
https://github.com/espressif/esp-idf/blob/release/v5.3/examples/protocols/http_server/restful_server/main/rest_server.c

### What is the actual behavior?

E (34498) my_server_http: Failed to open file : /root/web_serve/index.html
my path is
E (34498) my_server_http: Failed to open file : /root/web_serve**r**/index.html

### Steps to reproduce.

step by step example

nileshkale123, 2024-11-29:
Hello @sova32 ,

I tested it locally, and it is working as described in the README file. Could you please try again carefully and let me know if the issue persists?

If you encounter the issue again, it would be helpful if you could share your sdkconfig file and any patch file for the changes you have made to this example.

nileshkale123, 2024-12-05:
Hello @sova32 ,

We have followed your instructions and did not encounter any issues during execution. We recommend carefully retrying the steps as outlined in the README.

For now, we are closing this issue. However, please feel free to reopen it if further assistance is needed.

Thank you!

closed 2024-12-05 by nileshkale123