espressif/esp-idf #14808: Invalidate specific OTA slot.

ddomnik, 2024-10-31:
### Is your feature request related to a problem?

Let's assume we have two valid OTA partitions (ota0 and ota1) and both ota_state's are set to ESP_OTA_IMG_VALID. 
Now ota1 partition gets erased and we perform this incomplete update sequence:
- esp_ota_begin (does not update otadata)
- esp_partition_write (does not update otadata)
- esp_ota_end (does not update otadata)
- esp_image_verify (does not update otadata)
- esp_ota_set_boot_partition is **not** called. Because of a reset or a failed custom "header" check after esp_image_verify.

This would assume we have a valid image in ota1, even tho the update sequence has not been completed.

The same issue would appear if a factory partition is used to reflash ota0 that previously was marked as valid.

### Describe the solution you'd like.

esp_ota_begin() should mark the to be flashed partition as ESP_OTA_IMG_INVALID.
To be more flexible and also to be able to mark specific partitions as invalid a generic function like these would be better:

esp_ota_mark_app_invalid(const esp_partition_t *partition);

esp_ota_set_state(const esp_partition_t *partition, esp_ota_img_states_t ota_state);

### Describe alternatives you've considered.

1. Rewriting the ota_data partition manually, but this seems to be hacky.
2. Erase the partition content, but then the ota_state flag is still valid and the bootloader tries to load it.

### Additional context.

The logic of the OTA data is quite complex to me and I am not even sure if ota_data actually behaves like this. As I think not every ota partition has a dedicated slot where the ota_state is stored? If so I may makes sense to make a function available like esp_ota_mark_app_invalid_rollback_and_reboot without the rollback and reboot part.

mahavirj, 2024-11-26:
@ddomnik 

OTA data partition contains 2 identical flash sectors each storing copies of `esp_ota_select_entry_t otadata[2]` data structure. Sequence number (`ota_seq`) and the state (`ota_state`) are 2 important fields in this data structure that helps to define the active partition. Following code should help to clear the logic behind the usage of sequence number:

https://github.com/espressif/esp-idf/blob/f420609c332fbd2d2f7f188c6579d046c9560e42/components/app_update/esp_ota_ops.c#L490-L504

API `esp_ota_set_boot_partition` is responsible to correctly update both the fields `ota_seq` and `ota_state` in the OTA data partition. If there is a failure in calling this API then the new partition won't be activated yet and hence state (valid or invalid) does not really matter. 

May I  ask if you are observing any problem due to incorrect state of the image in the OTA data partition? If yes, could you please supply the detailed console logs here?

Alvin1Zhang, 2024-12-10:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

KonstantinKondrashov, 2024-12-20:
Hi @ddomnik!
You might use `esp_ota_erase_last_boot_app_partition` it erases **the inactive ota_data slot** and app partition as well. So you can use it.
Also, I see that we can improve the OTA lib as well. And erase t**he inactive ota_data slot** in **esp_ota_begin**.

ddomnik, 2024-12-20:
@KonstantinKondrashov  thank you for your reply.

> Also, I see that we can improve the OTA lib as well. And erase t**he inactive ota_data slot** in **esp_ota_begin**.

This sounds great!

closed 2025-01-09 by espressif-bot

ddomnik, 2025-01-09:
@KonstantinKondrashov  Thank you very much for this feature. Just for clarification regarding this note:
https://github.com/espressif/esp-idf/commit/c9ea30e6c09229acde6d47fbace6b08ff827e170#diff-1ffda434369319674b70b70d59a051dd6e93e2ec2b2652d8c90a9dc632c18ca2R86

Assuming we have the same update process as below, how can this (`esp_ota_begin `) be used together with the Rollback feature? Or is there now a different suggested way of doing updates when the Rollback feature is enabled? Thanks again.

> * esp_ota_begin (does not update otadata)
> * esp_partition_write (does not update otadata)
> * esp_ota_end (does not update otadata)
> * esp_image_verify (does not update otadata)
> * esp_ota_set_boot_partition is **not** called. Because of a reset or a failed custom "header" check after esp_image_verify.

KonstantinKondrashov, 2025-01-10:
Hi @ddomnik!
Yes, we have the same OTA update process, there is no difference.
The esp_ota_begin will take care about this use case, it will invalidate the slot that is currently used for downloading a new image.