espressif/esp-idf #14809: Unable to build (make) because of ncurses

wilmer-suarez-simplisafe, 2024-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm on macOS Sonoma 14.6.1. MacBook Pro with an Apple M2 Pro chip.
ESP-IDF version: `v3.0-dev-7220-gcc5440f6a2` (Unable to upgrade atm but will in the future)

Seeing this error when running `make`.

```
Makefile:20: warning: overriding commands for target `print_flash_cmd'
/Users/wilmer.suarez/Desktop/Repositories/<repo>/esp-idf/make/project.mk:641: warning: ignoring old commands for target `print_flash_cmd'
/Users/wilmer.suarez/Desktop/Repositories/<repo>/esp-idf/tools/kconfig/lxdialog/check-lxdialog.sh -check cc   -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MMD -MP -lncurses -lncurses
 *** Unable to find the ncurses libraries or the
 *** required header files.
 *** 'make menuconfig' requires the ncurses libraries.
 *** 
 *** Install ncurses (ncurses-devel) and try again.
 *** 
make[1]: *** [dochecklxdialog] Error 1
make: *** No rule to make target `/Users/wilmer.suarez/Desktop/Repositories/<repo>/build/include/config/auto.conf', needed by `/Users/wilmer.suarez/Desktop/Repositories/<repo>/build/bootloader/bootloader.bin'.  Stop.
make: *** Waiting for unfinished jobs....
/Users/wilmer.suarez/Desktop/Repositories/<repo>/esp-idf/tools/check_python_dependencies.py:22: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements from /Users/wilmer.suarez/Desktop/Repositories/<repo>/esp-idf/requirements.txt are satisfied.
```

I've tried installing ncurses using brew. Error is still present.

igrr, 2024-10-31:
Hi @wilmer-suarez-simplisafe, I guess you are using v3.3.6, just some tags are missing in your checkout, so `git describe` shows `v3.0-dev-...`.

This version is already EOL and according to our support policy we don't provide further support for it. However I happen to know the solution for this case. Please try to apply the following patch

```patch
diff --git a/tools/kconfig/lxdialog/check-lxdialog.sh b/tools/kconfig/lxdialog/check-lxdialog.sh
index 9e98145983f..964551aa332 100755
--- a/tools/kconfig/lxdialog/check-lxdialog.sh
+++ b/tools/kconfig/lxdialog/check-lxdialog.sh
@@ -53,9 +53,9 @@ trap "rm -f $tmp ${tmp%.tmp}.d" 0 1 2 3 15

 # Check if we can link to ncurses
 check() {
-        $cc -x c - -o $tmp 2>/dev/null <<'EOF'
+        $cc -x c - -o $tmp  <<'EOF'
 #include CURSES_LOC
-main() {}
+int main() {return 0;}
 EOF
        if [ $? != 0 ]; then
            echo " *** Unable to find the ncurses libraries or the"       1>&2
```

and then run `make -C tools/kconfig` from ESP-IDF root directory. After this, building your project should succeed as well.

If `make -C tools/kconfig` still fails after this change, please attach the output again, it should contain more information.

dobairoland, 2024-11-01:
@wilmer-suarez-simplisafe Did the above resolve your issue?

wilmer-suarez-simplisafe, 2024-11-01:
Yes thank you!

closed 2024-11-01 by dobairoland

HankB, 2025-04-13:
I've run into the same issue on Debian Testing (Trixie) and the fix above worked for me too. I'm on the following version of ESP-IDF

```text
(espressif_venv) hbarta@rocinante:~/esp/ESP8266_RTOS_SDK$ git describe --tags
v3.4-104-g2f586ea4
(espressif_venv) hbarta@rocinante:~/esp/ESP8266_RTOS_SDK$ 
```