espressif/esp-idf #14811: C270 UVC Configuration description caught by esp32s3 wrong

shenqili, 2024-11-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DEV-KIT-N8R8

### Power Supply used.

External 5V

### What is the expected behavior?

I want to use Logitech C270 with esp32s3 to capture video and pictures

### What is the actual behavior?

I have run the two project with the Logitech C270 which is supported by esp-idf and both failed.
https://gitcode.com/gh_mirrors/es/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc

In project of usb_camera_mic_spk
I have got the wrong part of the UVC Configuration description without the video format descriptions part.

And in project of uvc.
I have got the wrong part of the UVC Configuration description without the video format descriptions part. The supported video formats are not gotten by the esp32s3, which only shows the 160*120, 10frames,Uncompressed format. 

[ucvLog.txt](https://github.com/user-attachments/files/17595010/ucvLog.txt)
[usb_camera_mic_spk.txt](https://github.com/user-attachments/files/17595014/usb_camera_mic_spk.txt)

### Steps to reproduce.

1. Set the C270 and the esp32s3 (Use a USB hub with power supply to power the camera and ESP32 at the same time
[image]
)
[image]
2. Build the project and finish the menuconfig

3. Run the project

Here is the main.c:

/*
 * SPDX-FileCopyrightText: 2022-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */
[... 654 lines cut ...]
W (1634) uvc_mic_spk_demo: UVC: get frame list size = 0
I (1635) uvc_mic_spk_demo: UAC MIC: get frame list size = 1, current = 0
I (1635) uvc_mic_spk_demo:       [0] ch_num = 1, bit_resolution = 16, samples_frequence = 32000, samples_frequence_min = 0, samples_frequence_max = 0
I (1635) uvc_mic_spk_demo: UAC MIC: use frame[0] ch_num = 1, bit_resolution = 16, samples_frequence = 32000
I (1635) uvc_mic_spk_demo: UAC MIC: use frame[0] ch_num = 1, bit_resolution = 16, samples_frequence = 32000
W (1636) uvc_mic_spk_demo: UAC SPK: get frame list size = 0
I (1636) uvc_mic_spk_demo: Device connected
I (1637) USB_STREAM: Set Device Interface = 3, Alt = 1
I (1645) USB_STREAM: Set frequency endpoint 0x82: (32000) Hz
I (1676) USB_STREAM: USB Device Connected
I (80480) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (80481) wifi:station: f0:d4:15:9f:d4:3b join, AID=1, bgn, 40U
I (80482) camera wifi: station f0:d4:15:9f:d4:3b join, AID=1
I (80676) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (81258) wifi:<ba-add>idx:2 (ifx:1, f0:d4:15:9f:d4:3b), tid:0, ssn:10, winSize:64
W (137377) httpd_uri: httpd_uri: URI '/favicon.ico' not found
W (137377) httpd_txrx: httpd_resp_send_err: 404 Not Found - Nothing matches the given URI
```

### More Information.

I use the windows usb descriptior dumper to get the C270 descrption which is right。 And use amcap to connect the camera and set the format which is also successful.
[image]

[C270.txt](https://github.com/user-attachments/files/17595078/C270.txt)

shenqili, 2024-11-01:
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk
is same as
https://gitcode.com/gh_mirrors/es/esp-iot-solution/tree/master/examples/usb/host/usb_camera_mic_spk

shenqili, 2024-11-01:
I have found the reason is that uvc device has Other Speed Configuration Descriptor (for full speed). And the ideal video format can only be gotten at the high speed.
So the question changes to another. Why C270 is in the supported list and how can I use the C270 with esp32s3? I really want to know, thanks a lot. 
[image]

tore-espressif, 2024-11-01:
I uncommented this line
https://github.com/espressif/esp-idf/blob/9106c43accd9f5e75379f62f12597677213f5023/examples/peripherals/usb/host/uvc/main/main.c#L264
And this is my output with Logitech C270

<details>

<summary>Logitech C270 FS CFG descriptor</summary>

```
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0xef
bDeviceSubClass 0x2
bDeviceProtocol 0x1
bMaxPacketSize0 64
idVendor 0x46d
idProduct 0x825
bcdDevice 0.10
iManufacturer 0
iProduct 0
iSerialNumber 2
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 1117
bNumInterfaces 4
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 500mA
*** Interface Association Descriptor ***
bLength 8
bDescriptorType 11
bFirstInterface 0
bInterfaceCount 2
bFunctionClass 0xe
bFunctionSubClass 0x3
bFunctionProtocol 0x0
iFunction 0
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0xe
        bInterfaceSubClass 0x1
        bInterfaceProtocol 0x0
        iInterface 0
        *** Class-specific VC Interface Descriptor ***
        bLength 0xd
        bDescriptorType 0x24
        bDescriptorSubType 1
        bcdUVC 100
        wTotalLength 160
        dwClockFrequency 48000000
        bFunctionProtocol 1
        bInCollection 1
        baInterfaceNr 18
        *** Input Terminal Descriptor (Camera) ***
        bLength 0x12
        bDescriptorType 0x24
        bDescriptorSubType 2
        bTerminalID 1
        wTerminalType 201
        bAssocTerminal 0
        iTerminal 0
[... 698 lines cut ...]
                          default interval: 1/15
                          interval[0]: 1/15
                          interval[1]: 1/10
                          interval[2]: 1/5
                        FrameDescriptor(12)
                          capabilities: 01
                          size: 800x600
                          bit rate: 38400000-115200000
                          max frame size: 960000
                          default interval: 1/15
                          interval[0]: 1/15
                          interval[1]: 1/10
                          interval[2]: 1/5
```
</details>

You can see many JPEG and Uncompressed formats available

> And the ideal video format can only be gotten at the high speed.

What do you mean by 'ideal video format'?

ESP32-S2 and S3 are Full-Speed only, so it is not possible to stream higher quality (eg. 1280x720) video.

(ESP32-P4 is High-Speed and this is possible)

shenqili, 2024-11-03:
> I uncommented this line
> 
> https://github.com/espressif/esp-idf/blob/9106c43accd9f5e75379f62f12597677213f5023/examples/peripherals/usb/host/uvc/main/main.c#L264
> 
> 
> And this is my output with Logitech C270
> Logitech C270 FS CFG descriptor
> You can see many JPEG and Uncompressed formats available
> 
> > And the ideal video format can only be gotten at the high speed.
> 
> What do you mean by 'ideal video format'?
> 
> ESP32-S2 and S3 are Full-Speed only, so it is not possible to stream higher quality (eg. 1280x720) video.
> 
> (ESP32-P4 is High-Speed and this is possible)

Thanks for responding. The Ideal video format should include MJPEG video format, like 640*480. For the C270 is supported by the esp32s3 which is demonstrated in usb_host_uvc 1.0.4(https://components.espressif.com/components/espressif/usb_host_uvc).
And a youtuber have successfully used it for video tranfering. The link is below:
https://www.youtube.com/watch?v=Hi-6Y12Jhgs.
He used esp32s3 generic and also the C270 and run the project. So I have the doubt what is the wrong with me. I am willing for your feedback. Thx again.

shenqili, 2024-11-04:
[C270-windows vs esp32-log.docx](https://github.com/user-attachments/files/17612579/C270-windows.vs.esp32-log.docx)
I have upload the whole C270 configuration (captured by the windows usb description dumper) and compared to the print log of the ESP32 line by line. And as you seen, the configuration descrption gotten by esp32 is the 'Other Speed Configuration Descriptor'. It should work if get the configuration in the upper part. So I really want to how the examples of usb_host uvc 1.0.4 ror the youtube video can successfully use the C270.

tore-espressif, 2024-11-04:
It appears that your camera does not support any additional video formats in Full-Speed mode. This is unusual, as my Logitech C270 camera, which has a bcdDevice version of 0x12, does support a wider range of formats. In contrast, your camera's bcdDevice version is 0x21, indicating a newer version of the USB device. However, despite being newer, it seems to lack support for certain video formats in Full-Speed.

The discrepancy between the configuration descriptors seen by Windows and ESP-IDF is also notable. While Windows detects the High-Speed descriptor with the full range of video formats, ESP-IDF only accesses the Full-Speed configuration descriptor, which has a reduced subset of formats.

I’m not sure why Logitech chose to limit the available formats in Full-Speed mode for the newer version of the camera. If the data you provided is accurate, this would seem to be the most plausible explanation.

shenqili, 2024-11-04:
Or is there any possible to do some configuration or adjustment in ESP-IDF to access the High Speed Configuration descriptor of C270 I used.

tore-espressif, 2024-11-12:
There is no easy way how you can add USB HighSpeed support to ESP32-S3. 
Transferring your design to Raspberry Pi, or any other linux capable architecture will also give you a lot more computing power for video postprocessing. But it will greatly depend on what do you need to actually do

tore-espressif, 2024-12-02:
I'll close this issue as there wans't any bug. Feel free to reopen if you have more questions