espressif/esp-idf #14815: ESP32-S3 MCPWM has a separate prescaler for each timer, buf IDF code doesn't use them.

eriksl, 2024-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Espressif SoC revision.

ESP32-S3 0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

LilyGO S3 T7

### Power Supply used.

USB

### What is the expected behavior?

I expected to be able to run different timers within one MCPWM group to be able to run at different frequencies. I have selected APB frequency for timer0 and APB / 16 for timer1, so there should be no problem. The technical reference manual says (register overview) there is a common prescaler for all timers and a specific prescaler for each timer, so it should be possible. This is also what the text says, although the diagram of the MCPWM module is indecisive on this matter. 

### What is the actual behavior?

These reports and abort called:
```
E (10) mcpwm: mcpwm_set_prescale(242): group prescale conflict, already is 2 but attempt to 1
E (10) mcpwm: mcpwm_new_timer(125): set prescale failed
```
Besides the first report being very cryptic, it looks like the code simply doesn't take the timer-specific prescaler into account. Or there is none and the documentation is incorrect.

### Steps to reproduce.

1. create a timer on mcpwm group 0, frequency is 80 MHz (1x APB prescaler)
2. create a second timer on mcpwm group, frequency is 5 MHz (16x APB prescaler)

Nothing more required than this.

When you create both timers with the same frequencies it works. If the timers are on separate groups, it works as well.

### Debug Logs.

```plain
E (10) mcpwm: mcpwm_set_prescale(242): group prescale conflict, already is 2 but attempt to 1
E (10) mcpwm: mcpwm_new_timer(125): set prescale failed
```

### More Information.

I am forced to use MCPWM for my lighting ("LED") purposes as the "LEDPWM" has a resolution of only 14 bits, which is really too little. I don't understand why the "fast LEDPWM" channels where dropped (with the high resolution timers). I am now investigating ways to achieve similar PWM resolutions. The 16 bit width of MCPWM is step one, but I'd really like even more. I am thinking of having the cpu updating the PWM "value" on the fly after one or more timer overflow interrupts, but it sounds complex to implement. On the other hand, it may even be done on the LEDPWM module.

Other possible paths:
- use the modulator + duty cycle of the MCPWM to add 3 bits of resolution
- use the modulator of the RMT module, but I can't find maximum parameters
- use the UART, but you can't select 0 stop bits. If that was possible, it could work (although it would require a lot of memory or a lot of data pumping).

I am interested in any interesting hints to workaround this omission.

suda-morris, 2024-11-04:
Didn't reproduce the issue with the following init code:

```c
TEST_CASE("mcpwm timer with different resolutions", "[mcpwm]")
{
    mcpwm_timer_config_t timer_config = {
        .group_id = 0,
        .clk_src = MCPWM_TIMER_CLK_SRC_DEFAULT,
        .resolution_hz = 80 * 1000 * 1000,
        .period_ticks = 20 * 1000,
        .count_mode = MCPWM_TIMER_COUNT_MODE_UP,
    };
    mcpwm_timer_handle_t timer0, timer1;
    printf("create mcpwm timers\r\n");
    TEST_ESP_OK(mcpwm_new_timer(&timer_config, &timer0));
    timer_config.resolution_hz = 5 * 1000 * 1000;
    TEST_ESP_OK(mcpwm_new_timer(&timer_config, &timer1));

    printf("delete timers\r\n");
    TEST_ESP_OK(mcpwm_del_timer(timer0));
    TEST_ESP_OK(mcpwm_del_timer(timer1));
}
```

eriksl, 2024-11-04:
That's interesting. I have tested it more than once. I have run the same code on one and on two mcpwm groups, using two groups it works like a charm, on one group it doesn't.

eriksl, 2024-11-04:
I think I know where's the difference. I am always using a 16 bit (max) resolution timer.

Please try with:
```
mcpwm_timer_config_t timer_config =
{
	.group_id = 0,
	.clk_src = MCPWM_TIMER_CLK_SRC_DEFAULT,
	.resolution_hz = FREQUENCY,
	.count_mode = MCPWM_TIMER_COUNT_MODE_UP,
	.period_ticks = TIMER_TICKS,
	.intr_priority = 0,
	.flags =
	{
		.update_period_on_empty = 0,
		.update_period_on_sync = 0,
	},
};
```
where either
- FREQUENCY = 160000000 / 16 and TIMER_TICKS = 65535
- FREQUENCY = 160000000 / 1  and  TIMER_TICKS = 65535

This is where the problem occurs.

suda-morris, 2024-11-06:
@eriksl this can be workaround by set the higher resolution first then the smaller resolution. 

e.g. 

```c
TEST_CASE("mcpwm timer with different resolutions", "[mcpwm]")
{
    mcpwm_timer_config_t timer_config = {
        .group_id = 0,
        .clk_src = MCPWM_TIMER_CLK_SRC_DEFAULT,
        .resolution_hz = 160 * 1000 * 1000,
        .period_ticks = 65535,
        .count_mode = MCPWM_TIMER_COUNT_MODE_UP,
    };
    mcpwm_timer_handle_t timer0, timer1;
    printf("create mcpwm timers\r\n");
    TEST_ESP_OK(mcpwm_new_timer(&timer_config, &timer0));
    timer_config.resolution_hz = 10 * 1000 * 1000;
    TEST_ESP_OK(mcpwm_new_timer(&timer_config, &timer1));

    printf("delete timers\r\n");
    TEST_ESP_OK(mcpwm_del_timer(timer0));
    TEST_ESP_OK(mcpwm_del_timer(timer1));
}
```

eriksl, 2024-11-19:
So you're saying this is actually undocumented behaviour?

BTW I am doing exactly as in the workaround you posted and exactly then it doesn't work. Maybe you mean the other way around, first create the **lower** frequency and then the higher **frequency** timer?

Kainarx, 2024-11-21:
Hi @eriksl 
Can you try this patch? 
[0001-change-default-mcpwm-group-prescale.patch](https://github.com/user-attachments/files/17845478/0001-change-default-mcpwm-group-prescale.patch)

eriksl, 2024-11-21:
I already worked around it in another way (by using two MCPWM instances). Not ideal, but it works. It's not so easy to change the code back to use one MCPWM instance. If it fixes the issue it's good enough for me ;-)

baldhead69, 2024-11-27:
I have a similar problem.

I am trying to put mcpwm carrier frequency to work with my already working mcpwm code.

[code]
 mcpwm_carrier_config_t carrier_config =
    {
        .clk_src = MCPWM_CARRIER_CLK_SRC_DEFAULT,
        .frequency_hz = 5000,
        .duty_cycle = 0.5,
        .first_pulse_duration_us = 100,
        .flags.invert_after_modulate = false,
        .flags.invert_before_modulate = false,
    };

    ESP_ERROR_CHECK( mcpwm_operator_apply_carrier(operators[0], &carrier_config) );
    ESP_ERROR_CHECK( mcpwm_operator_apply_carrier(operators[1], &carrier_config) );
    ESP_ERROR_CHECK( mcpwm_operator_apply_carrier(operators[2], &carrier_config) );
[/code]

But the following error appear: 
[code]
D (3792) mcpwm: group (0) calc prescale:250, module calc prescale:128
E (3799) mcpwm: mcpwm_set_prescale(242): group prescale conflict, already is 2 but attempt to 250
E (3808) mcpwm: mcpwm_operator_apply_carrier(203): set prescale failed
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4200b0b1
[/code]

Kainarx, 2024-11-28:
@baldhead69 Hi! Thank you for feedback.  
What is the timer resolution of your previous application? In fact, the carrier needs to share the group clk with the timer.  The difference in frequency between the two is too large. Can you try to increase the frequency of the carrier?

eriksl, 2024-11-28:
I think the documentation falls way short in details here. About what frequencies are even possible and also what's exactly possible with carrier modulation. IIRC I distilled it's actually a 3 bit PWM signal, is that correct (duty values 0 - 7)? But nothing about frequencies IIRC.

baldhead69, 2024-11-28:
> @baldhead69 Hi! Thank you for feedback.  
> May I ask under what circumstances such a low carrier frequency is needed? In fact, the carrier needs to share the group clk with the timer.  The difference in frequency between the two is too large. Can you try to increase the frequency of the carrier?
> BTW, the first pulse can't reach 100us in this case either, which is too short for the current frequency.

I am using mcpwm to control a triac, and i tried implement a train of pulses to trigger the triac, only how a option if i needed in a future use case, but i think that it is not possible.

IUnfortunately, Espressif saves on certain things that I don't understand, like in this case not having a prescaler or another clock source or a duty cycle option with a higher resolution, and I know that the function of carrier modulation is another. but......

Another unnecessary saving in my opinion is having a linked list with only 12 bits, they should have made a linked list with at least 24 bits, which was already an "error" in the esp32, so they didn't improve it in the esp32-s3.

https://www.esp32.com/viewtopic.php?f=13&t=42388&start=20

baldhead69, 2024-11-28:
> I think the documentation falls way short in details here. About what frequencies are even possible and also what's exactly possible with carrier modulation. IIRC I distilled it's actually a 3 bit PWM signal, is that correct (duty values 0 - 7)? But nothing about frequencies IIRC.

Yes, documentation is bad regarding the carrier modulation module in technical reference manual.

Yes, carrier modulation has only 8 duty cycles values per documentation.

eriksl, 2024-11-28:
Agreed that Espressif makes some decisions now and then that are hard to explain. I think there is a whole lot going on that we don't understand or can't understand. For example decisions based on business model and I am sure sometimes things don't go as planned. I can imagine, for example, that some chip types, in contrary to what the documentation says, do have a certain hardware module, but that it's implementation is flawed in a way that can't be worked around in software. I can imagine in that case they decide to hide the whole thing instead of marking it "inoperative" and correcting it in the next ECO.

My main gripe in this area is the LED PWM implementation. On the ESP32 there where two modules. Not really clear why you'd need to modules, as each have 8 channels already. So on the ESP32-S3 they dropped one. Fair enough, completely logical. What they dropped the "high speed" one, not the "low speed". And what is "high speed" and "low speed" in the context of LED PWM? Well, the width of the duty timer! So we're now stuck to a PWM that only has 14 bits duty width, which is really far too small for a decent lighting system. Contrary to the "high speed" version, that has 22 bits, which is perfect. The only difference is the width of the timer register! Really, guys. This is the reason I am now forced to use MC-PWM, for my LEDs, because it gives me two bits additional PWM duty width. And I really don't need all of this dead-time, A/B, fault, capturing stuff. Just because of the silly decision that only 14 bits would be needed for PWM LED.

Also the decision to drop Bluetooth Classic I am not entirely happy with. BT-C is really more suited than BT-LE for classic streaming situations like uploading firmware. I really had to kludge this now.

baldhead69, 2024-11-28:
> Agreed that Espressif makes some decisions now and then that are hard to explain. I think there is a whole lot going on that we don't understand or can't understand. For example decisions based on [...]

When I migrated from the ESP32 to the ESP32-S3 I also checked this in the PWM LED, they removed the high speed and left the low speed, for me it wasn't so much of a problem, because I use it for the backlight of an LCD, I just had to make some adaptations and reduce from 16 bits to 14 bits.

eriksl, 2024-11-28:
Yes, for LCD backlight it's pretty good. Even the PDM (sigma-delta) modulator is good enough for that.

But for your living room lightning, it's really a too small range. The thing is that actually the generator is fast enough, by a long shot, so I am not sure why they made this limitation, it looks artificial to me. For most purposes a refresh frequency of 80 Hz is sufficient (even though I mostly double this, to be sure). With 14 bits I now have a refresh of 2440 Hz!

closed 2025-06-01 by espressif-bot