espressif/esp-idf #14816: sta2eth example crash with wired ethernet

chegewara, 2024-11-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master and older

### Espressif SoC revision.

any

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

any

### Power Supply used.

USB

### What is the expected behavior?

To not crash.

### What is the actual behavior?

Hello espressif team.
I dont want to sound rude, but i cant believe this example passed internal tests
https://github.com/espressif/esp-idf/tree/master/examples/network/sta2eth

I wanted to prepare this example for esp32-P4, and maybe even to make PR for it, but it just crash because of very simple mistake.
Here is code that is registering eth events handler where last argument is `NULL`:
https://github.com/espressif/esp-idf/blob/master/examples/network/sta2eth/main/ethernet_iface.c#L282
and here is code which is trying to use that argument:
https://github.com/espressif/esp-idf/blob/master/examples/network/sta2eth/main/ethernet_iface.c#L47

### Steps to reproduce.

Build and flash with ethernet wired interface. (with USB wired it works)

Bartarian, 2024-11-03:
I think I have the same problem, esp-idf 5.3.1 and component ethernet_init 0.3.2 are not compatible.

For me downgrading ethernet_init to 0.3.0 fixed the issue, for now.

I guess it will be fixed when they will release a new esp-idf version, which includes 0738314

chegewara, 2024-11-04:
Hi @david-cermak 
No, i am not using PPP. It was briefly tested with ethernet as a wired connection.
Like I said, with USB as wired netif it works great and I love it, because I've been thinking about how to do it for quite some time.

Since I have only basic knowledge about networking I gave up when this failed with ethernet.

Thanks

closed 2024-11-13 by espressif-bot