espressif/esp-idf #14817: ESP32-C6 WiFi Station Error (`wifi:m f null`)

fuzeman, 2024-11-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

WiFi Station connects to an Access Point and continues operating normally without errors.

### What is the actual behavior?

WiFi Station connects to an Access Point, receives an IP Address , and after a 30 seconds starts logging "wifi:m f null" continuously blocking other tasks from operating.

### Steps to reproduce.

1. Build `examples/wifi/getting_started/station` with ssid + password, and upload to board.
2. Observe station successfully connects to the Access Point.
3. Wait 30 seconds for `wifi:m f null` to start being printed.

### Debug Logs.

```plain
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len�ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c110,len:0xe2c
load:0x4086e610,len:0x2e5c
entry 0x4086c11a
I (23) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (24) boot: compile time Nov  3 2024 15:59:40
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
[... 276 lines cut ...]
    <td>WiFi</td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
  </tr>
  <tr>
    <td rowspan="2">v5.3.1</td>
    <td>STA_GOT_IP</td>
    <td>✔️ OK</td>
    <td>✔️ OK</td>
    <td>✔️ OK</td>
    <td>✔️ OK</td>
  </tr>
  <tr>
    <td>WiFi</td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
    <td>❌ ERROR (>30s)<br/><i>(m f null, wdt)</i></td>
  </tr>
</tbody>
</table>

#12488, #12464, #11615 report similar `wifi:m f null` issues, but all appear to have fixes implemented or were caused by memory exhaustion outside of the esp-idf framework.

fuzeman, 2024-11-04:
Looking at other similar issues, there was a suggestion in https://github.com/espressif/esp-idf/issues/11615#issuecomment-1612718659 to log the result of `esp_get_free_internal_heap_size` inside `malloc_internal_wrapper` to confirm memory usage. Which was added as follows:

```
void *IRAM_ATTR esp_coex_common_malloc_internal_wrapper(size_t size)
{
    void * result = heap_caps_malloc(size, MALLOC_CAP_8BIT | MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);

    printf("m.%d\n", esp_get_free_internal_heap_size());
    fflush(stdout);

    return result;
}
```

This provides the following results which show free memory usage stabilizing to ~252 KB:

```
[...]
W (46481) wifi:m f null

m.252848
m.253032
m.253032
m.252848
W (46481) wifi:m f null

m.252848
m.252848
m.252848
m.252848
m.252848
m.252848
W (46491) wifi:m f null

m.252848
m.252848
m.252848
W (46491) wifi:m f null

m.252848
m.252848
m.252848
m.252848
m.252848
m.252848
W (46501) wifi:m f null

m.252848
m.252848
m.252848
m.252848
m.252848
W (46511) wifi:m f null

m.252848
m.252848
m.252848
W (46511) wifi:m f null

m.252848
m.252848
m.252848
m.252848
W (46521) wifi:m f null
[...]
```

AxelLin, 2024-11-07:
@fuzeman
Can you call  esp_wifi_statis_dump(0xffff) each some minutes and share the log when the issue happen? (reference https://github.com/espressif/esp-idf/issues/8953#issuecomment-1951618720)

fuzeman, 2024-11-07:
Thanks for the response.

See the statis dump below. Note that this is esp-idf v5.3.1 with the `wifi/getting_started/station` example in default power-save mode (e.g. `esp_wifi_set_ps` not set).

```
W (60410) wifi:m f null

W (60410) wifi:m f null

W (60410) wifi:m f null

W (60410) wifi:m f null
I (60420) wifi: BUFFER:
I (60420) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (60420) wifi: 1  32  0     12          12         0        0
I (60430) wifi: 2  32  0     1           1          0        0
I (60440) wifi: 3  32  32    12905       12873      0        11958
I (60440) wifi: 4  32  0     1           1          0        0
I (60450) wifi: 5  0   0     4           4          0        0
I (60460) wifi: 6  0   0     0           0          0        0
I (60460) wifi: 7  32  32    287         255        0        246
I (60470) wifi: 8  5   5     629         629        0        0
I (60470) wifi: 9  0   0     0           0          0        0
I (60480) wifi: 10 2   2     2           2          0        0
I (60490) wifi:
I (60490) wifi: LMAC TX:
I (60490) wifi: tx_all   lrc src age to
I (60490) wifi: 12919    0   0   24  0
I (60500) wifi: 0        0   0   0   0
I (60500) wifi: 0        0   0   0   0
I (60500) wifi: 0        0   0   0   0
I (60510) wifi: 0        0   0   0   0
I (60510) wifi:
I (60510) wifi: LMAC RX:
I (60520) wifi: ampdu    mdpu     mgmt     ctrl     frag
I (60520) wifi: 0        590      565      0        0
I (60530) wifi:
I (60530) wifi: HW RX:
I (60530) wifi: rx_suc   rx_end   full#1   full#2
I (60530) wifi: 484      4a6b     2        1
I (60540) wifi:
I (60540) wifi: Hardware:
I (60540) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (60550) wifi: 0        0        0        0        86c      ac40b99  e0       0
I (60560) wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
I (60570) wifi: 0        41e      0        0        3240     0        0        0
I (60570) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic
I (60580) wifi: 0        0        0        0        c        0        1        0        0
I (60590) wifi:
I (60590) wifi: DIAG:
I (60590) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6
I (60600) wifi: 0        0        fe00a9a  0        188b0001 100008c4 0
I (60610) wifi: diag7    diag8    diag9    diag10   diag12   diagsel
I (60610) wifi: 40       5b4d800  82c000   40       240000   0
I (60620) wifi:
I (60620) wifi: Power save:
I (60620) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent
I (60640) wifi: 578      556      0        252      296      8        7        8        7        30       304      304      28080031 2656627  30736658 91.36 %
I (60650) wifi:
I (60650) wifi: HMAC RX:
I (60660) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (60660) wifi: sta 909      344      0      565    558
I (60670) wifi: ap  0        0        0      0      0
I (60670) wifi:
I (60670) wifi: ifx auth     assoc    deauth action amsdu
I (60680) wifi: sta 1        1        0      5      1
I (60680) wifi: ap  0        0        0      0      0
I (60690) wifi:
I (60690) wifi: HMAC TX:
I (60690) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (60700) wifi: 12       0        0      0      0      0       0

W (60710) wifi:m f null

W (60710) wifi:m f null

W (60710) wifi:m f null
```

fuzeman, 2024-11-07:
Out of interest, I've also captured a statis dump from esp-idf v5.2.1 with power-save disabled (e.g. esp_wifi_set_ps set to `WIFI_PS_NONE`). Not sure if it has any other clues.

```
W (5413) wifi:m f null

W (5413) wifi:m f null

W (5413) wifi:m f null

W (5413) wifi:m f null
I (5423) wifi: BUFFER:
I (5423) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (5423) wifi: 1  32  0     2           2          0        0
I (5433) wifi: 2  32  0     1           1          0        0
I (5443) wifi: 3  32  32    2073        2041       0        1876
I (5443) wifi: 4  32  0     1           1          0        0
I (5453) wifi: 5  0   0     4           4          0        0
I (5453) wifi: 6  0   0     0           0          0        0
I (5463) wifi: 7  32  0     44          44         0        0
I (5473) wifi: 8  5   5     68          68         0        0
I (5473) wifi: 9  0   0     0           0          0        0
I (5483) wifi: 10 2   2     2           2          0        0
I (5493) wifi:
I (5493) wifi: LMAC TX:
I (5493) wifi: tx_all   lrc src age to
I (5493) wifi: 2077     0   0   10  0
I (5503) wifi: 0        0   0   0   0
I (5503) wifi: 0        0   0   0   0
I (5503) wifi: 0        0   0   0   0
I (5513) wifi: 0        0   0   0   0
I (5513) wifi:
I (5513) wifi: LMAC RX:
I (5513) wifi: ampdu    mdpu     mgmt     ctrl     frag
I (5523) wifi: 0        55       52       0        0
I (5523) wifi:
I (5523) wifi: HW RX:
I (5533) wifi: rx_suc   rx_end   full#1   full#2
I (5533) wifi: 6d       aaf      0        1
I (5543) wifi:
I (5543) wifi: Hardware:
I (5543) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (5553) wifi: 0        0        0        0        9e       1b001b8  9        0
I (5553) wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
I (5563) wifi: 0        a4       0        0        7f4      0        0        0
I (5573) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic
I (5583) wifi: 0        0        0        0        8        0        0        0        0
I (5593) wifi:
I (5593) wifi: DIAG:
I (5593) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6
I (5603) wifi: 3        0        fe00a63  0        188f8001 1000082b 0
I (5603) wifi: diag7    diag8    diag9    diag10   diag12   diagsel
I (5613) wifi: 40       0        82c000   40       240000   0
I (5623) wifi:
I (5623) wifi: Power save:
I (5623) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent
I (5633) wifi: 42       42       0        0        0        0        0        0        1989     22       0        0        0        0        0
   nan   %
I (5653) wifi:
I (5653) wifi: HMAC RX:
I (5653) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (5663) wifi: sta 107      55       0      52     47
I (5663) wifi: ap  0        0        0      0      0
I (5673) wifi:
I (5673) wifi: ifx auth     assoc    deauth action amsdu
I (5673) wifi: sta 1        1        0      3      0
I (5683) wifi: ap  0        0        0      0      0
I (5683) wifi:
I (5683) wifi: HMAC TX:
I (5693) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (5693) wifi: 2        0        0      0      0      0       0

W (5703) wifi:m f null

W (5703) wifi:m f null

W (5703) wifi:m f null

W (5713) wifi:m f null
```

fuzeman, 2024-11-27:
Are there any more details or tests which would be helpful to resolve this issue? Do you think it's related to the access point configuration?

Also, note that these issues aren't present in another project instead utilizing the ESP32-S3, so at this stage, it appears to be specific to the ESP32-C6.

fuzeman, 2025-02-01:
Thought I would revisit this issue after noticing ESP-IDF v5.4 had several WiFi changes, and can confirm this bug has been fixed. Have had it running for 8+ hours with no lock-ups encountered using `WIFI_PS_MIN_MODEM` or `WIFI_PS_MAX_MODEM` power-save modes.

Note that these errors are still present with `WIFI_PS_NONE `, but this matches the previous behaviour of ESP-IDF <= 5.2.1, so guessing this is a known limitation?

closed 2025-02-01 by fuzeman

fuzeman, 2025-02-01:
I found a similar discussion in https://github.com/espressif/esp-idf/issues/9595 that WiFi Modem Sleep is required for BT coexistence which makes sense - seems interesting that this is required on the ESP32C6 with no Bluetooth enabled though? maybe related to the addition of the IEEE 802.15.4 radio present on this chip?