espressif/esp-idf #14820: esp_ieee802154_set_ack_timeout isn't implemented

sethtroisi, 2024-11-04:
### Is your feature request related to a problem?

This is defined in [components/ieee802154/include/esp_ieee802154.h](https://github.com/espressif/esp-idf/blob/master/components/ieee802154/include/esp_ieee802154.h) but never implemented.

chshu, 2024-11-04:
The current 802.15.4 driver and hardware handle the Ack process internally:
- Tx Ack: When a unicast frame is received, the hardware generates and transmits the Ack frame, while the software configures the pending bit in the SFD event.
- Rx Ack: After a frame is transmitted, the hardware waits for the Ack frame. If the correct Ack is received, an `rx_ack_done` event is triggered. If the ack times out, a `tx failed due to ack timeout` event is triggered.

The user can't disable the hardware auto Ack with current APIs, so `esp_ieee802154_set_ack_timeout` is not applicable. Could you share your use case that requires modifying the Ack timeout?

Alvin1Zhang, 2025-01-09:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/9d1a4ebac40c035270500415f6567da170b70f21 and has been back ported until release/5.1, feel free to reopen.

closed 2025-01-09 by Alvin1Zhang