espressif/esp-idf #14823: Make KConfig configurable the esp_console tx/rx driver buffer size when using CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG

alessandro-al2tech, 2024-11-04:
### Is your feature request related to a problem?

If i have to log a string (e.g., a json) graeter than 256 bytes on esp console using  sb serial/jtag console i have to manually splits the string calling multiple times the ESP_LOGx function.

### Describe the solution you'd like.

1. the esp_console installs the usb serial_jtag driver with the  the default configuration: https://github.com/espressif/esp-idf/blob/v5.1.3/components/console/esp_console_repl.c#L154.
2. The default configuration is defined here: https://github.com/espressif/esp-idf/blob/v5.1.3/components/driver/usb_serial_jtag/include/driver/usb_serial_jtag.h#L27
3.Could you add KConfig to esp_console component in order to set a custom tx/rx buffer size (e.g.CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG_TX_BUFF_SIZE, CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG_RX_BUFF_SIZE)?

igrr, 2024-11-04:
> If i have to log a string (e.g., a json) geater than 256 bytes on esp console using usb serial/jtag console i have to manually splits the string calling multiple times the ESP_LOGx function.

Hi @alessandro-al2tech, could you please describe the behavior you observe if you don't manually split the string? If you observe an issue, please also attach the code which can be used to reproduce it. Thank you.

alessandro-al2tech, 2024-11-04:
Hi, @igrr  on esp32-c3 using esp idf 5.1.3 enable CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y, intitilizes the console like in https://github.com/espressif/esp-idf/blob/v5.0/examples/system/console/basic/main/console_example_main.c#L95:  and then simply log a string longer than 256 bytes, e.g:
          ```
      char* test = "Lorem ipsum dolor sit amet, consectetur adipiscing elit. Phasellus finibus, odio et tincidunt venenatis, orci nisi bibendum leo, id aliquam enim ex in felis. Nulla facilisi. Vivamus gravida orci at urna volutpat, eu commodo felis ultrices. Fusce egestas eget justo nec vehicula. Integer at accumsan nisl, vitae elementum arcu. Nam sit amet nisi ut lacus pharetra viverra. Maecenas et erat fringilla, sodales justo et, sagittis arcu.";
                ESP_LOGI(m_tag,"%s",test);
```
You should see that output is truncated