espressif/esp-idf #14824: I am not able  to buld the default program of the examples/openthread/ot_br

shubhamk-krishworks, 2024-11-04:
[image]

When i m trying to set the idf_target then it is showing the error.

zwx1995esp, 2024-11-05:
@gytxxsy Please take a look.

shubhamk-krishworks, 2024-11-05:
I have a question can we make the matter enabled router or hub through the ESP just like Aqara Hub M2 (https://dbgtechstore.in/products/aqara-hub-m2-1). So that hub execute all matter related command.

chshu, 2024-11-13:
@shubhamk-krishworks Regarding the compile issue, it seems that the IDF env is not setup properly. Could you setup the IDF env follow the steps here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html, and then compile the `hello_world` example to confirm the setup is ok.

> I have a question can we make the matter enabled router or hub through the ESP just like Aqara Hub M2 (https://dbgtechstore.in/products/aqara-hub-m2-1). So that hub execute all matter related command.

We do have similar solutions available. You can refer to the following example, it's a Rainmaker based Matter controller and Thread BR:
- https://github.com/espressif/esp-rainmaker/tree/master/examples/matter/matter_controller

closed 2025-02-26 by chshu