espressif/esp-idf #14825: ESP-IDF tusb_midi example stack overflow

kevvy-log-ins, 2024-11-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3 Chip rev v0.2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

As per tutorial, the board should show up as a MIDI device, and a sequence of MIDI notes should be transmitted.

### What is the actual behavior?

Board shows up as a MIDI device, and reports the following, as expected:
```
I (485) TinyUSB: TinyUSB Driver installed
I (495) example: USB initialization DONE
I (495) example: MIDI write task init
I (505) example: MIDI read task init
I (505) main_task: Returned from app_main()
I (785) example: Writing MIDI data 74
```

It then either crashes, or continues to successfully write a few more notes from the MIDI note sequence, then crashes.
The TinyUSB MIDI device listed in macOS' Audio MIDI Setup app flickers to show the device being enabled and disabled over and over again as the ESP example reboots itself and tries again.

### Steps to reproduce.

1. Follow the standard ["Start a Project" instructions](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/get-started/linux-macos-setup.html#start-a-project) from the IDF Programming Guide to copy, configure, build, flash, and monitor a project.
2. Follow the instruction at the bottom of the project's [README](https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/examples/peripherals/usb/device/tusb_midi/README.md): "Disconnect UART-to-USB port and connect the native USB port to a computer then the device should show up as a USB MIDI Device while outputting notes."

### Debug Logs.

```plain
I (31) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (31) boot: compile time Nov  4 2024 12:16:04
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.2
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 8MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a7ech ( 42988) map
I (111) esp_image: segment 1: paddr=0001a814 vaddr=3fc92800 size=02b14h ( 11028) load
I (114) esp_image: segment 2: paddr=0001d330 vaddr=40374000 size=02ce8h ( 11496) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1c048h (114760) map
I (150) esp_image: segment 4: paddr=0003c070 vaddr=40376ce8 size=0bab8h ( 47800) load
I (166) boot: Loaded app from partition at offset 0x10000
I (166) boot: Disabling RNG early entropy source...
I (178) cpu_start: Multicore app
I (187) cpu_start: Pro cpu start user code
I (187) cpu_start: cpu freq: 160000000 Hz
I (188) app_init: Application information:
I (190) app_init: Project name:     tusb_midi
I (195) app_init: App version:      1
I (200) app_init: Compile time:     Nov  4 2024 12:15:51
I (206) app_init: ELF file SHA256:  9d69322b5...
I (211) app_init: ESP-IDF:          v5.3.1
I (216) efuse_init: Min chip rev:     v0.0
I (221) efuse_init: Max chip rev:     v0.99 
I (226) efuse_init: Chip rev:         v0.2
I (230) heap_init: Initializing. RAM available for dynamic allocation:
I (238) heap_init: At 3FC960A8 len 00053668 (333 KiB): RAM
I (244) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (250) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (256) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (263) spi_flash: detected chip: gd
I (266) spi_flash: flash io: dio
I (271) sleep: Configure to isolate all GPIO pins in sleep state
I (277) sleep: Enable automatic switching of GPIO sleep configuration
I (285) main_task: Started on CPU0
I (315) main_task: Calling app_main()
I (315) example: USB initialization
W (315) tusb_desc: No Device descriptor provided, using default.
I (315) tusb_desc: 
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 0           │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 0           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 0           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4008      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (485) TinyUSB: TinyUSB Driver installed
I (495) example: USB initialization DONE
I (495) example: MIDI write task init
I (505) example: MIDI read task init
I (505) main_task: Returned from app_main()
I (785) example: Writing MIDI data 74
I (1075) example: Writing MIDI data 78
I (1355) example: Writing MIDI data 81
I (1645) example: Writing MIDI data 86
I (1935) example: Writing MIDI data 90
I (2215) example: Writing MIDI data 93
I (2465) example: Read - Time (ms since boot): 2197462, Data: 06 F0 F7 00
I (2465) example: Read - Time (ms since boot): 2197877, Data: 04 F0 7E 7F
I (2465) example: Read - Time (ms since boot): 2201602, Data: 04 0D 70 02
I (2475) example: Read - Time (ms since boot): 2209066, Data: 04 0E 02 3A
I (2475) example: Read - Time (ms since boot): 2216531, Data: 04 22 7F 7F
I (2485) example: Read - Time (ms since boot): 2223995, Data: 04 7F 7F 11
I (2495) example: Read - Time (ms since boot): 2231460, Data: 04 00 00 01
I (2505) example
***ERROR*** A stack overflow in task midi_task_read_ has been detected.

Backtrace: 0x40375b16:0x3fc9c160 0x40379e01:0x3fc9c180 0x4037ac86:0x3fc9c1a0 0x4037be8f:0x3fc9c220 0x4037ad4c:0x3fc9c240 0x4037ad42:0xa5a5a5a5 |<-CORRUPTED
--- 0x40375b16: panic_abort at /Users/kevin/esp/esp-idf/components/esp_system/panic.c:463
0x40379e01: esp_system_abort at /Users/kevin/esp/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4037ac86: vApplicationStackOverflowHook at /Users/kevin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:554
0x4037be8f: vTaskSwitchContext at /Users/kevin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
0x4037ad4c: _frxt_dispatch at /Users/kevin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x4037ad42: _frxt_int_exit at /Users/kevin/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246
```

### More Information.

This is on latest macOS, Sequoia 15.1 Intel.

tore-espressif, 2024-11-05:
Hello @kevvy-log-ins could you please try increasing the stack size to `4 * 1024` on this line?

https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/examples/peripherals/usb/device/tusb_midi/main/tusb_midi_main.c#L184

kevvy-log-ins, 2024-11-05:
Thank you so much, the example functions as expected when increasing the buffer size to 4 * 1024.

Do you know what effect this buffer size has on latency?

tore-espressif, 2024-11-06:
It has no effect on latency

Alvin1Zhang, 2024-12-17:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/be42634dc9c057c20e88fd05d3b8ea87b87a7eb4, feel free to reopen.

closed 2024-12-17 by Alvin1Zhang