espressif/esp-idf #14826: ESP32-C3 BLE server disconnect from client after 30s

yihan-liu, 2024-11-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**ESP-IDF version**: v5.2.3
**Python version**: 3.11.3
**Hardware**: ESP32-C3 custom board
**OS**: Win 11 / iOS

### What I expect

I have a custom device that uses ESP32-C3-MINI-1 module to run a BLE application. The device reads data from an AS7341 color sensor and send to the client via bluetooth notification. My main.c (shown below) works all good. It can connect to the client and send notification for 30s, before disconnect automatically. The problem appears with both iOS (nRF connect) and windows (bleak) client.

### What I have done

My main.c:

```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_bt.h"
#include "esp_gap_ble_api.h"
#include "esp_gatts_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_main.h"
#include "esp_gatt_common_api.h"
#include "driver/i2c.h"
#include "driver/i2c_master.h"
#include "driver/gptimer.h"
#include "driver/ledc.h"
#include "sdkconfig.h"

#include "sdm_as7341.h"

#define GPIO_LED_CTRL                       GPIO_NUM_0
#define GPIO_NIR_CTRL                       GPIO_NUM_4

#define LEDC_MODE                           LEDC_LOW_SPEED_MODE
#define LEDC_TIMER                          LEDC_TIMER_0
#define LEDC_DUTY_RESOLUTION                LEDC_TIMER_8_BIT
#define LEDC_FREQUENCY                      10000

// PWM for AS clock signal
#define LEDC_AS_OUTPUT_IO                   GPIO_NUM_10
#define LEDC_AS_CHANNEL                     LEDC_CHANNEL_0
#define LEDC_AS_DUTY                        127             // ((2 ** 8) - 1) * 50% = 127

// PWM for LED brightness control
#define LEDC_PWM_OUTPUT_IO                  GPIO_NUM_1
#define LEDC_PWM_CHANNEL                    LEDC_CHANNEL_1
#define LEDC_PWM_DUTY                       2               // ((2 ** 8) - 1) * 1% = 2

// I2C config
#define I2C_MASTER_SCL_IO                   GPIO_NUM_18
#define I2C_MASTER_SDA_IO                   GPIO_NUM_19
#define I2C_MASTER_FREQ_HZ                  1000000         // 1 MHz
#define I2C_MASTER_NUM                      I2C_NUM_0

[... 896 lines cut ...]
Frame 2002: 7 bytes on wire (56 bits), 7 bytes captured (56 bits) on interface TCP@127.0.0.1:24352, id 0
Bluetooth
    [Source: controller]
    [Destination: host]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: HCI Event (0x04)
Bluetooth HCI Event - Disconnect Complete

No.     Time           Source                Destination           Protocol Length Info
   2003 60.841683      remote ()             localhost ()          L2CAP    9      Rcvd Connection oriented channel

Frame 2003: 9 bytes on wire (72 bits), 9 bytes captured (72 bits) on interface TCP@127.0.0.1:24352, id 0
Bluetooth
    [Source: 00:00:00_00:00:00 (00:00:00:00:00:00)]
    [Destination: 00:00:00_00:00:00 (00:00:00:00:00:00)]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol

```

Any help is welcomed! Please inform if any further information is needed.

esp-zhp, 2024-11-08:
1. **Error Explanation**

The error code `(44605) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13` indicates that the peer device disconnected the connection. This means that it was the iOS or Windows 11 device that disconnected, rather than the ESP32-C3 actively terminating the connection.

2. **Solution**

In the code you provided, it seems there are no SMP (Security Manager Protocol) parameters set. To connect with a Windows 11 device, it requires security configuration. Specifically, you need to enable SMP:

Navigate to:
```
(Top) → Component config → Bluetooth → Bluedroid Options
[*]     Include BLE security module(SMP)
```

Then, configure the SMP settings in your code. You can refer to the example in `esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server` for guidance:

```c
/* Set the security parameters for IO capabilities, authentication requirements, key size, and key exchange */
esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND; // Bond with peer device after authentication
esp_ble_io_cap_t iocap = ESP_IO_CAP_NONE;           // Set IO capability to No output, No input
uint8_t key_size = 16;      // Key size should be between 7 and 16 bytes
uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;

// Set static passkey
uint32_t passkey = 123456;
uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
uint8_t oob_support = ESP_BLE_OOB_DISABLE;

esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));

/* If your BLE device is a slave, the init_key specifies the key types you expect the master to share with you.
   If your BLE device is a master, the init_key specifies the key types you will share with the slave. */
esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));
```

yihan-liu, 2024-11-08:
> 1. **Error Explanation**
>
> The error code `(44605) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13` indicates that the peer device disconnected the connection. This means that it was the iOS or Windows [...]

Thanks, I will give it a try and let you know the result shortly!

esp-zhp, 2024-11-13:
@yihan-liu 
Based on your feedback, I still believe this might be an SMP issue. Try flashing your board with either esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server or esp-idf/examples/bluetooth/bluedroid/ble/ble_hid_device_demo.

Once that’s done, connect the ESP device to both a PC and an iPhone. Do you still experience disconnections in this setup?

esp-zhp, 2024-11-13:
If the PC or iPhone disconnects from the ESP device after 30 seconds, this sounds like an ATT timeout. Typically, this happens when an ATT request packet doesn’t receive a response, such as when the peer sends a write request that the ESP device fails to respond to within 30 seconds. In this case, the iPhone or PC will disconnect from the ESP device. Could you check if this is happening in your setup?

Based on your description, this does not appear to be an issue with the BLE stack itself.

yihan-liu, 2024-11-13:
> If the PC or iPhone disconnects from the ESP device after 30 seconds, this sounds like an ATT timeout. Typically, this happens when an ATT request packet doesn’t receive a response, such as when the [...]

How do I check for that? Will a Wireshark log be useful? On my computer I am using the [Bleak](https://github.com/hbldh/bleak) api to read from the device if that helps you to know the procedure. In case you find this to be useful, here I attached the receiver code, I have not explicitly implement any write request but it still could happen:

```
import sys
import logging
import asyncio
from collections import deque

from bleak import BleakScanner, BleakClient
from PyQt5 import QtWidgets, QtCore
from pyqtgraph import PlotWidget, mkPen
import pyqtgraph as pg
from qasync import QEventLoop, asyncSlot

logger = logging.getLogger(__name__)

class BTVizApp(QtWidgets.QMainWindow):
    def __init__(self):
        super().__init__()
        self.setWindowTitle('BTViz')
        self.device_name = ''
        self.client = None
        self.characteristic_uuid = None
        self.data_buffer = []
        self.plots = []
        self.curves = []
        self.timestamps = []
        self.init_ui()

    def init_ui(self):
        # Create input dialog to get BLE device name
        self.device_name, ok = QtWidgets.QInputDialog.getText(
            self, 'Device Name Input', 'Enter BLE Device Name:', QtWidgets.QLineEdit.EchoMode.Normal, 'SpectraDerma')
        if not ok or not self.device_name:
            QtWidgets.QMessageBox.critical(
                self, 'Error', 'No device name entered. Exiting.')
            sys.exit(1)

        # Set up central widget and layout
        self.central_widget = QtWidgets.QWidget()
        self.setCentralWidget(self.central_widget)
        self.layout = QtWidgets.QVBoxLayout(self.central_widget)

        # Status label
        self.status_label = QtWidgets.QLabel('Connecting to device...')
        self.layout.addWidget(self.status_label)

        # Start BLE connection
        asyncio.ensure_future(self.start_ble())

    @asyncSlot()
    async def start_ble(self):
        # Scan for devices
        devices = await BleakScanner.discover()
        target_device = None
        for d in devices:
            if d.name == self.device_name:
                target_device = d
                break

        if not target_device:
            QtWidgets.QMessageBox.critical(
                self, 'Error', f'Device {self.device_name} not found.')
            sys.exit(1)

        self.status_label.setText(f'Connecting to {self.device_name}...')

        # Connect to device
        async with BleakClient(target_device, winrt=dict(use_cached_services=False)) as self.client:
            try:
                await self.client.connect()

                services = self.client.services
                # Assuming only one custom service and characteristic
                for service in services:
                    for char in service.characteristics:
                        if 'notify' in char.properties or 'indicate' in char.properties:
                            self.characteristic_uuid = char.uuid
                            break

                if not self.characteristic_uuid:
                    QtWidgets.QMessageBox.critical(
                        self, 'Error', 'No notifiable characteristic found.')
                    sys.exit(1)

                self.status_label.setText('Connected. Starting notifications...')
                await asyncio.sleep(1.0)
                await self.client.start_notify(self.characteristic_uuid, self.notification_handler)

            except Exception as e:
                QtWidgets.QMessageBox.critical(
                    self, 'Error', f'Failed to connect: {e}')
                print(e)
                sys.exit(1)

    def notification_handler(self, sender, data):
        """
        Handle incoming data from the BLE device.
        """
        # Decode and parse the data
        try:
            text = data.decode('utf-8').strip()
            values = [float(v) for v in text.split(',')]
        except Exception as e:
            print(f"Failed to parse data: {e}")
            return

        if not self.plots:
            # Initialize plots based on number of channels
            num_channels = len(values)
            for i in range(num_channels):
                plot_widget = PlotWidget()
                plot_widget.enableAutoRange(axis='y')
                plot_widget.showGrid(x=True, y=True)
                plot_widget.setTitle(f"Channel {i+1}")
                pen = mkPen(color=pg.intColor(i))
                curve = plot_widget.plot(pen=pen)
                self.layout.addWidget(plot_widget)
                self.plots.append(plot_widget)
                self.curves.append(curve)

                self.data_buffer.append(deque(maxlen=100))
                self.timestamps.append(deque(maxlen=100))

        # Append data and update plots
        timestamp = QtCore.QTime.currentTime().msecsSinceStartOfDay() / 1000.0
        for i, value in enumerate(values):
            self.data_buffer[i].append(value)
            self.timestamps[i].append(timestamp)
            self.curves[i].setData(list(self.timestamps[i]), list(self.data_buffer[i]))

def main():
    app = QtWidgets.QApplication(sys.argv)
    loop = QEventLoop(app)
    asyncio.set_event_loop(loop)
    window = BTVizApp()
    window.show()
    with loop:
        loop.run_forever()

if __name__ == "__main__":
    main()
```

It still will be nice if you can let me know whether a Wireshark log is necessary! Thanks!

esp-zhp, 2024-11-14:
如果有 Wireshark 抓包的话，我们可以通过分析数据包来定位问题。你可以**通过 Wireshark 抓包查看 ESP 设备是否有未回复的 ATT 包**。
对于 windows 电脑，你也可以获取 windows 电脑的 HCI log，来看看 windows 电脑没有得到哪些回复。
通过 Wireshark  来判断更好一些。

另外，如果你希望协议栈自动回复各种 ATT 请求，可以了解以下结构体：

```c
typedef struct {
    /**
     * @brief 控制对读/写操作的响应处理方式。
     *
     * - 如果设置为 @c ESP_GATT_RSP_BY_APP，应用层负责生成响应。
     * - 如果设置为 @c ESP_GATT_AUTO_RSP，GATT 协议栈会自动生成响应。
     */
    uint8_t auto_rsp;
} esp_attr_control_t;
```
具体用法可以参考：
`esp-idf/examples/bluetooth/bluedroid/ble/gatt_server_service_table/main/gatts_table_creat_demo.c`

将 `auto_rsp` 设置为 1 后，协议栈就会自动回复部分请求。

你当前使用的示例基于：
`esp-idf/examples/bluetooth/bluedroid/ble/gatt_server`

在这个示例中，需要通过 `esp_ble_gatts_send_response` 来回复部分 ATT 请求。如果未进行回复，对端设备可能会主动断开连接。

esp-zhp, 2024-11-15:
ok,Thank you for your feedback. If there are no further questions, this issue can be closed for now. Feel free to reach out anytime if you have additional concerns. Wishing you success in your work!

closed 2024-11-15 by yihan-liu