espressif/esp-idf #14829: LEDC PWM with LIGHT_SLEEP issue: GPIO can hold 1 or 0 but not but PWM duty

josthlee, 2024-11-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have researched how LEDC PWM works with light sleep mode and found a workaround. However, after testing it, I discovered it only works when the PWM duty cycle is set to either fully on or zero. From my understanding, this workaround seems effective only when the GPIO is either fully on (100% duty) or completely off (0% duty). I'm struggling with the solution that allows light sleep mode to support true PWM, where the GPIO can maintain the set duty cycle during light sleep.

Here is my test code with ESP32C6 on IDF5.3,

```
#include <stdio.h>
#include <unistd.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/ledc.h"
#include "esp_sleep.h"
#include "esp_timer.h"
#include "esp_log.h"

#define ESP_LOG_FLUSH(dev) (fflush(dev), fsync(fileno(dev)))
#define LEDC_CHANNEL        LEDC_CHANNEL_0
#define LEDC_TIMER          LEDC_TIMER_0
#define LEDC_MODE           LEDC_LOW_SPEED_MODE
#define LEDC_RESOLUTION     LEDC_TIMER_8_BIT // 8-bit resolution
#define LEDC_OUTPUT_IO      12          // GPIO number for the LED
#define LEDC_DUTY           256       // Duty cycle (100% of 256)
#define LEDC_FREQUENCY      100       // Frequency in Hz (100Hz)

static const char *TAG = "LEDC_Light_Sleep";

void configure_ledc()
{
    gpio_config_t io_conf = {};

    io_conf.intr_type = GPIO_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_INPUT_OUTPUT;
    //bit mask of the pins 
    io_conf.pin_bit_mask = (1ULL<<LEDC_OUTPUT_IO);
    //disable pull-down mode
    io_conf.pull_down_en = 1;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);

    // Configure the LEDC timer with APB_CLK as the clock source
    ledc_timer_config_t ledc_timer = {
        .speed_mode       = LEDC_MODE,
        .duty_resolution  = LEDC_TIMER_8_BIT, 
        .timer_num        = LEDC_TIMER,
        .freq_hz          = LEDC_FREQUENCY,
        .clk_cfg          = LEDC_USE_RC_FAST_CLK,  // Use LEDC_USE_RC_FAST_CLK to keep LEDC active during light sleep
    };
    ESP_ERROR_CHECK(ledc_timer_config(&ledc_timer));

    // Configure the LEDC channel
    ledc_channel_config_t ledc_channel = {
        .gpio_num       = LEDC_OUTPUT_IO,
        .speed_mode     = LEDC_MODE,
        .channel        = LEDC_CHANNEL,
        .timer_sel      = LEDC_TIMER,
        .duty           = LEDC_DUTY,
        .hpoint         = 0,
    };
    ESP_ERROR_CHECK(ledc_channel_config(&ledc_channel));

    // LEDC fade enable
    ledc_fade_func_install(0);

    ESP_ERROR_CHECK(esp_sleep_pd_config(LEDC_USE_RC_FAST_CLK, ESP_PD_OPTION_ON));
    ESP_ERROR_CHECK(gpio_sleep_sel_dis(LEDC_OUTPUT_IO));

    ESP_LOGI(TAG, "LEDC configured to stay on during light sleep with LEDC_USE_RC_FAST_CLK clock.");
}

void app_main()
{
    // Initialize LEDC with APB_CLK as clock source
    configure_ledc();

    // Optional: Set the LED brightness 
    ledc_set_fade_with_time(LEDC_MODE, LEDC_CHANNEL, LEDC_DUTY, 500);
    ledc_fade_start(LEDC_MODE, LEDC_CHANNEL, LEDC_FADE_WAIT_DONE);

    ESP_LOGI(TAG, "Hold in none sleep mode...");
    ESP_LOG_FLUSH(stdout);

    vTaskDelay(5000 / portTICK_PERIOD_MS);

    // Allow the ESP32C6 to enter light sleep while keeping LEDC running
    esp_sleep_enable_timer_wakeup(5000000);  // Wake up after 5 seconds for demonstration

    // Enter light sleep
    ESP_LOGI(TAG, "Entering light sleep mode...");
    ESP_LOG_FLUSH(stdout);

    esp_light_sleep_start();

    // After waking up
    ESP_LOGI(TAG, "Woke up from light sleep");
    ESP_LOG_FLUSH(stdout);

    // Off the light
    ledc_set_duty(LEDC_MODE, LEDC_CHANNEL, 0);
    ledc_update_duty(LEDC_MODE, LEDC_CHANNEL);    
}

``
#include <stdio.h>
#include <unistd.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/ledc.h"
#include "esp_sleep.h"
#include "esp_timer.h"
#include "esp_log.h"

#define ESP_LOG_FLUSH(dev) (fflush(dev), fsync(fileno(dev)))
#define LEDC_CHANNEL        LEDC_CHANNEL_0
#define LEDC_TIMER          LEDC_TIMER_0
#define LEDC_MODE           LEDC_LOW_SPEED_MODE
#define LEDC_RESOLUTION     LEDC_TIMER_8_BIT // 8-bit resolution
#define LEDC_OUTPUT_IO      12          // GPIO number for the LED
#define LEDC_DUTY           25       // Duty cycle (10% of 256)
#define LEDC_FREQUENCY      100       // Frequency in Hz (100Hz)

static const char *TAG = "LEDC_Light_Sleep";

void configure_ledc()
{
    gpio_config_t io_conf = {};

    io_conf.intr_type = GPIO_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_INPUT_OUTPUT;
    //bit mask of the pins 
    io_conf.pin_bit_mask = (1ULL<<LEDC_OUTPUT_IO);
    //disable pull-down mode
    io_conf.pull_down_en = 1;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);

    // Configure the LEDC timer with APB_CLK as the clock source
    ledc_timer_config_t ledc_timer = {
        .speed_mode       = LEDC_MODE,
        .duty_resolution  = LEDC_TIMER_8_BIT, 
        .timer_num        = LEDC_TIMER,
        .freq_hz          = LEDC_FREQUENCY,
        .clk_cfg          = LEDC_USE_RC_FAST_CLK,  // Use LEDC_USE_RC_FAST_CLK to keep LEDC active during light sleep
    };
    ESP_ERROR_CHECK(ledc_timer_config(&ledc_timer));

    // Configure the LEDC channel
    ledc_channel_config_t ledc_channel = {
        .gpio_num       = LEDC_OUTPUT_IO,
        .speed_mode     = LEDC_MODE,
        .channel        = LEDC_CHANNEL,
        .timer_sel      = LEDC_TIMER,
        .duty           = LEDC_DUTY,
        .hpoint         = 0,
    };
    ESP_ERROR_CHECK(ledc_channel_config(&ledc_channel));

    // LEDC fade enable
    ledc_fade_func_install(0);

    ESP_ERROR_CHECK(esp_sleep_pd_config(LEDC_USE_RC_FAST_CLK, ESP_PD_OPTION_ON));
    ESP_ERROR_CHECK(gpio_sleep_sel_dis(LEDC_OUTPUT_IO));

    ESP_LOGI(TAG, "LEDC configured to stay on during light sleep with LEDC_USE_RC_FAST_CLK clock.");
}

void app_main()
{
    // Initialize LEDC with APB_CLK as clock source
    configure_ledc();

    // Optional: Set the LED brightness 
    ledc_set_fade_with_time(LEDC_MODE, LEDC_CHANNEL, LEDC_DUTY, 500);
    ledc_fade_start(LEDC_MODE, LEDC_CHANNEL, LEDC_FADE_WAIT_DONE);

    ESP_LOGI(TAG, "Hold in none sleep mode...");
    ESP_LOG_FLUSH(stdout);

    vTaskDelay(5000 / portTICK_PERIOD_MS);

    // Allow the ESP32C6 to enter light sleep while keeping LEDC running
    esp_sleep_enable_timer_wakeup(5000000);  // Wake up after 5 seconds for demonstration

    // Enter light sleep
    ESP_LOGI(TAG, "Entering light sleep mode...");
    ESP_LOG_FLUSH(stdout);

    esp_light_sleep_start();

    // After waking up
    ESP_LOGI(TAG, "Woke up from light sleep");
    ESP_LOG_FLUSH(stdout);

    // Off the light
    ledc_set_duty(LEDC_MODE, LEDC_CHANNEL, 0);
    ledc_update_duty(LEDC_MODE, LEDC_CHANNEL);    
}
```
According to the documentation, both esp_sleep_pd_config(LEDC_USE_RC_FAST_CLK, ESP_PD_OPTION_ON) and ESP_ERROR_CHECK(gpio_sleep_sel_dis(LEDC_OUTPUT_IO)) should be working correctly. When I set LEDC_DUTY to 256 (fully on in LEDC_TIMER_8_BIT resolution) or 0 (fully off), it works perfectly. However, if I set LEDC_DUTY to 128 (half on), it is pulled up to 256 (fully on) when entering light sleep. Similarly, if I set LEDC_DUTY to 10, it is pulled down to zero (fully off) during light sleep.

The documentation states that gpio_sleep_sel_dis will "Disable SLP_SEL to change GPIO status automantically in lightsleep." So, I am wondering why this behavior occurs. Did I miss something? Is there a way to enable LEDC to maintain different duty cycles in light sleep, or is it limited to ON/OFF hold instead of true PWM?

I hope I can make it work, please help and thank you.

suda-morris, 2024-11-05:
Not sure if there's any regression in the v5.3, but recently we added a test case on the master branch for testing the ledc working under light sleep: https://github.com/espressif/esp-idf/blob/master/components/esp_driver_ledc/test_apps/ledc/main/test_ledc_sleep.c#L129

Could you please try that first?

josthlee, 2024-11-05:
Just downloaded the master for esp_driver_ledc test suite. Now I just have C6 and H2 with me, and I met the following errors when idf.py set-target esp32c6:

-- Configuring incomplete, errors occurred!
HINT: The component 'perfmon' could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager, the component has been removed and refactored into some other component or the component may not be supported by the selected target.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.

Cannot search out the component "perfmon". How to solve it or any workaround? Thank you!

songruo, 2024-11-05:
@josthlee Please refer to this post https://github.com/espressif/esp-idf/issues/13698#issuecomment-2119607575 (C6 requires additional workaround). On master, this feature has been implemented in the LEDC driver ([4a90deb](https://github.com/espressif/esp-idf/commit/4a90deb2278322e6c605484770caafebfbbc615f), so you don't need to take care of adding any workaround anymore. Instead, just set the flag `LEDC_SLEEP_MODE_KEEP_ALIVE` during LEDC channel configuration.

For your build error, please delete the `dependencies.lock` file in the folder, and try building the test app again.

josthlee, 2024-11-05:
> @josthlee Please refer to this post [#13698 (comment)](https://github.com/espressif/esp-idf/issues/13698#issuecomment-2119607575) (C6 requires additional workaround). On master, this feature has been implemented in the LEDC driver ([4a90deb](https://github.com/espressif/esp-idf/commit/4a90deb2278322e6c605484770caafebfbbc615f), so you don't need to take care of adding any workaround anymore. Instead, just set the flag `LEDC_SLEEP_MODE_KEEP_ALIVE` during LEDC channel configuration.
> 
> For your build error, please delete the `dependencies.lock` file in the folder, and try building the test app again.

Thanks for the bolt reply.
I just remove the ”perfmon“ from the makefile and fortunately now it comes out the bin.
Noted and I will try then and get back to you!

josthlee, 2024-11-05:
> @josthlee Please refer to this post [#13698 (comment)](https://github.com/espressif/esp-idf/issues/13698#issuecomment-2119607575) (C6 requires additional workaround). On master, this feature has [...]

Great LEDC_SLEEP_MODE_KEEP_ALIVE flag, it works perfectly.
Now it holds the duty after light sleeping, and I will test it more and get back to you!

> > For your build error, please delete the `dependencies.lock` file in the folder, and try building the test app again.
> 
> Thanks for the bolt reply. I just remove the ”perfmon“ from the makefile and fortunately now it comes out the bin. Noted and I will try then and get back to you!

josthlee, 2024-11-08:
I have tested it for a while and I think it is now fully satisfied with my work.
Thank you so much!

closed 2024-11-08 by josthlee

wangutian, 2024-12-25:
> Not sure if there's any regression in the v5.3, but recently we added a test case on the master branch for testing the ledc working under light sleep: [...]

您好，我使用的ESP-IDF版本为5.4，芯片为esp32c3，我配置sleep_mode为LEDC_SLEEP_MODE_KEEP_ALIVE时失败，需要打开下边这个宏定义，我应该怎么配置sdkconfig，还是说esp32c3不支持在轻睡眠的时候输出pwm波形？
`#if !SOC_LEDC_SUPPORT_SLEEP_RETENTION
    ESP_RETURN_ON_FALSE(ledc_conf->sleep_mode != LEDC_SLEEP_MODE_NO_ALIVE_ALLOW_PD, ESP_ERR_NOT_SUPPORTED, LEDC_TAG, "register back up is not supported");
#endif`

songruo, 2024-12-30:
@wangutian 

esp32c3 支持在轻睡眠的时候输出 pwm 波形，配置 sleep_mode 为 `LEDC_SLEEP_MODE_KEEP_ALIVE` 不应该失败，相关测试: https://github.com/espressif/esp-idf/blob/release/v5.4/components/esp_driver_ledc/test_apps/ledc/main/test_ledc_sleep.c#L135

esp32c3 不支持 `LEDC_SLEEP_MODE_NO_ALIVE_ALLOW_PD` 模式。