espressif/esp-idf #14830: Image distortion on a 128x32 SSD1306 OLED screen

henkjannl, 2024-11-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.3.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

MH-ET minikit

### Power Supply used.

USB

### What is the expected behavior?

Display images on my OLED display

### What is the actual behavior?

Images look distorted

### Steps to reproduce.

I have an I²C controlled SSD1306 OLED display with 128x32 pixels.
[image]

I tried to get the /peripherals/lcd/i2c_oled/main/i2c_oled_example_main.c example working, but images are distorted and screen updates were very slow. It appeared from the logfile that the I²C clock frequency was too high, and I could not manage to get the clock frequency down.  By switching back to the legacy I²C driver, I managed to get the clock frequency at 400.000 Hz as specified in the SSD1306 datasheet. I also tried lower frequencies to no avail.

I also switched back to a more basic example without LVGL: ../components/esp_lcd/test_apps/i2c_lcd/main/test_i2c_lcd_panel.c
 
When using the ` ` `esp_lcd_panel_draw_bitmap()` ` ` function, images still look distorted.

I modified the test program to clearify the type of distortions:

```c
#include <stdio.h>
#include "driver/i2c.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "esp_system.h"
#include "esp_lcd_io_i2c.h"
#include "esp_err.h"
#include "esp_log.h"

#define tag "main"

#define MN_LCD_H_RES          128
// #define MN_LCD_V_RES          32
[... 155 lines cut ...]
I (286) heap_init: At 4008DEA8 len 00012158 (72 KiB): IRAM
I (294) spi_flash: detected chip: generic
I (297) spi_flash: flash io: dio
W (301) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (312) main_task: Started on CPU0
I (322) main_task: Calling app_main()
I (492) main: i=0 byte=1 j=0
I (1502) main: i=0 byte=1 j=1
I (1652) main: i=0 byte=1 j=2
I (1802) main: i=0 byte=1 j=3
I (1952) main: i=0 byte=1 j=4
I (2102) main: i=0 byte=1 j=5
I (2252) main: i=0 byte=1 j=6
I (2402) main: i=0 byte=1 j=7
I (2552) main: i=1 byte=2 j=0
I (2702) main: i=1 byte=2 j=1
I (2852) main: i=1 byte=2 j=2
I (3002) main: i=1 byte=2 j=3
I (3152) main: i=1 byte=2 j=4
I (3302) main: i=1 byte=2 j=5
```

### More Information.

The hardware is working well when using the U8G2 driver in the Arduino programming environment, so I suspect something is wrong with the driver settings.

henkjannl, 2024-11-05:
Just for information:
* The repository of the project that uses this display: https://github.com/henkjannl/BedClock
* Process of porting from Arduino to ESP-IDF: https://github.com/henkjannl/BedClock/tree/master/04_Software/readme.md
* Code is in: https://github.com/henkjannl/BedClock/tree/master/04_Software/Bedclock_IDF_V11

suda-morris, 2024-11-06:
I remember SSD1306 of different resolution may need slightly different initialization: https://github.com/espressif/esp-idf/pull/12450

henkjannl, 2024-11-06:
Thanks!! It indeed has to do with the difference in initialisation between 128x64 and 128x32 displays.

I modified the code in this way:

```c
    esp_lcd_panel_ssd1306_config_t esp_lcd_panel_ssd1306_config = {
        .height = 32,
    };

    esp_lcd_panel_dev_config_t panel_config = {
        .bits_per_pixel = 1,
        .reset_gpio_num = -1,
        .vendor_config = &esp_lcd_panel_ssd1306_config,
    };
```
and now it is showing the 8x8 bitmaps correctly. This helped a lot!

closed 2024-11-06 by henkjannl