espressif/esp-idf #14833: Looking for help modifing registers for GPIO

ulao, 2024-11-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I had to use the register to change my pin state because of my timing requirement. I have most of it working but I can not get the input to release the state of the pin.

my goal is something like

set to output so I can hold pins high or low
set to input so I can read the start of an external pin.

I got the output working and my input is not, these are my defines. 

#define PINPULLUP(p)    asm volatile ("ee.setbitgpioout %0" : : "I"(p) : );
#define PIN_PULLDOWN(p)  __asm __volatile ("ee.clr_bit_gpio_out %0" : : "I"(p) : ); 
 

#define PIN_MODE_OUTPUT(p)  do { GPIO.enable_w1ts = (uint32_t)(1 << p); WRITE_PERI_REG((REG_IO_MUX_BASE + ( (p+1) * 4) ), (READ_PERI_REG(REG_IO_MUX_BASE + ( (p+1) * 4))|(1<<9)));  } while (0);

#define PIN_MODE_INPUT(p)   do { GPIO.enable_w1tc = (uint32_t)(1 << p); WRITE_PERI_REG((REG_IO_MUX_BASE + ( (p+1) * 4) ), (READ_PERI_REG(REG_IO_MUX_BASE + ( (p+1) * 4)) & ~(1<<9)));  } while (0);

to write a pin state I do this
PIN_MODE_OUTPUT(1<<1);
then I can use either PIN_PULLDOWN or PIN_PULLUP
This works

to read I do 
PIN_MODE_INPUT(1<<1);
but my wires stay low. and my external resister can not pull up.

ulao, 2024-11-05:
why was this missing?
/**
 * @brief Enable input for an IO
 *
 * @param gpio_num GPIO number
 *
 * @return
 *      - ESP_OK Success
 *      - ESP_ERR_INVALID_ARG GPIO number error
 */
esp_err_t gpio_input_enable(gpio_num_t gpio_num);
esp_err_t gpio_output_disable(gpio_num_t gpio_num);<--------------------

Anyways that works but I'd like to know if there is a simpler way to doable that? Especially since I can not use it on the master repo.

songruo, 2024-11-07:
Please check our Technical Reference Manual, `IO MUX and GPIO Matrix (GPIO, IO MUX)` chapter should have the steps stated clearly.

Take ESP32S3 TRM as an example:
[image]
[image]

ulao, 2024-11-07:
Hi could you post a link to the manual I'm confused where that is. 
[image]

but just for understanding the "n" I figured was an indication I replace that with a number. but in searching it seems the register name has the letter n in it. i.e GPIO_FUNCn_OUT_SEL_CFG_REG. Also the (x) is confusing still, is index (0x100) what I need for the GPIO? In other words if I want to set pin 1 I do 

GPIO_FUNCn_OUT_SEL_CFG_REG(0x100) ;
WRITE_PERI_REG( GPIO_OUT_REG[0], REG_READ(GPIO_OUT_REG[0] |= 0x02) );

songruo, 2024-11-08:
S2 TRM link: https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf

ESP-IDF Programming Guide is mainly for those uses our peripheral driver APIs. Since you want to work with registers directly, I would say TRM is your reference.

The 'n' and 'x' both correspond to the gpio number. You should first configure the pin as a simple input, and then also configure it as a simple output, following the guide stated in the TRM.

BTW, you sure you don't want to try our upper GPIO driver APIs, but play around with the registers? :)

ulao, 2024-11-08:
"BTW, you sure you don't want to try our upper GPIO driver APIs, but play around with the registers? :)"
-- my only requirement is speed.  I work with many device that require a transition in ttl around 50ns. If you think the API is fast enough and easier to work with I can try.

ulao, 2024-11-09:
so I as playing around with a few registers and I found the issue I had was my GPIO bundles. Apparently once you set those up, you can not use the ports as input. Also the register modification are not as precise as the bundle code, nor is it as fast.  I'm wondering if I can set up input bundles?

This is how I set them up

dedic_gpio_bundle_handle_t  createBundle( const int arr[], int size, bool makeBundle)
{ 
  // configure GPIO DEBUG
    gpio_config_t io_conf;

 
    for (int i = 0; i <size; i++) 
    {
        io_conf.intr_type = GPIO_INTR_DISABLE;
        io_conf.mode = GPIO_MODE_INPUT_OUTPUT;
        io_conf.pin_bit_mask = (1ULL<<i);
        io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
        io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
        gpio_config(&io_conf);
    }

    // Create bundleA, output only
    dedic_gpio_bundle_handle_t bundleA = NULL;
    dedic_gpio_bundle_config_t bundleA_config =
    {
        .gpio_array = arr,
        .array_size = size,
        .flags = {
        .out_en = 1,
        .in_en = 1,
        },
    };
    if (makeBundle)  dedic_gpio_new_bundle(&bundleA_config, &bundleA);
     return bundleA;
}

I thought that was both input and output?

songruo, 2024-11-11:
If you want to use Dedicated GPIO module, then it is a different story than using GPIO module to read/write the IOs. For configuring an IO to be used as a dedicated GPIO, you don't need to call `gpio_config`. Your call to `dedic_gpio_new_bundle` looks correct, it should be able to make IO1 both readable and writeable. Try reading the states with `dedic_gpio_bundle_read_in` first, see if you can get desired result.

ulao, 2024-11-12:
for a test, I change everything to open drain
io_conf.mode = GPIO_MODE_INPUT_OUTPUT_OD;
and everything works, expect in OD mode my transition for high to low is a max of 696 nS

But in normal GPIO_MODE_INPUT_OUTPUT my transition are much much faster, however I can not find a way to get my input wire conditions? If I do a read, it just shows what I see on the wires. So its working but I need to tell the GPIO to stop pulling high or low on the wires so my device can operate. Externally I use a 1k so that should be strong enough. but in this mode my device is not able to send data over the wire. 

to read I can use
  __asm__ __volatile__ ( "ee.get_gpio_in %[r]" : [r] "=r" (r));

but I still need a way to disable the external pulling, or at least stop what seems to be preventing my device from using the lines.

wanckl, 2024-11-13:
@ulao Hi, I think just adjust your external resister to 100~300 ohm and have a try!   

1k will work like `3v3/1k=3.3mA` which still weak compare to GPIO.

To allow more than one node output on one wire correctly, operate as `wire and`, you need config pin as OC not Push_Pull,     
just like IIS bus, with 4.7k pull up work with OC, can speed up to 400K

You set GPIO_MODE_INPUT_OUTPUT will config pin as Push_Pull, which let your device can't output.

ulao, 2024-11-17:
Hi thx, my external pullups are fixed ( they are part of another device ) I can not change them or change the values. 

but in my tests bitbaning the pins from the registers works just fine.  I can change back to GPIO_MODE_INPUT_OUTPUT but I still do not know what registers change the input output operations?

wanckl, 2024-11-18:
@ulao I'm confiused,   

what you mean `just fine in GPIO_MODE_INPUT_OUTPUT mode`, in this mode you can NOT read.

I mean decrase your external resister, so even if now you have a fixed internal pullup in other device, you can also parallio another resister do pullup, and change gpio mode to `OD`

ulao, 2024-11-18:
Sorry for any confusion will try t itemize here.

1) I can not decrees the external because my other devise will not work properly. But this works for avr and its a 30k.
2) in  GPIO_MODE_INPUT_OUTPUT I do not know how to change to input. This is my main issue. For AVR its simple and I know its an 8 bit mcu but life is so much simpler there..

output
DDRD |= 0x01;

to set pins 
PORTD |= 0x01; //high
PORTD |= 0x01; //low

input
DDRD &= ~0x01;
PORTD |= 0x01; //pull up. 30k

Everything works. I just change from output to input and I can bitbang. But with  the ESP, things are a bit more complicated. I get that the bundle code is needed to group my pairs, that is very understandable. I also now understand GPIO_MODE_INPUT_OUTPUT is the best mode. But what is still unclear, is how to change modes? It woudl seem by default mode is output and I can use 

#define PIN_PULLUP(p)    __asm__ __volatile__ ("ee.set_bit_gpio_out %0" : : "I"(p) : ); 
#define PIN_PULLDOWN(p)  __asm__ __volatile__ ("ee.clr_bit_gpio_out %0" : : "I"(p) : ); 

I have tested this and it works great.  but I'm not sure how to change from input to output.  I need to swtich relatively fast. So I can not delete my bundle and remake it.  I tried modifying the REG_IO_MUX_BASE  and READ_PERI_REG registers but that didnt work. I'm just very confused how the registers work in the ESP for GPIO. There must be a simple way to change the pin to inputs and use no pullups. I was able to do this on a stm32 platform once before but documentation was a bit more available. If that does not exist is there a way to change to OD mode without rebuilding the bundle?

ulao, 2025-01-04:
Wanted ti bump this and see if my last post was any clearer in what I'm hoping to do . Does my question at least make sense?

ginkgm, 2025-01-06:
Regarding your "really fast" requirement, not sure which is your real requirement, realtime performance (responding to an event, or the toggling speed). 

To eliminate the effect of insufficient current of pull-ups, and avoid  I suggest to try our driver on a hardware that does not have other devices connected, with OUTPUT mode instead of OD. Try to emulate a reasonable use case and see if the toggling speed meet your requirements. For example if you are running in a task without other interrupts, then you can just polling and toggling the GPIO in the task; if you rely on some interrupts, then you can respond to one GPIO interrupt to see how long it will take to respond (usually I'll assume the performance is around 100us max)

If you find the performance is much faster than your hardware, the problem may be caused by your OD+pullup. Consider modifying the hardware. Otherwise we can try other solutions, for example raw registers or dedicated GPIO.

ginkgm, 2025-01-06:
To realize it, you can try setting it up with https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#_CPPv411gpio_configPK13gpio_config_t , and use [gpio_set_level](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#_CPPv414gpio_set_level10gpio_num_t8uint32_t) / [gpio_get_level ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html#_CPPv414gpio_get_level10gpio_num_t) or their alternatives later.

- gpio_mode_t = GPIO_MODE_INPUT_OUTPUT_OD
- gpio_pullup_t = GPIO_PULLUP_DISABLE
- gpio_pulldown_t = GPIO_PULLDOWN_DISABLE

This will disable the pullup and pulldown, enable the input and output as OD. 

To read, gpio_set_level(1) first, then read. Please note that due to the possible delay caused by the pullup, you may need to delay (esp_rom_delay_us) until the external pullup to pull the line to a reasonable level.

To output, you can use gpio_set_level(0) or 1. 

If you want to make it faster, you can use the LL directly, for example GPIO LL for esp32: https://github.com/espressif/esp-idf/blob/master/components/hal/esp32/include/hal/gpio_ll.h#L515

I believe the normal GPIO is fast enough compared to the external pullup. Dedicated GPIO is more complicated and overkilling.

ulao, 2025-01-06:
My "really fast" requirement is just that I need to change output pin states quickly. The bundle code gave me that ability. In some cases I will wait for pin.1 to fall, and then I need to set the states of pins 2,3,4,5,6 etc. Much like a multiplexor. 

I'm currently at the stage where my external hardware is absent. My confusion is with code not hardware. I'm just trying to write some code logic that puts the esp32 into a output state, and change to input state. I can not use the built in functions because It takes a few hundred microseconds. That is too late. 

When I configure the bundle code for output I can do what I need as show above with the ee.set_bit_gpio_out regsiters. But to go to input mode,  I can not delete my bundle and remake it. I was hoping to use registers to do that. 

like with AVR I can do 

output
DDRD |= 0x01;

input
DDRD &= ~0x01;

So my current question or confusion, is how to change the pin state mode without rebuilding the bundle. 

**UPDATE**: 

I just saw the second reply, those look like function I need to call, can I get some of the over head removed by accessing registers? In input mode I do not need any pullups, the external pull up should work. I just need to read the condition of the wire. 

_To read, gpio_set_level(1) first, then read. Please note that due to the possible delay caused by the pullup, you may need to delay (esp_rom_delay_us) until the external pullup to pull the line to a reasonable level._
**yeah I get there maybe some delay here that is ok.**

so is gpio_set_level(1) out put and gpio_set_level(0) input? and this works with a bundle set up?

ulao, 2025-01-09:
I tried using the functions, but I'd rather just use registers. Is there a way to set w1ts.val  ?

I do not know what gpio_num wants,.. say for pin 3, is is 3, or 0x04? My IDE can not find the define. 
I tried both
gpio_set_level(0x04,1);
and
gpio_set_level(3,1);

neither of these changed the pin state. 

This is how I set up my bundle.
   for (int i = 0; i <size; i++) 
    {
        io_conf.intr_type = GPIO_INTR_DISABLE;
        io_conf.mode = GPIO_MODE_INPUT_OUTPUT_OD; 
        io_conf.pin_bit_mask = (1ULL<<i);
        io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
        io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
        gpio_config(&io_conf);
    }

ulao, 2025-01-17:
ah ok, no problem. This was much slower, I won;t be able to use the device that way. 

[image]

I think using register gets me closer to 200ns, but Hopeing to do better.  I will need to loop a condition of a wire state and then set the condition of another. I only have 300ns to do it in. I know the esp32 clock cycles are closer to 10- 12ns and I also know there will be over head because of other things running, so the faster the better. With bundles I got near that speed but can not figure out how to do the open drain.

songruo, 2025-01-27:
@ulao I guess I finally get your problem. 

Please try to configure the IO with dedicate GPIO driver first with both `in_en` and `out_en` flags set. When it is the time to read, call `gpio_ll_output_disable` first, then do the read. It should be able to disable the drive from the pin, so that read is possible. After you finish the read, call `gpio_ll_output_enable` to re-enable the output for the pin, so that you can write again.

`gpio_ll_output_disable` and `gpio_ll_output_enable` are low-level functions, which should be pretty fast to execute.

ulao, 2025-01-28:
Ok sounds good. Thx for helping!

Running in to some confusion I'm only able to output onj 1 pin, and its not the pin I'm write to. So before I try a read, I want to figure out the confusion.

how I set up my bundler
 const int bundleA_gpios[] = {0,1,2,3};
bundles[0] = createBundle( bundleA_gpios,sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]), true );

   dedic_gpio_bundle_handle_t bundles[8]; 
dedic_gpio_bundle_handle_t  createBundle( const int arr[], int size, bool makeBundle)
{ 
  // configure GPIO DEBUG
    gpio_config_t io_conf;

 
    for (int i = 0; i <size; i++) 
    {
        io_conf.intr_type = GPIO_INTR_DISABLE;
        //if (i==3) io_conf.mode = GPIO_MODE_INPUT_OUTPUT; else 
        io_conf.mode = GPIO_MODE_INPUT_OUTPUT; 
        io_conf.pin_bit_mask = (1ULL<<i);
        io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
        io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
        gpio_config(&io_conf);
    }

    // Create bundleA, output only
    dedic_gpio_bundle_handle_t bundleA = NULL;
    dedic_gpio_bundle_config_t bundleA_config =
    {
        .gpio_array = arr,
        .array_size = size,
        .flags = {
        .out_en = 1,
        .in_en = 1,
        },
    };
    if (makeBundle)  dedic_gpio_new_bundle(&bundleA_config, &bundleA);
     return bundleA;
}

to enable output on all 4 I use this
gpio_ll_output_enable (GPIO_HAL_GET_HW(0), GPIO_NUM_0 ); 
gpio_ll_output_enable (GPIO_HAL_GET_HW(0), GPIO_NUM_1 );
gpio_ll_output_enable (GPIO_HAL_GET_HW(0), GPIO_NUM_2 );
gpio_ll_output_enable (GPIO_HAL_GET_HW(0), GPIO_NUM_3 );

for a test I loop this ( I know the nameing is bad here, its just test code)
#define PIN_PULLUP(p)    __asm__ __volatile__ ("ee.set_bit_gpio_out %0" : : "I"(p) : ); 
#define PIN_PULLDOWN(p)  __asm__ __volatile__ ("ee.clr_bit_gpio_out %0" : : "I"(p) : ); 

PIN_PULLDOWN(1 << 0);_delay_us(100);
PIN_PULLUP(1 << 0);_delay_us(100);

PIN_PULLDOWN(1 << 1);_delay_us(200);
PIN_PULLUP(1 << 1);_delay_us(100);

PIN_PULLDOWN(1 << 2);_delay_us(300);
PIN_PULLUP(1 << 2);_delay_us(100);

PIN_PULLDOWN(1 << 3);_delay_us(400);
PIN_PULLUP(1 << 3);_delay_us(100);

Then on my scope the only pin that works is GPIO00 and it does down for 400us.

So 
1) Why is GPIO00 going down, and not GPIO03 ?
2) Why only the one pin?

I suspect something in my group is set wrong but I'm not seeing it.  I know what pins my scope is on, and its pin 0 that drops low for 400us, I'm sure of it, but unclear why.

songruo, 2025-01-31:
Please don't mix GPIO driver API with the dedicated GPIO driver API.

Configure the IOs with dedicated GPIO driver:
```
const int bundleA_gpios[] = {0, 1, 2, 3};
dedic_gpio_bundle_handle_t bundleA = NULL;
dedic_gpio_bundle_config_t bundleA_config = {
    .gpio_array = bundleA_gpios,
    .array_size = sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]),
    .flags = {
        .in_en = 1,
        .out_en = 1,
    },
};
dedic_gpio_new_bundle(&bundleA_config, &bundleA));
```
Output:
```
dedic_gpio_bundle_write(bundleA,0x0f,0x0f);_delay_ns(100)
dedic_gpio_bundle_write(bundleA,0x0f,0x00);_delay_ns(100)
```
Input:
```
gpio_ll_output_disable(&GPIO, 0);
gpio_ll_output_disable(&GPIO, 1);
gpio_ll_output_disable(&GPIO, 2);
gpio_ll_output_disable(&GPIO, 3);

dedic_gpio_bundle_read_in(bundleA);
...
// after you finish reading period, time to switch back to output
gpio_ll_output_enable(&GPIO, 0);
gpio_ll_output_enable(&GPIO, 1);
gpio_ll_output_enable(&GPIO, 2);
gpio_ll_output_enable(&GPIO, 3);
```

ulao, 2025-01-31:
ok all exmaples I saw on this matter used the gpio_config(), That maybe part of my issue then I will remove all of that code and try these exmaples .

ulao, 2025-02-07:
This worked fine, I just have two concerns left hope you can help. 

1) pin 3 does not work, 0,1,2 do. 3 is always low? 

bundle

const int a[] = {1,2,3,4};
     // Create bundleA, output only
    dedic_gpio_bundle_handle_t bundleA = NULL;
    dedic_gpio_bundle_config_t bundleA_config =
    {
        .gpio_array = a,
        .array_size =  sizeof(a) / sizeof(a[0]),
        .flags = 
        {
            .out_en = 1,
            .in_en = 1,
        },
    };
    // ESP_LOGW("start", "Start_LLAPI..");  try this?
    if (makeBundle)  dedic_gpio_new_bundle(&bundleA_config, &bundleA);

code
dedic_gpio_bundle_write(bundleHandle,0x0f,0x0f);

whether its out is enabled or not, it does not work.  in input mode its always low, even if I do the above. The other pins work fine. 

2) I guess my timer functions are crap.

#define _delay_ns(a) do {long long  i=0; i = a * 10; while (--i) {} }while (0);

#define _delay_us(a)  do {uint64_t  microseconds = esp_timer_get_time(); while (((uint64_t) esp_timer_get_time() - microseconds) <= a) {}} while (0);

Is there any way to get a more accurate time?

songruo, 2025-02-08:
Which chip are you using...? I tried the following code on S2 and S3, and checked with the logic analyzer, all four pins (IO0, IO1, IO2, IO3) work fine.

```
const int bundleA_gpios[] = {0, 1, 2, 3};
dedic_gpio_bundle_handle_t bundleA = NULL;
dedic_gpio_bundle_config_t bundleA_config = {
    .gpio_array = bundleA_gpios,
    .array_size = sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]),
    .flags = {
        .in_en = 1,
        .out_en = 1,
    },
};
dedic_gpio_new_bundle(&bundleA_config, &bundleA);

while (1) {
    dedic_gpio_bundle_write(bundleA,0x0f,0x0f);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    dedic_gpio_bundle_write(bundleA,0x0f,0x00);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
}
```

ulao, 2025-02-08:
s3 yeah. using 0,1,2,3

**1) test with mask F and pins f**
const int bundleA_gpios[] = {0,1,2,3};
dedic_gpio_bundle_handle_t bundleA = NULL;
dedic_gpio_bundle_config_t bundleA_config = {
    .gpio_array = bundleA_gpios,
    .array_size = sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]),
    .flags = {
        .in_en = 1,
        .out_en = 1,
    },
};
dedic_gpio_new_bundle(&bundleA_config, &bundleA);

while (1) {
    dedic_gpio_bundle_write(bundleA,0x0f,0x0f);
    vTaskDelay(100 / portTICK_PERIOD_MS);
    dedic_gpio_bundle_write(bundleA,0x0f,0x00);
    vTaskDelay(100 / portTICK_PERIOD_MS);
}

result

[image]

**2) test with mask F and pins 0x01**
const int bundleA_gpios[] = {0,1,2,3};
dedic_gpio_bundle_handle_t bundleA = NULL;
dedic_gpio_bundle_config_t bundleA_config = {
    .gpio_array = bundleA_gpios,
    .array_size = sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]),
    .flags = {
        .in_en = 1,
        .out_en = 1,
    },
};
dedic_gpio_new_bundle(&bundleA_config, &bundleA);

while (1) {
    dedic_gpio_bundle_write(bundleA,0x0f,0x0f);
    vTaskDelay(100 / portTICK_PERIOD_MS);
    dedic_gpio_bundle_write(bundleA,0x0f,0x00);
    vTaskDelay(100 / portTICK_PERIOD_MS);
}

result

[image]

same result  with **2) test with mask F and pins 0x02**

same result  with **2) test with mask F and pins 0x04**

**2) test with mask F and pins 0x08**
const int bundleA_gpios[] = {0,1,2,3};
dedic_gpio_bundle_handle_t bundleA = NULL;
dedic_gpio_bundle_config_t bundleA_config = {
    .gpio_array = bundleA_gpios,
    .array_size = sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]),
    .flags = {
        .in_en = 1,
        .out_en = 1,
    },
};
dedic_gpio_new_bundle(&bundleA_config, &bundleA);

while (1) {
    dedic_gpio_bundle_write(bundleA,0x0f,0x0f);
    vTaskDelay(100 / portTICK_PERIOD_MS);
    dedic_gpio_bundle_write(bundleA,0x0f,0x00);
    vTaskDelay(100 / portTICK_PERIOD_MS);
}

result

[image]

as expected
same result  with **2) test with mask F and pins 0x0F**

[image]

songruo, 2025-02-10:
I could not reproduce. The dev board pins were not connect to any external device, right? Are you using the default sdkconfig, or is there anything special you selected in the menuconfig?

ulao, 2025-02-10:
I'm not sure how to check these thing, will add as I learn how. 

External device, no its just the board. 

I think the menu config is a viewer for the sdkconfig ?

sdkconfig's are in every path so I figured may its the one at the root of my project? Its huge but will try to post it.

Woudl it be easier to somehow find a defeult one or will that break my comport speeds's and opther settings. 

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.4.0 Project Configuration
#
CONFIG_SOC_MPU_MIN_REGION_SIZE=0x20000000
CONFIG_SOC_MPU_REGIONS_MAX_NUM=8
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_PHY_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_AHB_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_LCDCAM_I80_LCD_SUPPORTED=y
CONFIG_SOC_LCDCAM_RGB_LCD_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_MPU_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
[... 1311 lines cut ...]
CONFIG_IPC_TASK_STACK_SIZE=1280
CONFIG_TIMER_TASK_STACK_SIZE=3584
CONFIG_TIMER_TASK_PRIORITY=1
CONFIG_TIMER_TASK_STACK_DEPTH=2048
CONFIG_TIMER_QUEUE_LENGTH=10
# CONFIG_ENABLE_STATIC_TASK_CLEAN_UP_HOOK is not set
# CONFIG_HAL_ASSERTION_SILIENT is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# End of deprecated options

songruo, 2025-02-11:
Still not reproducible on my end. What else you can try is:

- Try to run our dedicated gpio test cases (https://github.com/espressif/esp-idf/blob/master/components/esp_driver_gpio/test_apps/gpio_extensions/main/test_dedicated_gpio.c), see if they can pass, to confirm your hardware is fine.
- In the code I showed, after writing to the pins with `dedic_gpio_bundle_write`, call `dedic_gpio_bundle_read_in` to read the IO status, see if the write_out value matches with the read_in value.

ulao, 2025-02-11:
I tried the write and read, is there somethign wrong with how I did this?

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40377bda  PS      : 0x00060830  A0      : 0x82004fa1  A1      : 0x3fc979b0       
--- 0x40377bda: dedic_gpio_bundle_read_in at C:/Users/Administrator/Desktop/esp/esp-idf/v5.4/esp-idf/components/esp_driver_gpio/src/dedic_gpio.c:409

A2      : 0x00000000  A3      : 0x3c021dc1  A4      : 0x3c023966  A5      : 0x0000013d       
A6      : 0x3c021dc1  A7      : 0x0000000f  A8      : 0x00000000  A9      : 0x3fc97990       
A10     : 0x0000000a  A11     : 0x3c021dc1  A12     : 0x3c023966  A13     : 0x3fc979b0       
A14     : 0x3fc97990  A15     : 0x0000000c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c       
EXCVADDR: 0x00000008  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff       
--- 0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

--------------------------------------------------------------------------------------------

ESP_LOGW("first", "%lu", dedic_gpio_bundle_read_in(bundleHandle ));

dedic_gpio_bundle_write(bundleHandle,0x0f,0x04);_delay_us(5);
ESP_LOGW("writing 4", "%lu", dedic_gpio_bundle_read_in(bundleHandle));

dedic_gpio_bundle_write(bundleHandle,0x0f,0x00);_delay_us(5);

ESP_LOGW("clearing 4", "%lu", dedic_gpio_bundle_read_in(bundleHandle));

ulao, 2025-02-11:
I think I just found my issue.

[image]

songruo, 2025-02-12:
Hmm, additional to what I said in https://github.com/espressif/esp-idf/issues/14833#issuecomment-2626947868, after the configuration of the IOs with dedicated GPIO driver, please add the following code, and try again.

```
GPIO.func_out_sel_cfg[0].oen_sel = 1;
GPIO.func_out_sel_cfg[1].oen_sel = 1;
GPIO.func_out_sel_cfg[2].oen_sel = 1;
GPIO.func_out_sel_cfg[3].oen_sel = 1;
```

ulao, 2025-02-12:
guessing you meana for use of input as that is the issue im having, let me try real fast here.

ulao, 2025-02-12:
yeah I think that did it?
I used
GPIO.func_out_sel_cfg[x].oen_sel = 1;  for input
GPIO.func_out_sel_cfg[x].oen_sel = 0; for out put

whjat did that do?

songruo, 2025-02-13:
`GPIO.func_out_sel_cfg[x].oen_sel` is used to control the source of output enable signal for the GPIO pin. Set to 1 means to force the output enable signal to be sourced from bit n of GPIO_ENABLE_REG (i.e. gpio_ll_output_disable would take affect); Set to 0 means to use output enable signal from peripheral (i.e. the dedicated GPIO signal controls whether output or not, and since the dedicated GPIO signal is configured to output, so it will drive the IO).

ulao, 2025-02-14:
ok, thx for the explination. All is working, 

I know this is off topic but one other issue I'm having is acurate delay timing. 

i.e #define _delay_us(a)  do {uint64_t  microseconds = esp_timer_get_time(); while (((uint64_t) esp_timer_get_time() - microseconds) < a) {}} while (0);

I only get 3 us resolution here. I'd really like to do ;
_delay_us(1)
_delay_us(100)
_delay_us(1.5)
_delay_us(.1)
etc
Is there anythign on the ESP that can do that?

songruo, 2025-02-17:
[esp_rom_delay_us()](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp_rom_sys.h#L89) is better than what you used, since it counts the cpu ticks internally. However, the best resolution is 1us. For any smaller time ticks, you may need to use 'nop'.

closed 2025-02-17 by songruo

ulao, 2025-04-06:
@songruo 
Just resumed this project and all is working as discussed but I saw the timeing is not as expected, is this the best I'm going to get?

I made some test code

 PIN_PULLUP(LLAPI_DATA);
_delay_us(2);
 dedic_gpio_bundle_write(bundleHandle,bundle_Mask, 1); dedic_gpio_bundle_write(bundleHandle,bundle_Mask, ~1);
_delay_us(2);
 dedic_gpio_bundle_write(bundleHandle,bundle_Mask, ~1); ASM("NOP");  dedic_gpio_bundle_write(bundleHandle,bundle_Mask, ~1);

and saw this.

[image]

this means its 136 ns for the commands to change the pin state. and the nop adds 8. Is that the best I can hope for?  I could bypass one fucntion and call the dedic_gpio_cpu function but can I was hoping to get around 14ns by hitting a register.

songruo, 2025-04-07:
`dedic_gpio_bundle_write` and `dedic_gpio_bundle_read_in` contains quite a few operations. If you want to further reduce the timing, yo have to dive into these two APIs, and call the dedicated GPIO assembly code by yourself.