espressif/esp-idf #14838: Change the include directories for linked components to use SYSTEM includes rather than normal ones.

BIGduzy, 2024-11-06:
### Is your feature request related to a problem?

When using a language server such as clangd, the language server adds inline warning/error diagnostics to the code in your editor.
These language servers usually parse your compile_commands.json to figure out which files belong to your projects and which compiler flags you've used.
If a macro is imported by a header that is not included using `-isystem` rather than `-I` warnings stemming from the expanded macro will be shown in the file where the macro is used.

An example is that when you use clang-tidy, all ESP_LOG* macros will be highlighted with a warning for clang-tidy's `avoid-do-while` rule:
[image]

As an example, I want the rule enabled because our internal code standards require it, but I don't have to change vendor code so the warning is just useless/added noise.

### Describe the solution you'd like.

I would appreciate it if esp-components (such as esp_log) would export their include directories as SYSTEM so that I don't get warnings from code I can't change in my own code.

### Describe alternatives you've considered.

I've attempted to use CMake scripting to edit all include directories on the `main` target that starts with `/opt/esp` (my installation directory) so that they are removed from my target and re-added as a system include but couldn't get it to work.

igrr, 2024-11-06:
@BIGduzy As an alternative solution, could you try excluding IDF root directory from analysis in clang-tidy?

> If a macro is imported by a header that is not included using -isystem rather than -I warnings stemming from the expanded macro will be shown in the file where the macro is used.

I'm afraid that in some cases this can mask genuine errors.

Even in your case, technically you do get these do/while loops injected into your application code via macros, so the warning is probably justified? If your coding standard doesn't allow do/while loops you probably should create a wrapper function and call it instead of the macros IDF provides, then disable this check just for the file which defines the wrapper function.

BIGduzy, 2024-11-07:
Thank you for your reply @igrr. It seems we require a new clang version to exclude directories from clang-tidy. Which for now is not prioritized. 

Wrapper functions would be a better solution perhaps, but it would require us to refactor the code we already have. For now I'll close the issue. 
Thank you.

closed 2024-11-07 by BIGduzy