espressif/esp-idf #14839: esp_https_ota command ignores http status code 304 (Not Modified)

nozols, 2024-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The `esp_https_ota` component tries to download the binary when it receives a `304` HTTP status code. I think it should stop the OTA progress, because this status code should indicate that the firmware is already up to date.

According to [MDN](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/304):
> The HTTP 304 Not Modified redirection response status code indicates that there is no need to retransmit the requested resources.

Maybe the same should be done for status code 204, which according to [MDN](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/204) means:
> The HTTP 204 No Content successful reponse status code indicates that a request has succeeded, but the client doesn't need to navigate away from its current page.

Returning 404 will stop the OTA in the current esp-idf, but I don't think that is the correct status code to use, as the URL is valid. I would only expect a 404 if the URL doesn't point to a valid location.

Is this something that can be implemented in ESP-IDF?

nileshkale123, 2024-11-11:
Hello @nozols,

We’ve raised a merge request to handle the 304 (NOT_MODIFIED) response code, and it will be merged soon.

Typically, a 204 (NO_CONTENT) response is returned by the server in DELETE or POST requests, so we haven’t included handling for it in the OTA process. However, if you have a specific reproducible case where a 204 response is encountered during OTA, please let us know. We’ll be happy to add support for it if needed.

Thank you!

nozols, 2024-11-11:
Thanks! If status code 304 is supported we don't need 204. Thanks a lot for implementing this.

closed 2024-12-05 by espressif-bot