espressif/esp-idf #14840: OS.h Defines for min and max break c++ compilation

kenozz-igs, 2024-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1-638-ga0f798cfc4

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

C++ build successful when nimble is enabled.  

### What is the actual behavior?

Cannot build project due to defines for min and max.  Release/5.2 builds OK, Release/5.3 does not. Defines for min and max in os.h in 5.3 are the issue.  5.2 had a commit 155931f to remove them but looks to have been re-introducted in 5.3 commit fd31133.

5.2 - https://github.com/espressif/esp-nimble/blob/e08302d66d77a29e65dffe2c336e4dcda4e80e51/porting/nimble/include/os/os.h

5.3 - https://github.com/espressif/esp-nimble/blob/32b2124f54238e70ecd4c72a753698babdd939f7/porting/nimble/include/os/os.h

### Steps to reproduce.

Bulid C++ project with Nimble and compiler error is observed. 

### More Information.

Looks to be same issue as [in ticket](https://github.com/espressif/esp-idf/issues/11634)

ESPAbhinav, 2024-11-07:
Hi @kenozz-igs ,
I have removed the header definition from the OS.h file, just like in v5.2. The following is the patch file for changes. Can you please confirm if the patch provided works for you? 
[remove_min_max_defs.zip](https://github.com/user-attachments/files/17657888/remove_min_max_defs.zip)

kenozz-igs, 2024-11-07:
Hi @ESPAbhinav thanks for the quick response.  I can confirm the patch does resolve the build issue.  Is there any plans to merge into the 5.3 release?

ESPAbhinav, 2024-11-07:
Hi @kenozz-igs , already raised the MR to merge this change into release 5.3 .Thanks for identifying the issue.

closed 2024-11-19 by espressif-bot