espressif/esp-idf #14841: ability to read GPIO holder status

bryghtlabs-richard, 2024-11-06:
### Is your feature request related to a problem?

I need the ability to read the GPIO holder status per pin, so that I can discover the state of different peripherals after a restart.

### Describe the solution you'd like.

I'd like a nice ESP-IDF GPIO interface to take a GPIO number and receive a boolean value if the pin-holder was enabled or disabled.

bryghtlabs-richard, 2024-11-06:
There is a related function, `gpio_ll_is_digital_io_hold()`, but nothing at the gpio.h level.

songruo, 2024-11-07:
Hi @bryghtlabs-richard,

From your PR description, I get your scenario is you need to hold the SPI LCD CS line during a restart (core reset), is that correct? Though you have held the CS pad before restart, after the system restarts, all the signal connection information will still get lost, so you will still have to re-configure the pin (assuming you are using `esp_lcd_new_panel_io_spi()`), and then eventually release the hold.

Is there any reason why you don't want to call `esp_lcd_new_panel_io_spi()` after the restart?

bryghtlabs-richard, 2024-11-07:
Hi @songruo ,

It's close but not exactly what you described. I'm using `esp_lcd_new_panel_io_spi()` for every startup.

If the CS/RST pins were held with pin-holders, then I know an earlier software run has already configured the display, so I need to avoid calling `esp_lcd_panel_reset()` and avoid initial startup register scripts to prevent displaying corruption to user.

If the CS/RST were not held with pin-holders, then the display needs to be fully booted, with `esp_lcd_panel_reset()` and initial startup register scripts and delays.

songruo, 2024-11-08:
@bryghtlabs-richard Do you think checking the chip reset reason with [esp_rom_get_reset_reason()](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp_rom_sys.h#L91) can also achieve the purpose?

bryghtlabs-richard, 2024-11-18:
@songruo , no, because the chip may have been reset before or after the display driver IC startup-up.

bryghtlabs-richard, 2024-11-19:
Sorry for the late response, I was out of the office for a bit, but I'm back now.

bryghtlabs-richard, 2025-03-04:
>Though you have held the CS pad before restart, after the system restarts, all the signal connection information will still get lost, so you will still have to re-configure the pin (assuming you are using esp_lcd_new_panel_io_spi()), and then eventually release the hold.

To explain about this part a little further: my LCD had internal memory that will keep the image displayed, as long as the ESP32 does not reset the LCD controller or send a garbage packet due to chip select signal changing during reset. As long as I can hold those two pins, and tell that they have been held, then I can re-configure the SPI bus and associated pins, then release the holds, and resume communications with LCD controller.

`esp_rom_get_reset_reason()` cannot tell a crash with display bus holds apart from a crash before display bus holds were enabled.