espressif/esp-idf #14842: bad link in issue template

bryghtlabs-richard, 2024-11-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected to read the documentation.

### What is the actual behavior?

Issue templated included:

> Use this format to link issue numbers: Fixes #123 - https://docs.github.com/en/free-pro-team@latest/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue-using-a-keyword

But the link was unavailable:

[image]

### Steps to reproduce.

1. Signed in to a standard(non-pro) github account
2. Tried to create an ESP-IDF issue and follow request
3. got HTTP404

Alvin1Zhang, 2024-11-29:
@bryghtlabs-richard Thanks for reporting, I checked all the templates and followed the steps you shared, but failed to reproduce the issue you mentioned, will close due to can't reproduce, feel free to reopen with more details, thanks.

closed 2024-11-29 by Alvin1Zhang

bryghtlabs-richard, 2024-12-02:
Thanks for checking in - they're all working for me too now 👍

bryghtlabs-richard, 2024-12-02:
@Alvin1Zhang , sorry, I found another - the following text with the bad link is also in the pull-request pre-filled form:

```
<!--
- Use this format to link issue numbers: Fixes #123 - https://docs.github.com/en/free-pro-team@latest/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue-using-a-keyword
- Mention any other PRs related to this one.
- If there is related documentation, add the link here.
- If there is a public chat where changes in this PR were initiated, you can include the link here.
-->
```

This looks like a real URL, just one I cannot see.

Alvin1Zhang, 2024-12-03:
@bryghtlabs-richard Thanks for sharing the updates, will check and get back to you, thanks.