espressif/esp-idf #14846: 'f_chdir' can not work well

hibfh, 2024-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I need to use the api "f_chdir" to change the directory In my FTP server part.
I mount the emmc to fatfs vfs already
,when I try f_chdir(MOUNT_POINT),It do works,but in trouble with f_chdir(MOUNT_POINT/subdir).It return 5 which means path invalid，and i'm sure it exists.
And I change the config below
#define FF_FS_RPATH		2
#define FF_STR_VOLUME_ID	2
#define FF_VOLUME_STRS		"MOUNT_POINT1","MOUNT_POINT2"
Need some helps ，thx

nopnop2002, 2024-11-08:
>when I try f_chdir(MOUNT_POINT),It do works

Have you checked the return value of f_chdir?

adokitkat, 2024-11-17:
Hi @hibfh , do you still experience your issue?

hibfh, 2024-11-20:
@adokitkat yep, I have no time to solve it ,so I change the method and not use the 'f_chdir' anymore.

RathiSonika, 2024-11-25:
Hello @hibfh, With the configuration you mentioned above, the f_chdir function is working correctly and is tested with following piece of code. Let me know if you have any questions.

```
void app_main(void) {
    const char *base_path = "";
    ESP_LOGI(TAG, "Mounting FAT filesystem");

    const esp_vfs_fat_mount_config_t mount_config = {
            .max_files = 4,
            .format_if_mount_failed = true,
            .allocation_unit_size = CONFIG_WL_SECTOR_SIZE,
            .use_one_fat = false,
    };

    wl_handle_t wl_handle = WL_INVALID_HANDLE;
    esp_err_t err = ESP_OK;

    err = esp_vfs_fat_spiflash_mount_rw_wl(base_path, "storage", &mount_config, &wl_handle);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to mount FATFS (%s)", esp_err_to_name(err));
        return;
    }

    char cwd[128];
    f_getcwd(cwd, sizeof(cwd));
    printf("Current working directory: %s\n", cwd);

    const char* new_dir = "subdir";
    rmdir(new_dir);
    ESP_LOGI(TAG, "Creating a new directory");
    if (mkdir(new_dir, 0777) < 0) {
        ESP_LOGE(TAG, "Failed to create a new directory: %s", strerror(errno));
        return;
    }

    // Change to a subdirectory
    FRESULT res = f_chdir(new_dir);

    if (res == FR_OK) {
        printf("Successfully changed to directory: %s\n", new_dir);

        // Verify the current directory
        if (f_getcwd(cwd, sizeof(cwd)) == FR_OK) {
            printf("Current working directory: %s\n", cwd);
        } else {
            printf("Failed to get current working directory\n");
        }
    } else {
        printf("Failed to change directory: %s (Error: %d)\n", new_dir, res);
    }

    ESP_LOGI(TAG, "Unmounting FAT filesystem");
    ESP_ERROR_CHECK(esp_vfs_fat_spiflash_unmount_rw_wl(base_path, wl_handle));
}
```

Following is the output log
```I (309) example: Mounting FAT filesystem
Current working directory: /MOUNT_POINT1/
I (489) example: Creating a new directory
Successfully changed to directory: /MOUNT_POINT1/subdir
Current working directory: /MOUNT_POINT1/subdir
I (769) example: Unmounting FAT filesystem
I (859) main_task: Returned from app_main()
```

hibfh, 2024-11-27:
@RathiSonika I am truly grateful for your help。I‘ll try it later

closed 2024-11-27 by hibfh