espressif/esp-idf #14848: Error when reading from NVS after flash encryption and NVS encryption are enabled

mejha, 2024-11-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello.

We are using ESP32-C3 devices that need to be configured before production use.
We have enabled SecureBoot, flash encryption and NVS encryption.
We are using the following partition to store sensitive data that needs to be protected:
```
key,type,encoding,value
DPS,namespace,,<-- First entry should be of type "namespace"
P1,data,string,11111
P2,data,string,22222
P3,data,string,33333
```

Our partition table is as follows:
```
nvs,data,nvs,0xd000,16K,
otadata,data,ota,0x11000,8K,
ota_0,app,ota_0,0x20000,1536K,
ota_1,app,ota_1,0x1a0000,1536K,
nvs_keys,data,nvs_keys,0x320000,16K,
```

The command we are using to generate the partition is:
```
nvs_partition_gen.py generate esp32c3_nvs.csv esp32c3_nvs.bin 0x4000
```

Above generated *esp32c3_nvs.bin* is then flashed at address 0xd000.

After flashing everything (bootloader, partition table, app, etc.), we are encountering an issue when using the `nvs_get_str` function; it returns error 4354.
However, if we use `nvs_set_str` and then call `nvs_get_str`, it returns without an error. So, calling `nvs_get_str` alone results in an error, but after using `nvs_set_str`, `nvs_get_str` works as expected.

Is there a step we may have missed?

Thank you in advance.
Best regards,
Miha

rrtandler, 2024-11-08:
Hi @mejha ,

Try to put the string values in .csv into double quotes.
`P1,data,string,"11111"
P2,data,string,"22222"
P3,data,string,"33333"`

mejha, 2024-11-08:
Hello @rrtandler,
Just to note: when encryption options are not enabled, everything works fine, meaning that putting string values in quotes will not solve the issue.

rrtandler, 2024-11-08:
Hello @mejha,
I see, so the missing data is not subject of the binary generation. Then the potential data loss may happen during the data encryption. 
What is the sequence of steps on host and on the target you perform in order to encrypt the NVS data ?
You can check it and compare it with IDF example in `examples/security/flash_encryption`.

mejha, 2024-11-10:
We would like to use **_"Automatic generation of NVS encryption keys"_** that is mentioned in the example `examples/security/flash_encryption`, meaning that keys are generated internally by the esp32.
Example only shows how to flash empty partition. In our case partition is not empty, P1, P2, P3 have some values set that are extremely important for our application.

Our sequence of step are like so:
- setting menuconfig (NVS encryption, flash encryption, enabing SecureBoot, ...)
- generating NVS partition that contains predefined values (in our case: P1, P2, P3) with `nvs_partition_gen.py`
- building bootloader (`idf.py bootloader`) and application (`idf.py build`)
- generating SecureBoot private keys with `espsecure.py`
- appending generated private keys to the bootloader with `espsecure.py`
- application singing with one of the generated private keys
- flashing bootloader
- flashing application -> here all .bins are flashed (application, nvs, partition table, ota_data)

mahavirj, 2024-11-11:
@mejha 

Relevant: https://github.com/espressif/esp-idf/issues/14551

NVS encryption scheme do not support encryption of the pre-flashed NVS contents on the device. If you would like to pre-flash NVS data then the partition must be encrypted on the host side and that in turn means the the NVS keys must also be generated on the host side. 

Please refer to following documentation guides:

- [NVS Partition Generator](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html#generate-encryption-keys-partition)
- [Mass Manufacturing Tool](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/mass_mfg.html)
- [NVS Encryption Workflow Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/security-features-enablement-workflows.html#enable-nvs-encryption-externally)

mejha, 2024-11-11:
@mahavirj thank you for provided information.
It would be a nice feature if this could be a setting in menuconfig to encrypt preflashed NVS.

closed 2024-11-11 by mejha