espressif/esp-idf #14849: Missing gethostname function

DBMTC, 2024-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The function gethostname is not defined anywhere, exactly as reported here: https://github.com/espressif/esp-idf/issues/1102 except in newer releases the lwip_tcp_ip_adapter component has been removed and the provided solution cannot be used anymore (nor should be in older releases as that component has been deprecated a long time ago).

atanisoft, 2024-11-08:
esp_netif_get_hostname could replace the older approach for retrieving the hostname.

DBMTC, 2024-11-11:
> esp_netif_get_hostname could replace the older approach for retrieving the hostname.

Okay, I will get that tested as soon as possible. Is it possible to have the standard function call this function? As it is a bit weir having to do this in a library with code shared among many microcotrollers.

atanisoft, 2024-11-11:
> Is it possible to have the standard function call this function?

Possibly, that may be a good use for this ticket (implement a default version of the API).

In the projects I maintain I've included a standard version of this function in common code shared across a few projects. But it would be nice to have it included by default (same would go for `gai_strerror` which wouldn't be too much to include either)

igrr, 2024-11-11:
Linking https://github.com/espressif/esp-protocols/pull/671 for gai_strerror

david-cermak, 2024-12-09:
We will support these two functions (`gethostname()` and `gai_strerror()`) via an external component called [sock_utils](https://components.espressif.com/components/espressif/sock_utils), which provides a very simplified implementation of some common socket/network/system related functionality. The idea is to provide "some" functionality first (not necessarily 100% correct, but good enough) to simplify porting efforts.

For instance `gai_strerror()` just returns a textual representation of the error number and `gethostname()` uses a preferred network interface hostname (unlike standard POSIX `gethostname()`, which returns a system wide, global hostname)

PS: Adding `gethostname()` in https://github.com/espressif/esp-protocols/pull/707

closed 2024-12-20 by espressif-bot