espressif/esp-idf #14850: ESP32-P4. Problems with MIPI DSI

Maxim88S, 2024-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-P4  v 1.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-P4  v 1.4 

### Power Supply used.

USB

### What is the expected behavior?

Good day,
I wanted to use the example code from examples/peripherals/lcd/mipi_dsi  to connect the display from the board kit ESP32-P4.
I expected the display to start without any code changes.

### What is the actual behavior?

 I encountered errors in serial console.
```
I (1584) main_task: Started on CPU0
I (1604) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1604) main_task: Calling app_main()
I (1604) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1614) example: MIPI DSI PHY Powered on
I (1624) example: Install MIPI DSI LCD control panel
E (6604) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6604) task_wdt:  - IDLE0 (CPU 0)
E (6604) task_wdt: Tasks currently running:
E (6604) task_wdt: CPU 0: main
E (6604) task_wdt: CPU 1: IDLE1
E (6604) task_wdt: Print CPU 0 (current core) backtrace
Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x40047d80  RA      : 0x40047d7a  SP      : 0x4ff279d0  GP      : 0x4ff10a80
TP      : 0x4ff27b70  T0      : 0x4fc19b4c  T1      : 0x48000c54  T2      : 0x00000000
S0/FP   : 0x00000001  S1      : 0x48000c58  A0      : 0x48000c58  A1      : 0x00000000
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x500a0000  A5      : 0x00001801
A6      : 0x00000000  A7      : 0x00000008  S2      : 0x4ff27a4b  S3      : 0x00000000
S4      : 0x00000006  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000004  T4      : 0x00000003  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00000001  MTVEC   : 0x00000000  MCAUSE  : 0x4ff27ae8  MTVAL   : 0x48000c70
MHARTID : 0x00000000
E (11604) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (11604) task_wdt:  - IDLE0 (CPU 0)
[... 160 lines cut ...]
E (21604) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (21604) task_wdt:  - IDLE0 (CPU 0)
E (21604) task_wdt: Tasks currently running:
E (21604) task_wdt: CPU 0: main
E (21604) task_wdt: CPU 1: IDLE1
E (21604) task_wdt: Print CPU 0 (current core) backtrace
Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x40047d80  RA      : 0x40047d7a  SP      : 0x4ff279d0  GP      : 0x4ff10a80
TP      : 0x4ff27b70  T0      : 0x4fc19b4c  T1      : 0x48000c54  T2      : 0x00000000
S0/FP   : 0x00000001  S1      : 0x48000c58  A0      : 0x48000c58  A1      : 0x00000000
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x500a0000  A5      : 0x00001801
A6      : 0x00000000  A7      : 0x00000008  S2      : 0x4ff27a4b  S3      : 0x00000000
S4      : 0x00000006  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000004  T4      : 0x00000003  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00000001  MTVEC   : 0x00000000  MCAUSE  : 0x4ff27ae8  MTVAL   : 0x48000c70
MHARTID : 0x00000000
E (26604) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (26604) task_wdt:  - IDLE0 (CPU 0)
E (26604) task_wdt: Tasks currently running:
E (26604) task_wdt: CPU 0: main
E (26604) task_wdt: CPU 1: IDLE1
E (26604) task_wdt: Print CPU 0 (current core) backtrace
```

suda-morris, 2024-11-08:
Hi @Maxim88S 

If you're working with release v5.3.1, the example there is using a different MIPI LCD from the one inside the ESP32-P4-Function-EV-Board. In v5.3.2 (commin soon) this example will support the P4-Function-EV-Board out of the box. 

For now, can you try the example from the master branch or from the release/v5.3 branch?

Maxim88S, 2024-11-08:
Thanks for the advice.
The master version launched without this error.
Now the output on the COM port is like this
```
I (1585) main_task: Started on CPU0
I (1605) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1605) main_task: Calling app_main()
I (1605) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1615) example: MIPI DSI PHY Powered on
I (1625) example: Install MIPI DSI LCD control IO
I (1625) example: Install MIPI DSI LCD data panel
I (1635) ek79007: version: 1.0.0
I (1795) example: Initialize LVGL library
I (1795) example: Allocate separate LVGL draw buffers
I (1795) example: Register DPI panel event callback for LVGL flush ready notification
I (1805) example: Use esp_timer as LVGL tick timer
I (1805) example: Create LVGL task
I (1815) example: Starting LVGL task
I (1875) example: Display LVGL Meter Widget
I (2025) main_task: Returned from app_main()

```
But at the same time, only the backlight is on, and the display itself is black. Should something be displayed on the screen when running this example and if not, can you tell me how to modify the code to display, for example, an image.

suda-morris, 2024-11-11:
@Maxim88S  Have you read the user guide of the P4-Function-Board? https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/user_guide.htm For example, how do you connect the `RST_LCD`?

closed 2024-11-14 by suda-morris