espressif/esp-idf #14851: ESP-IDF v5.3.1 i2c_master.h Causing Unexpected Nack

K0I05, 2024-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDFv5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32s3box

### Power Supply used.

USB

### What is the expected behavior?

I am polling two I2C sensors (SHT40 and BMP280) on the same bus at a 3-second interval.  In addition, I placed a 50-ms delay between sensor polling i.e. the SHT40 polls, delay for 50-ms, then the BMP280 polls. 

### What is the actual behavior?

After about 20 polls there is an odd error that appears but doesn't break my code or crash the MCU.

E (1066671) i2c.master: I2C transaction unexpected nack detected
E (1066671) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1066671) i2c.master: i2c_master_receive(1117): I2C transaction failed

### Steps to reproduce.

1. Step SHT40 and BMP280 components are loaded [github](https://github.com/K0I05/esp32-s3)
2. Step setup a task to run every 3-seconds and poll the sensors at a 3-second interval.

### Debug Logs.

```plain
W (881661) I2C-MQTT MACHBASE [APP]: Free Memory: 8526540 bytes (32 bytes Consumed)
W (881661) I2C-MQTT MACHBASE [APP]: Free Stack Memory: 1016 bytes (heap_size_task)
W (881671) I2C-MQTT MACHBASE [APP]: Free Stack Memory: 1504 bytes (sample_sensor_task)
W (881671) I2C-MQTT MACHBASE [APP]: Free Stack Memory: 2148 bytes (publish_sensor_task)
E (883671) i2c.master: I2C transaction unexpected nack detected
E (883671) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (883671) i2c.master: i2c_master_receive(1117): I2C transaction failed
I (883681) I2C-MQTT MACHBASE [APP]: SHT40 Air Temperature:       22.85 C
I (883681) I2C-MQTT MACHBASE [APP]: SHT40 Relative Humidity:     39.33 %
I (883691) I2C-MQTT MACHBASE [APP]: SHT40 Dewpoint Temperature:  8.28 C
I (883751) I2C-MQTT MACHBASE [APP]: BMP280 Atmospheric Pressure: 997.55 hPa
I (886671) I2C-MQTT MACHBASE [APP]: SHT40 Air Temperature:       22.82 C
I (886671) I2C-MQTT MACHBASE [APP]: SHT40 Relative Humidity:     39.32 %
I (886681) I2C-MQTT MACHBASE [APP]: SHT40 Dewpoint Temperature:  8.25 C
I (886731) I2C-MQTT MACHBASE [APP]: BMP280 Atmospheric Pressure: 997.55 hPa
I (889671) I2C-MQTT MACHBASE [APP]: SHT40 Air Temperature:       22.83 C
[... 230 lines cut ...]
I (1054731) I2C-MQTT MACHBASE [APP]: BMP280 Atmospheric Pressure: 997.45 hPa
I (1057671) I2C-MQTT MACHBASE [APP]: SHT40 Air Temperature:       22.87 C
I (1057671) I2C-MQTT MACHBASE [APP]: SHT40 Relative Humidity:     39.23 %
I (1057681) I2C-MQTT MACHBASE [APP]: SHT40 Dewpoint Temperature:  8.26 C
I (1057731) I2C-MQTT MACHBASE [APP]: BMP280 Atmospheric Pressure: 997.44 hPa
I (1060671) I2C-MQTT MACHBASE [APP]: SHT40 Air Temperature:       22.86 C
I (1060671) I2C-MQTT MACHBASE [APP]: SHT40 Relative Humidity:     39.24 %
I (1060681) I2C-MQTT MACHBASE [APP]: SHT40 Dewpoint Temperature:  8.26 C
I (1060731) I2C-MQTT MACHBASE [APP]: BMP280 Atmospheric Pressure: 997.39 hPa
W (1061661) I2C-MQTT MACHBASE [APP]: Free Memory: 8526540 bytes (32 bytes Consumed)
W (1061661) I2C-MQTT MACHBASE [APP]: Free Stack Memory: 1016 bytes (heap_size_task)
W (1061671) I2C-MQTT MACHBASE [APP]: Free Stack Memory: 1504 bytes (sample_sensor_task)
W (1061671) I2C-MQTT MACHBASE [APP]: Free Stack Memory: 2148 bytes (publish_sensor_task)
I (1063671) I2C-MQTT MACHBASE [APP]: SHT40 Air Temperature:       22.88 C
I (1063671) I2C-MQTT MACHBASE [APP]: SHT40 Relative Humidity:     39.22 %
I (1063681) I2C-MQTT MACHBASE [APP]: SHT40 Dewpoint Temperature:  8.27 C
I (1063731) I2C-MQTT MACHBASE [APP]: BMP280 Atmospheric Pressure: 997.42 hPa
E (1066671) i2c.master: I2C transaction unexpected nack detected
E (1066671) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1066671) i2c.master: i2c_master_receive(1117): I2C transaction failed
```

### More Information.

I've tried this only mulple ESP32-S3 boards.  Since the release of v5.3.1 I noticed some odd behaviours with timing.  I2C sensors that used to work with master_transmit_receive no longer work.  I had to recode with descreet transmit, delay for 10-ms, then receive functions to resolve the problems.  Is anyone else experiencing this?

K0I05, 2024-11-08:
This is most likely a hardware issue with the SHT40 but I never had this problem before, I am using adafruit SHT40 and BMP280 MEMS boards with STEMMA QT.  I just wired up an AHT20+BMP280 board to an ESP32-S3 and they get along well, no unexpected nack at a 3 second polling interval.   However, I can longer poll the status register on SHTXX (i.e. SHT10, 20, 21) MEMS sensors any more for data ready, it causes unexpected nack when polled, the hard coded delay seems to work but I was able poll for data ready before ESP-IDF v5.3.1.  I have an SHT45 that I am going to try next to see if the problem is reproduced.

mythbuster5, 2024-11-11:
Then I may ask do you know why the nack should happen? You may check this via logic analyzer. I guess it's the same situation with https://github.com/espressif/esp-idf/issues/14715

K0I05, 2025-01-20:
The older the sensor model, the more likely a NACK will occur after running various test cases with different sensors and models.  However, I've implemented a retry routine which seems to solve the problem in most cases.  If no one else is experiencing this intermittent issue we can close this issue given it appears to be hardware-related.

mythbuster5, 2025-01-21:
Yeah, that is hardware related, it's not the esp-idf driver issue, so I'm going to close it.

closed 2025-01-21 by mythbuster5

wyp1207, 2025-05-15:
[image]
大佬帮我看看怎么办，下载的源码编译没问题，烧录报错