espressif/esp-idf #14852: ESP-IDF v5.3.1 I2C的transmit方法报错

BasicGearwheel, 2024-11-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 S3 WROOM- 1 DevKitC-1 

### Power Supply used.

USB

### What is the expected behavior?

正常发送一个字符。

### What is the actual behavior?

当我使用transmit方法时， 开发板报黄字并且重启。

### Steps to reproduce.

#include "driver/i2c_master.h"
#include "driver/i2c_slave.h"
#include "esp_log.h"

i2c_master_bus_config_t i2c_mst_config = {
    .clk_source = I2C_CLK_SRC_DEFAULT,
    .i2c_port = I2C_NUM_0,
    .scl_io_num = GPIO_NUM_10,
    .sda_io_num = GPIO_NUM_3,
    .glitch_ignore_cnt = 7,
    .flags.enable_internal_pullup = true,
};
i2c_master_bus_handle_t bus_handle;

i2c_device_config_t mpu_cfg = {
    .dev_addr_length = I2C_ADDR_BIT_LEN_7,
    .device_address = 0x68,
    .scl_speed_hz = 100000,
};

i2c_master_dev_handle_t mpu_handle;

uint8_t a = 0x80;

void app_main (void) {
    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
    ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &mpu_cfg, &mpu_handle));
    ESP_ERROR_CHECK(i2c_master_transmit(mpu_handle, a, sizeof(a), -1));
}

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200bf83  PS      : 0x00060333  A0      : 0x8200cbf4  A1      : 0x3fc985f0
0x4200bf83: i2c_ll_write_txfifo at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/hal/esp32s3/include/hal/i2c_ll.h:682
 (inlined by) s_i2c_write_command at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2c/i2c_master.c:164

A2      : 0x3fc99a90  A3      : 0x3fc99dc4  A4      : 0x00000080  A5      : 0x3fc99dbc
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000030  A9      : 0x00000000
A10     : 0x00000080  A11     : 0x60013000  A12     : 0x00000900  A13     : 0x00060323
A14     : 0xffff0000  A15     : 0x0000cdcd  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000080  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x4200bf80:0x3fc985f0 0x4200cbf1:0x3fc98640 0x4200ce48:0x3fc98670 0x4200cf28:0x3fc986b0 0x4200da4a:0x3fc986e0 0x42008ded:0x3fc98740 0x4201a9db:0x3fc98760 0x4037a941:0x3fc98790
0x4200bf80: i2c_ll_write_txfifo at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/hal/esp32s3/include/hal/i2c_ll.h:682
 (inlined by) s_i2c_write_command at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2c/i2c_master.c:164
0x4200cbf1: s_i2c_send_commands at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2c/i2c_master.c:457
0x4200ce48: s_i2c_transaction_start at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2c/i2c_master.c:584
0x4200cf28: s_i2c_synchronous_transaction at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2c/i2c_master.c:872
0x4200da4a: i2c_master_transmit at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_driver_i2c/i2c_master.c:1074
0x42008ded: app_main at E:/Develop/Project/ESP-IDF/IIC/main/main.c:29
0x4201a9db: main_task at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/app_startup.c:208
0x4037a941: vPortTaskWrapper at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

mythbuster5, 2024-11-11:
Hello? Could you please check the function declaration `esp_err_t i2c_master_transmit(i2c_master_dev_handle_t i2c_dev, const uint8_t *write_buffer, size_t write_size, int xfer_timeout_ms);` and find why?

closed 2024-11-11 by mythbuster5

BasicGearwheel, 2024-11-16:
@mythbuster5 

Thank you.I had used the method in correct way before.But it still didn't work normally.The message from ESP32-S3 is actually the following text .Now the problem has been handled, because I have found that the GPIO which connect IIC devices are wrong.

E (50) i2c.master: I2C transaction unexpected nack detected
E (50) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (60) i2c.master: i2c_master_transmit(1074): I2C transaction failed
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x42008e01
0x42008e01: app_main at E:/Develop/Project/ESP-IDF/IIC/main/main.c:30 (discriminator 1)

file: "./main/main.c" line 30
func: app_main
expression: i2c_master_transmit(mpu_handle, ac, sizeof(ac), -1)

abort() was called at PC 0x40379e47 on core 1
0x40379e47: _esp_error_check_failed at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/esp_err.c:49

Backtrace: 0x403759ea:0x3fc98650 0x40379e51:0x3fc98670 0x40380705:0x3fc98690 0x40379e47:0x3fc98700 0x42008e01:0x3fc98730 0x4201a9e7:0x3fc98760 0x4037a941:0x3fc98790
0x403759ea: panic_abort at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/panic.c:463
0x40379e51: esp_system_abort at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/port/esp_system_chip.c:92
0x40380705: abort at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/newlib/abort.c:38
0x40379e47: _esp_error_check_failed at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/esp_err.c:49
0x42008e01: app_main at E:/Develop/Project/ESP-IDF/IIC/main/main.c:30 (discriminator 1)
0x4201a9e7: main_task at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/app_startup.c:208
0x4037a941: vPortTaskWrapper at E:/Develop/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134