espressif/esp-idf #14856: blink: RGB LED doesn't stay on on ESP32-C6-DevKitC-1

dmage, 2024-11-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-4076-gce6085349f

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

The LED blinks, but it stays on for about a second.

### What is the actual behavior?

The LED flashes and instantly turns off. Video: https://www.youtube.com/watch?v=3eTQghAk8Ro (visible only on 60fps)

### Steps to reproduce.

```
~/esp/master/esp-idf (master)$ cd ./examples/get-started/blink/
~/esp/master/esp-idf/examples/get-started/blink (master)$ idf.py set-target esp32c6
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
<...>
-- Configuring done (253.8s)
-- Generating done (1.5s)
-- Build files have been written to: /mnt/storage/user_esp/master/esp-idf/examples/get-started/blink/build
~/esp/master/esp-idf/examples/get-started/blink (master)$ idf.py -p /dev/ttyACM0 build flash monitor
<...>
I (245) main_task: Started on CPU0
I (245) main_task: Calling app_main()
I (245) example: Example configured to blink addressable LED!
I (255) example: Turning the LED OFF!
I (1255) example: Turning the LED ON!
I (2255) example: Turning the LED OFF!
I (3255) example: Turning the LED ON!
```

### More Information.

If I change color in `main/blink_example_main.c`

```diff
-led_strip_set_pixel(led_strip, 0, 16, 16, 16);
+led_strip_set_pixel(led_strip, 0, 16, 0, 0);
```

it starts to work as expected. It works with (16,0,0), (0,16,0), (0,0,16), but it doesn't work with (16,16,0) or (16,16,16).

suda-morris, 2024-11-11:
I can't reproduce the issue on my ESP32-C6-DevKitC-1 V1.1 board. I watched your video, maybe your LED (WS2812) is broken somehow?

Sourtoast, 2025-03-05:
Hi I have the exact same issue with my board

Im on macOS, ESP32-C6-DevKitC-1
IDF version: ESP-IDF v5.5-dev-2182-ga6c3a9cbbb
Chip version: ESP32-C6 (QFN40) (revision v0.1)

AlbertVos, 2025-03-11:
I have the same issue too. 
When I received my boards they were programmed with some code that seemed to work well, it looped through some colors , including red, which were shown correct. I don't have that code however.

Sourtoast, 2025-03-23:
I had the same issue - it acted just like in your video. The only way I could get it to work was by using values below 12% and avoiding mixing colors. The red one was especially moody.

To rule out any software or board architecture issues, I connected an onboard LED to the ATmega328p microcontroller using pin headers and programmed the microcontroller to light up different colors. It had the same problem.

I ended up returning the board and ordering a new one. **It seems like the issue was with the LED itself** because my new board doesn’t have the problem.

If anyone decides to replace the LED on their board instead of returning it, please let me know how it goes. I'm curious if that solves the issue.

chanduCR, 2025-04-28:
Same issue, setting the RGB values to 5, 5, 5 and 2, 2, 2 works as expected. (IDF v5.4.1), will try to do some more tests