espressif/esp-idf #14857: RGB Display ISR's running on core that is not the app core.

kdschlosser, 2024-11-10:
### Is your feature request related to a problem?

RGB displays are problematic as we know and that is due to needing a constant stream of data being passed to it. The problem is exponentially worse when dealing with displays that have a large pixel count (800 x 480). If software rotation is thrown into the mix that makes things all kinds of worse to deal with and it consumes a much larger amount of memory because of the copying from one buffer to another.

### Describe the solution you'd like.

ISR's are a great thing when wanting to have code run immediately as the result of some kind of a hardware change that is taking place on the ESP32. There is a downside to them and that is too many ISR's happening will cause a very large amount of wasted processing time due to saving the core state, and reloading that saved core state. It also stops the execution of user code while the ISR is running. 

### Describe alternatives you've considered.

When using an ESP32-S3 there are 2 processor cores that are available. One core is traditionally used for the application code and the other is used for things that are running form the ISP-IDF internally. Blueteeth and WiFi have portions of it that run on that other core if I am not mistaken.

ISR's occur on the same core the ISR was registered on correct? If that is the case can we move the RGB ISR's so they run on the non application core? I believe this is how it should be done ideally. While I don't think that it will make an "Oh Wow" can of an impact that is going to be easily seen I do feel that it will make a difference and one that is big enough that when added to other user code optimizations would make a noticeable difference.

### Additional context.

As always thoughts on this are always welcome. I am no expert that's for sure and if my understanding on how things work is not correct please by all means let me know and explain to me how it does work. I am 100% onboard with learning.

There has to be some way of boosting the performance when using the RGB display driver.

suda-morris, 2024-11-11:
Hi @kdschlosser Here're some useful tips for using the LCD driver: https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/peripherals/lcd.html#how-can-i-improve-the-display-frame-rate-of-lcd-screens

> ISR's occur on the same core the ISR was registered on correct?

Yes. So if you want to install the LCD's ISR on a specific CPU core. You can create a temporary task by `xTaskCreatePinnedToCore`, because of the task is pined to a specific core, when you add your LCD initialization code in that task function, its interrupt will be installed on that core as well.

kdschlosser, 2024-11-11:
> Yes. So if you want to install the LCD's ISR on a specific CPU core. You can create a temporary task by xTaskCreatePinnedToCore, because of the task is pined to a specific core, when you add your LCD initialization code in that task function, its interrupt will be installed on that core as well.

My suggestion is to have this done in the ESP-IDF. The ESP-IDF has mechanisms to separate IO related code so it runs on a different core. This is done for WiFi and I want to say Bluetooth as well. Seeing as how demanding the LCD's IO can be especially the RGB driver how come this is not something that is built in?

My request was to have this as a built in feature of the IDF.

I also do wish it was as simple as just adding a task and pinning it to a different core it's really not that cut and dry. There are ESP32's that only have a single core so those would have to be handled differently. a Mutex would also have to be added in order to hold up the application core until the other core finished running the display setup code... 

One thing that could be problematic are the callback functions for things like the vsync. Those would need to run on the application core I would think. There is user code that would run ion that callback and having it run on the core that is not for the application could cause issues no?

Multicore processing can be tricky to get working properly and it would take a person that has a fair bit of knowledge of the code in the IDF to get it done properly. This is the reason why ideally it should be done in the IDF and no in user code.

kdschlosser, 2024-11-11:
one of the things that has made the ESP32 so successful is the SDK and it handling most of the complicated stuff. Things that a new user would have a hard time getting to work properly. 

I have spent some time working with the RGB driver and I am able to keep everything synchronized correctly without needing to stall the application. I made the suggestion to add an additional function prototype that only gets used the first time a vsync occurs after a buffer change occurs.

RGB panel example for IDFv5.4
https://github.com/espressif/esp-idf/blob/906f36bdb9bcc572fff5ac26a305e2d166895f7c/examples/peripherals/lcd/rgb_panel/main/rgb_lcd_example_main.c#L97

example in the master branch
https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/examples/peripherals/lcd/rgb_panel/main/rgb_lcd_example_main.c#L104

you can see there is no semaphore being used that stalls the application until the vsync occurs after passing a buffer to the RGB panel driver.

 
I was able to produce the same result as what this change does without needing to edit the IDF. It was a rather hack kind of a thing but it does work. 

My point is handling things like what is seen above or this specific feature request would be easier to do as part of the IDF code rather than a user having to implement it.

atanisoft, 2024-11-11:
> You can create a temporary task by `xTaskCreatePinnedToCore`

Another alternative to this is to use `esp_ipc` as shown below for a custom TWAI driver:
```
static void esp32_twai_isr_init(void *param)
{
    LOG(VERBOSE, "ESP-TWAI: Allocating ISR");
    ESP_ERROR_CHECK(
        esp_intr_alloc(ETS_TWAI_INTR_SOURCE, TWAI_INTERRUPT_FLAGS, twai_isr,
            nullptr, &twai.isr_handle));
}

void esp_twai_hw_init()
{
....

#if SOC_CPU_CORES_NUM > 1
    ESP_ERROR_CHECK(
        esp_ipc_call_blocking(preferredIsrCore_, esp32_twai_isr_init, nullptr));
#else
    esp32_twai_isr_init(nullptr);
#endif // SOC_CPU_CORES_NUM > 1

...
}
```

This doesn't require a throw away task or (possibly) heap churn.

closed 2026-09-16 by kdschlosser