espressif/esp-idf #14860: esp_lcd RGB driver callback on_color_trans_done is not going to work right,

kdschlosser, 2024-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

current master branch

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

N/A

### Power Supply used.

USB

### What is the expected behavior?

for the callback to be called after the buffer has been written to the display

### What is the actual behavior?

it is being called before the buffer has been written to the display

### Steps to reproduce.

OK so the purpose of that callback is to signal when the buffer has been flushed to the display. calling the callback prior to it actually being sent is NOT going to work and you will end up with tearing.

so if you register an `on_vsync` callback and also register the `on_color_trans_done` callback and you time how long it takes between each of those callbacks being made you will find that it is actually a fairly substantial amount of time. 

so say you are using double buffering and the buffers are located in DMA RAM. That would mean that while one buffer is being transmitted the other buffer is going to be transmitting. If the `rgb_panel_draw_bitmap` function gets called just as the sending buffer starts it's transmit and say the display resolution is 800 x 480 x 24bit. that is going to take a while to transmit. The only work performed by the `rgb_panel_draw_bitmap` could be simply changing the current buffer index and not much more. That is going to finish really fast. Way faster than the buffer that is sending is going to be able to finish in. The purpose to the `on_color_trans_done` callback is to be able to signal that the buffer has finished transferring not before it has even started. So what ends up happening is the GUI code gets signaled that it is OK to write to the next buffer when it really isn't because this buffer can still be in the process of being written to the display. If the GUI code renders data to the buffer that is transmitting it is going to cause issues. 

### More Information.

to make it work properly the following would need to be done.

to make it work properly 2 fields would need to be added to `esp_rgb_panel_t`

```c
uint8_t real_fb_index;
uint8_t trans_done_index;
```

2 lines need to be added to the `esp_lcd_new_rgb_panel` function

```c
rgb_panel->real_fb_index = 255;
rgb_panel->trans_done_index = 255;
```

the use of `on_color_trans_done` in the `rgb_panel_draw_bitmap` function would need to be removed

and the the `rgb_lcd_default_isr_handler` ISR handler needs to be written like this.

```c
IRAM_ATTR static void rgb_lcd_default_isr_handler(void *args)
{
    esp_rgb_panel_t *rgb_panel = (esp_rgb_panel_t *)args;
    bool need_yield = false;

    // clear the interrupt status
    uint32_t intr_status = lcd_ll_get_interrupt_status(rgb_panel->hal.dev);
    lcd_ll_clear_interrupt_status(rgb_panel->hal.dev, intr_status);

    // VSYNC event happened
    if (intr_status & LCD_LL_EVENT_VSYNC_END) {
        if (rgb_panel->real_fb_index == 255) {
             rgb_panel->real_fb_index = rgb_panel->cur_fb_index;
        }
        if (rgb_panel->trans_done_index != rgb_panel->real_fb_index) {
            rgb_panel->trans_done_index = rgb_panel->real_fb_index;

            if (rgb_panel->on_color_trans_done) {
                if (rgb_panel->on_color_trans_done(&rgb_panel->base, NULL, rgb_panel->user_ctx)) {
                    need_yield = true;
                }
            }
        }
        // call user registered callback
        if (rgb_panel->on_vsync) {
            if (rgb_panel->on_vsync(&rgb_panel->base, NULL, rgb_panel->user_ctx)) {
                need_yield = true;
            }
        }

        if (rgb_panel->real_fb_index != rgb_panel->cur_fb_index) {
            rgb_panel->real_fb_index = rgb_panel->cur_fb_index;
        }
        // check whether to update the PCLK frequency, it should be safe to update the PCLK frequency in the VSYNC interrupt
        lcd_rgb_panel_try_update_pclk(rgb_panel);

        if (rgb_panel->flags.stream_mode) {
            // check whether to restart the transmission
            lcd_rgb_panel_try_restart_transmission(rgb_panel);
        }

    }
    // yield if needed
    if (need_yield) {
        portYIELD_FROM_ISR();
    }
}
```

The assumption I am making here is there is no way the current index would change after a vsync has finished and before the next vsync starts to run. I find it highly unlikely that this would happen but maybe it could? if it could then a mutex would be needed to guard against the current frame buffer index changing between vertical syncing of the display. I imagine the amount of time from when one vsync ends and the next starts would be incredibly small so if the application needs to be stalled the amount of time it would be stalled for would be really tiny.

The reason for the `real_fb_index` is that would hold the index of the frame buffer that is actually being sent instead of the index of the frame buffer that is next to be sent. because the current index is able to be changed while a buffer is transmitting is the reason why this needs to exist. The `trans_done_index` not matching the real index is what triggers the callback to get called. this way if the same buffer keeps on being sent over and over again the callback would only get called a single time only after the index has changed. This is written for double buffering handling and not for single buffer handling. That is something that would need to added or figured out when to call the callback. I don't believe that single buffer is going to be used due to tearing happening so it really becomes moot for adding handling of a single buffer. Unless the display is super small but in those cases an RGB bus is not going to be used with it SPI would be.

suda-morris, 2025-01-13:
Hi @kdschlosser Thanks for the detailed explanation. 

The LCD driver itself doesn't handle the tearing effect on purpose, but providing many kinds of callback functions. We meet some users who don't care about the tearing but really care about the FPS. Enabling tearing effect can affect the FPS by a lot. So we let the user do the decision.

But I agree, when we use double frame buffer, we can't only rely on the `on_color_trans_done` callback, but also need the `on_frame_buf_complete` callback. Because the buffer we passed to the "esp_lcd_panel_draw_bitmap" function is not a "draw buffer" but a "frame buffer (partially)".

kdschlosser, 2025-01-13:
If you take a look at the code I wrote to handle the RGB driver I will say it definitely faster. It does however come with the cost of additional memory. How the tradeoff works is the 2 frame buffers the esp_lcd RGB driver creates resides in DMA PSRAM and then 2 more partial frame buffers get created for LVGL in the internal memory. I initialize the RGB diver in a FreeRTOS task running on Core 0 and LVGL runs on core 1. using semaphores I am able to control the task that is running on core 0 from core 1. When LVGL flushes a partial buffer LVGL is then told it can write to the second partial buffer. The first partial buffer is handed off to the task running on core 0. A check is made to see if that partial is the last in a series of them. The buffer then gets copied and rotated if needed to the idle full frame buffer from the RGB driver. If the partial buffer is the last in the series I then tell the RGB driver to swap the buffers and the core 0 thread then gets stalled. I use the VSYNC callback to know when the swap has actually taken place. When it has taken place I then copy the buffer that is transmitting to the now idle full frame buffer. This keeps the buffers in sync. All of this work is done on core 0 while LVGL is rendering to a partial buffer.

This works the best because LVGL only has to render small areas of the buffer, the things that have changed. That cuts down on the time it takes LVGL to update the display. I am essentially killing 2 birds with one stone by copying the buffer at the same time it is transmitting. There is more than enough time for the copy of the full buffer to take place because there are 2 partial buffers that LVGL would have to render before the copying has finished. The only case where that would happen is if the partial buffers are sized too small. The size the partial needs to be depends on how much of the GUI is changing at once. if it takes more than 3 partial buffer renders to update the UI then the buffer size should be increased. Keeping everything running in sync without causing any stalls took a bit of time to put together.

Users that have tested it say it is faster than having LVGL write to the full frame buffers and it has been reported that there is only a small decrease in speed when performing a software rotation. 

I think that this type of thing should be considered for use in the ESP-IDF. It's use would be optional of course. 

The other thing that needs to be done is the RGBPanel driver needs to really be a bus driver. a separate panel driver should be made for RGB just like you have the SPI bus driver and I8080 bus driver and you are able to have the display drivers work without knowing anything about the underlying bus that is being used. The reason why I say this is because there are quite a few RGB displays that use some kind of a bridge IC and that IC needs to be set up using SPI3wire. With those displays and the RGB panel driver getting things to all work together correctly gets a bit more challenging especially if the 3wire pins are shared with the display bus data pins.

Take a look at the code I wrote that handles the RGB displays. It might provide some ideas to refine the RGB driver to make it easier to use and possibly make it perform better. 

https://github.com/lvgl-micropython/lvgl_micropython/blob/main/ext_mod/lcd_bus/esp32_src/rgb_bus_rotation.c

I am using IDF 5.2.3 in that project. It binds LVGL to MicroPython. It works better than the "official" LVGL MicroPython binding. It is way easier to compile, it handles collecting all of the dependencies that are needed. It has a grip of build in drivers for display ICs, touch ICs and IO expander ICs. It supports SPI, I2C, I8080 and RGB displays. It also works with displays that have bridge ICs using SPI3wire. You should give an RGB display a test drive using it, It's actually really surprising how quick it is.

closed 2026-09-16 by kdschlosser