espressif/esp-idf #14861: Unhandled interrupt 58 on CPU 0

Lyp82ndbskl, 2024-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Espressif SoC revision.

ESP32-WROOM-32E-N4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E-N4

### Power Supply used.

USB

### What is the expected behavior?

This error message should not be printed

### What is the actual behavior?

When the device is powered on again and the WiFi and Bluetooth functions are initialized, the device may print this error message.

### Steps to reproduce.

When the device is powered on again and the WiFi and Bluetooth functions are initialized, the device may print this error message.

### Debug Logs.

```plain
Unhandled interrupt 58 on cpu 0!
Unhandled interrupt 58 on cpu 0!
```

SoucheSouche, 2024-11-11:
Hi @Lyp82ndbskl, thank you for reporting the issue. Could you maybe be more specific concerning the steps to reproduce? Maybe post a code sample that triggers the issue so we can reproduce it on our side?

ESP-Marius, 2024-11-19:
Closing this one due to lack of information. If anyone has any more info to help us debug/reproduce this, then please feel free to reopen the ticket.

closed 2024-11-19 by ESP-Marius

udoudou, 2025-07-04:
@Lyp82ndbskl Do you still have this problem? I have a fix that may be related to this. I can provide it for you to test