espressif/esp-idf #14862: how to disable CONFIG_ESP_TLS_SERVER?

zzzh, 2024-11-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

in menu config of esp-idf, being able to unselect the "Enable ESP-TLS Server" option(showing [*] instead of -*- before this item)

### What is the actual behavior?

in menu config of esp-idf, it's -*- before "Enable ESP-TLS Server", make it impossible to unselect it.

### Steps to reproduce.

it's enabled by default after select ESP_HTTPS_SERVER_ENABLE

### More Information.

Got "httpd: httpd_server_init: error in listen (112)" when running httpd_start(&server, &config), suspect it's because CONFIG_ESP_TLS_SERVER=y

KaeLL, 2024-11-12:
Press `?` and check which other kconfig option is y-selecting it.

zzzh, 2024-11-12:
> Press `?` and check which other kconfig option is y-selecting it.

Good idea,  I tried that, turns out it's ESP_HTTPS_SERVER_ENABLE make it fixed, which should not be select to use only a http server, I selected it to see if it can solve the "httpd: httpd_server_init: error in listen (112)" issue..
Any ideas on "httpd: httpd_server_init: error in listen (112)"?