espressif/esp-idf #14864: Allow customize NimBLE configs with `MYNEWT_` prefix

Tnze, 2024-11-11:
### Is your feature request related to a problem?

I'm using NimBLE to create a BLE server with DIS service. It turns out that I have to enable following configs to have all characteristics in DIS.

- MYNEWT_VAL_BLE_SVC_DIS_FIRMWARE_REVISION_READ_PERM
- MYNEWT_VAL_BLE_SVC_DIS_HARDWARE_REVISION_READ_PERM
- MYNEWT_VAL_BLE_SVC_DIS_SOFTWARE_REVISION_READ_PERM
- MYNEWT_VAL_BLE_SVC_DIS_MANUFACTURER_NAME_READ_PERM

### Describe the solution you'd like.

These configs are defined in "esp-idf\components\bt\host\nimble\port\include\esp_nimble_cfg.h", which (in my opinion) should not be modified directly.

I want some method to customize the configurations without modify esp-idf itself.

Tnze, 2024-11-12:
Thanks, I tested the patch and it works.

rahult-github, 2024-12-11:
Changes pushed on master branch. Closing issue.

closed 2024-12-11 by rahult-github