espressif/esp-idf #14869: Why isn't the new I2C API deemed as thread safe?

wuyuanyi135, 2024-11-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

According to the [doc](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-reference/peripherals/i2c.html#thread-safety), only bus creation was deemed as thread safe. Tracing the source code I found that the API are all semaphore-guarded that prevent racing access on the same bus:

https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/components/esp_driver_i2c/i2c_master.c#L898

If so, why are we recommended to put extra locks as per the [doc](https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/components/esp_driver_i2c/i2c_master.c#L898)?

mythbuster5, 2024-11-12:
Yes, the doc is wrong and it will be updated soon

```
The factory function :cpp:func:`i2c_new_master_bus` and :cpp:func:`i2c_new_slave_device` are guaranteed to be thread safe by the driver, which means that the functions can be called from different RTOS tasks without protection by extra locks.

I2C master operation functions are also guaranteed to be thread safe by bus operation semaphore.

- :cpp:func:`i2c_master_transmit`
- :cpp:func:`i2c_master_multi_buffer_transmit`
- :cpp:func:`i2c_master_transmit_receive`
- :cpp:func:`i2c_master_receive`
- :cpp:func:`i2c_master_probe`
```

closed 2024-11-12 by mythbuster5

wuyuanyi135, 2024-11-12:
@mythbuster5  Thank you for the confirmation