espressif/esp-idf #14875: NimBLE examples panics when connected from an iPhone using `nRF Connect` app

Tnze, 2024-11-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-H2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Waveshare ESP32-H2-DEV-KIT-N4

### Power Supply used.

USB

### What is the expected behavior?

The example doesn't panic. And the BLE works.

### What is the actual behavior?

When connected from an iPhone, the ESP-32 panic and reboot.
After crashing and restarting several times, and connecting several times, the connection can then be successful.

### Steps to reproduce.

1. Run the `power save` example
2. Run the ESP-IDF Monitor
3. Scan BLE devices on iPhone using the "nRF Connect" app (I'm using iPhone 12 Pro Max), and try connecting the device `nimble-bleprph-e`
4. The firmware fatal and reboot

### Debug Logs.

```plain
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM20 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM20 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x17ac
load:0x4083cad0,len:0xefc
load:0x4083efd0,len:0x2ce4
entry 0x4083cada
I (23) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (24) boot: compile time Nov 12 2024 21:19:21
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
[... 828 lines cut ...]

I (4893) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (5316) NimBLE: connection established; status=0
I (5317) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (5367) NimBLE: 74:4d:bd:62:05:5f
I (5368) NimBLE:  our_id_addr_type=0 our_id_addr=
I (5369) NimBLE: 74:4d:bd:62:05:5f
I (5371) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (5376) NimBLE: 7c:f8:9a:ef:5c:68
I (5380) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (5386) NimBLE: 7c:f8:9a:ef:5c:68
I (5390) NimBLE:  conn_itvl=21 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0

I (5400) NimBLE:

I (5407) NimBLE: advertise complete; reason=0
I (5411) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (5480) NimBLE: Reattempt advertising; reason: 0x3e, status = 0
```

### More Information.

It's noticed that the rear part of the logs is quite similar to #12911

Shreeyash-17, 2024-11-13:
Based on the IDF version and chip name you provided, I was able to reproduce the issue. I have included the log below. Also, please ensure that your codebase is clean and properly synced. Could you kindly share the sdkconfig file?

Log that I got :

I (23) bot: ESP-IDF v5.3.1 2nd stage bootloader
I (24) boot: compile time Nov 13 2024 12:50:40
I (25) boot: chip revision: v0.2
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=1b0f4h (110836) map
I (126) esp_image: segment 1: paddr=0002b11c vaddr=40800000 size=04efch ( 20220) load
I (134) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=7e4dch (517340) map
I (295) esp_image: segment 3: paddr=000ae504 vaddr=40804efc size=0f608h ( 62984) load
I (317) esp_image: segment 4: paddr=000bdb14 vaddr=40814510 size=02cbch ( 11452) load
I (322) esp_image: segment 5: paddr=000c07d8 vaddr=50000000 size=00050h (    80) load
I (324) esp_image: segment 6: paddr=000c0830 vaddr=50000050 size=00008h (     8) load
I (338) boot: Loaded app from partition at offset 0x10000
I (339) boot: Disabling RNG early entropy source...
I (355) cpu_start: Unicore app
W (1892) clk: 32 kHz clock not found, switching to internal 150 kHz oscillator
I (1915) cpu_start: Pro cpu start user code
I (1916) cpu_start: cpu freq: 96000000 Hz
I (1916) app_init: Application information:
I (1919) app_init: Project name:     power_save
I (1924) app_init: App version:      v5.3.1
I (1929) app_init: Compile time:     Nov 13 2024 12:50:35
I (1935) app_init: ELF file SHA256:  5a25a9141...
I (1940) app_init: ESP-IDF:          v5.3.1
I (1945) efuse_init: Min chip rev:     v0.0
I (1950) efuse_init: Max chip rev:     v0.99 
I (1955) efuse_init: Chip rev:         v0.2
I (1960) heap_init: Initializing. RAM available for dynamic allocation:
I (1967) heap_init: At 40819F90 len 000333F0 (204 KiB): RAM
I (1973) heap_init: At 4084D380 len 00002B60 (10 KiB): RAM
I (1981) spi_flash: detected chip: generic
I (1984) spi_flash: flash io: dio
W (1988) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (2002) sleep: Enable automatic switching of GPIO sleep configuration
I (2009) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (2031) pm: Frequency switching config: CPU_MAX: 96, APB_MAX: 32, APB_MIN: 32, Light sleep: DISABLED
I (2032) coexist: coex firmware version: 4482466
I (2037) main_task: Started on CPU0
I (2040) main_task: Calling app_main()
I (2054) pm: Frequency switching config: CPU_MAX: 96, APB_MAX: 32, APB_MIN: 32, Light sleep: ENABLED
W (2055) BLE_INIT: 32.768kHz XTAL not detected, fall back to main XTAL as Bluetooth sleep clock
I (2063) BLE_INIT: Using main XTAL as clock source
I (2075) BLE_INIT: ble controller commit:[a95ec45]
W (2075) BLE_INIT: BLE modem sleep is enabled
I (2080) BLE_INIT: Modem BLE MAC retention initialization
W (2085) BLE_INIT: Enable light sleep, the wake up source is BLE timer
I (2104) phy: phy_version: 310,2, 8fa9cb4, Jun  5 2024, 14:19:10
I (2122) phy: libbtbb version: a024da6, Jun  5 2024, 14:19:22
I (2123) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (2128) NimBLE_BLE_PRPH: BLE Host Task Started
I (2135) NimBLE: Device Address: 
I (2136) NimBLE: 60:55:f9:f7:39:cc
I (2139) NimBLE: 

I (2143) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (2153) uart: queue free spaces: 8
I (2155) main_task: Returned from app_main()
[... 126 lines cut ...]
I (123825) NimBLE: connection updated; status=0 
I (123826) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (123876) NimBLE: 60:55:f9:f7:39:cc
I (123879) NimBLE:  our_id_addr_type=0 our_id_addr=
I (123880) NimBLE: 60:55:f9:f7:39:cc
I (123882) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (123888) NimBLE: 50:6e:89:2d:aa:68
I (123892) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (123898) NimBLE: 50:6e:89:2d:aa:68
I (123902) NimBLE:  conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

I (123914) NimBLE: 

I (304007) NimBLE: disconnect; reason=531 
I (304009) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (304059) NimBLE: 60:55:f9:f7:39:cc
I (304060) NimBLE:  our_id_addr_type=0 our_id_addr=
I (304061) NimBLE: 60:55:f9:f7:39:cc
I (304063) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (304069) NimBLE: 50:6e:89:2d:aa:68
I (304073) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (304079) NimBLE: 50:6e:89:2d:aa:68
I (304083) NimBLE:  conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

I (304093) NimBLE:

Tnze, 2024-11-13:
Looks like you didn't reproduce the assertion failure and panic.

While capturing the logs my codebase is not clean because I applied a patch from #16864. But I can ensure the issue can be reproduced on unmodified esp-idf release. I will clean the code and recapture the logs, sharing the config file later.

Tnze, 2024-11-13:
I cleanup my esp-idf and I can still reproduce this issue.
And this is my project (with built artifacts).
[power_save.zip.001.zip](https://github.com/user-attachments/files/17737285/power_save.zip.001.zip)
[power_save.zip.002.zip](https://github.com/user-attachments/files/17737288/power_save.zip.002.zip)

```
40829150: 0x66207472 0x656c6961 0x65203a64 0x625f7478 0x7270656c 0x615f6870 0x72657664 0x65736974
40829170: 0x69616d20 0x3a632e6e 0x20393231 0x20637228 0x30203d3d 0x00000029 0x40828378 0x4080e234
0x4080e234: prvCopyDataToQueue at F:/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2470

40829190: 0x00000000 0x00000000 0x40828378 0x4080e864 0x408291f4 0x00002036 0x0000020c 0x00000000
0x4080e864: xQueueGenericSend at F:/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1057

408291b0: 0x00000019 0x00002036 0x00000000 0x4201ea94 0x0000020c 0x408291f0 0x00000001 0x00000000
0x4201ea94: ble_hs_hci_cmd_tx at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:361

408291d0: 0x0000020c 0x408293e8 0x40828378 0x4080ea26 0x00000000 0x40829244 0x408201e0 0x408036b4
0x4080ea26: xQueueGiveMutexRecursive at F:/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:796
0x408036b4: npl_freertos_mutex_release at F:/esp/v5.3.1/esp-idf/components/bt/porting/npl/freertos/src/npl_os_freertos.c:470 (discriminator 1)

408291f0: 0x00000000 0x40829244 0x00000000 0x42012c3a 0x00000000 0x00000000 0x01020001 0x42012c56
0x42012c3a: ble_hs_unlock_nested at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:209
0x42012c56: ble_hs_unlock at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:240

40829210: 0x00000000 0x00000000 0x00000000 0x42000000 0x00000012 0x00000000 0x00000000 0x00000000
40829230: 0x00000012 0x408241a6 0x408292ec 0x4200e480 0x00000000 0x00000001 0x00000030 0x00000030
0x4200e480: ext_bleprph_advertise at F:/Embedded/ESP32/power_save/main/main.c:135 (discriminator 1)

40829250: 0x00000000 0x00000000 0x02010000 0x00000100 0x4208376c 0x00000000 0x408292ec 0x4200e8e2
0x4200e8e2: bleprph_gap_event at F:/Embedded/ESP32/power_save/main/main.c:345

40829270: 0x408292cc 0x00002016 0x00000000 0x00000000 0x408292e8 0x408293e8 0x40828378 0x4080ea26
0x4080ea26: xQueueGiveMutexRecursive at F:/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:796

40829290: 0x408292e8 0x408292e4 0x408201e0 0x408036b4 0x00000000 0x408292e4 0x00000000 0x42012c3a
0x408036b4: npl_freertos_mutex_release at F:/esp/v5.3.1/esp-idf/components/bt/porting/npl/freertos/src/npl_os_freertos.c:470 (discriminator 1)
0x42012c3a: ble_hs_unlock_nested at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:209

408292b0: 0x40818931 0x40819a60 0x40818918 0x42012c56 0x00000000 0x000000fc 0x00000000 0x42018446
0x42012c56: ble_hs_unlock at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:240
0x42018446: ble_gap_slave_extract_cb at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1013

408292d0: 0x00000012 0x408241a6 0x00000000 0x420187ba 0x408293e0 0x00000000 0x4200e5d0 0x00000009
0x420187ba: ble_gap_adv_finished at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1038
0x4200e5d0: bleprph_gap_event at F:/Embedded/ESP32/power_save/main/main.c:280

408292f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40829310: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xb4970101 0x408241a6 0x42019428
0x42019428: ble_gap_rx_adv_set_terminated at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1829

40829330: 0x00000012 0x408241a6 0x408241a6 0x42013aee 0x00000000 0x00000000 0x00000006 0x42013460
0x42013aee: ble_hs_hci_evt_le_adv_set_terminated at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:956
0x42013460: ble_hs_hci_evt_le_meta at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:440

40829350: 0x40818000 0x0000003e 0x408241a4 0x42013fbc 0x40818000 0x408241a4 0x40818864 0x42012b16
0x42013fbc: ble_hs_hci_evt_process at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1124
0x42012b16: ble_hs_event_rx_hci_ev at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:532

40829370: 0x00000000 0x00000000 0x40818864 0x40802d86 0x00000000 0x00000000 0x40818000 0x40802c6a
0x40802d86: npl_freertos_event_run at F:/esp/v5.3.1/esp-idf/components/bt/porting/npl/freertos/src/npl_os_freertos.c:403
0x40802c6a: nimble_port_run at F:/esp/v5.3.1/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:307

40829390: 0x420837a8 0x00000865 0x00000000 0x4200e276 0x00000000 0x00000000 0x00000000 0x40810ace
0x4200e276: bleprph_host_task at F:/Embedded/ESP32/power_save/main/main.c:534
0x40810ace: vPortTaskWrapper at F:/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:258

408293b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408293d0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x40829000 0x00000000
408293f0: 0x408176f8 0x408176f8 0x408293e8 0x408176f0 0x00000004 0x40820218 0x40820218 0x408293e8
40829410: 0x00000000 0x00000015 0x408283e4 0x626d696e 0x685f656c 0x0074736f 0x00000000 0x408293e0
[... cut ...]
I (6024) NimBLE: 74:4d:bd:62:05:5f
I (6026) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (6033) NimBLE: 7c:9f:3e:33:b4:97
I (6040) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (6042) NimBLE: 7c:9f:3e:33:b4:97
I (6046) NimBLE:  conn_itvl=24 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0

I (6056) NimBLE:

I (6062) NimBLE: advertise complete; reason=0
I (6067) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (6106) NimBLE: mtu update event; conn_handle=0 cid=4 mtu=256

```

Shreeyash-17, 2024-11-14:
@Tnze I have added print statements to help identify the cause of the crash, and I’m sharing a patch below. Please apply this patch and share the log with us. Let me know the results after doing this.

path you need to use while applying patch is :
esp-idf/components/bt/host/nimble/nimble

[patch.txt](https://github.com/user-attachments/files/17747329/patch.txt)

Tnze, 2024-11-14:
> @Tnze I have added print statements to help identify the cause of the crash, and I’m sharing a patch below. Please apply this patch and share the log with us. Let me know the results after doing [...]

@Shreeyash-17 Your patch cannot be applied even by hand because file `nimble/host/src/ble_gap.c` doesn't have a function named `ble_gap_ext_adv_params_tx_v1`.

> https://github.com/espressif/esp-nimble/blob/2cf7edde63866f5663bbcdfb8776cf29c0809456/nimble/host/src/ble_gap.c

Shreeyash-17, 2024-11-14:
@Tnze The previous patch I shared was not compatible with version 5.3.1. I am sending a new patch—please apply this one and share the log and results with us. 

Path : esp-idf/ components/bt/host/nimble/nimble : 
[ble_gap_nimble.txt](https://github.com/user-attachments/files/17749086/ble_gap_nimble.txt)
Path : esp-idf :
[power_save.txt](https://github.com/user-attachments/files/17749094/power_save.txt)

Tnze, 2024-11-14:
I have patched my codebase. I tested it twice. During the first test, due to setting too few Scrollback lines on the terminal, the previous logs were flushed out, so I tested again for the second time. I saw `Failed rc!=0` in both logs.

<details>
<summary>Log1</summary>
ext_bleprph_advertise -> ble_gap_ext_adv_configure() : 0
I (18408) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (18419) NimBLE: Reattempt advertising; reason: 0x3e, status = 0
I (18423) NimBLE: GAP procedure initiated: stop extended advertising.

ble_gap_ext_adv_configure -> (instance >= BLE_ADV_INSTANCES) :0
ble_gap_ext_adv_configure -> (!ble_hs_is_enabled()) : 0 
ble_gap_ext_adv_params_validate -> !params : 0
ble_gap_ext_adv_params_validate -> params->own_addr_type > BLE_HCI_ADV_OWN_ADDR_MAX : 0     
ble_gap_ext_adv_params_validate -> params->itvl_min : 48
ble_gap_ext_adv_params_validate -> (params->itvl_max && params->itvl_max > 0xFFFFFF) : 0    
ble_gap_ext_adv_params_validate -> (params->connectable && !ble_hs_conn_can_alloc()) : 0
ble_gap_ext_adv_params_validate -> (params->legacy_pdu) : 0
ble_gap_ext_adv_params_validate -> (params->directed) : 0
ble_gap_ext_adv_params_validate -> (!params->legacy_pdu) : 1  , (params->connectable && params->scannable) : 0  , (params->high_duty_directed) : 0
ble_gap_ext_adv_configure -> ble_gap_adv_active_instance() : 0
ble_gap_ext_adv_params_tx -> ble_hs_hci_cmd_tx() : 0
ble_gap_ext_adv_configure -> rc = ble_gap_ext_adv_params_tx() =  0
I (18512) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (18524) NimBLE: connection established; status=0
I (18530) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (18531) NimBLE: 74:4d:bd:62:05:5f
I (18534) NimBLE:  our_id_addr_type=0 our_id_addr=
I (18540) NimBLE: 74:4d:bd:62:05:5f
I (18544) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (18551) NimBLE: 46:62:1e:cc:75:f1
I (18554) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (18560) NimBLE: 46:62:1e:cc:75:f1
I (18564) NimBLE:  conn_itvl=21 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0

I (18574) NimBLE: 

I (18579) NimBLE: advertise complete; reason=0
ble_gap_ext_adv_configure -> (instance >= BLE_ADV_INSTANCES) :0
ble_gap_ext_adv_configure -> (!ble_hs_is_enabled()) : 0 
ble_gap_ext_adv_params_validate -> !params : 0
ble_gap_ext_adv_params_validate -> params->own_addr_type > BLE_HCI_ADV_OWN_ADDR_MAX : 0     
ble_gap_ext_adv_params_validate -> params->itvl_min : 48
ble_gap_ext_adv_params_validate -> (params->itvl_max && params->itvl_max > 0xFFFFFF) : 0    
ble_gap_ext_adv_params_validate -> (params->connectable && !ble_hs_conn_can_alloc()) : 0    
ble_gap_ext_adv_params_validate -> (params->legacy_pdu) : 0
ble_gap_ext_adv_params_validate -> (params->directed) : 0
ble_gap_ext_adv_params_validate -> (!params->legacy_pdu) : 1  , (params->connectable && params->scannable) : 0  , (params->high_duty_directed) : 0
ble_gap_ext_adv_configure -> ble_gap_adv_active_instance() : 0
ble_gap_ext_adv_params_tx -> ble_hs_hci_cmd_tx() : 0
ble_gap_ext_adv_configure -> rc = ble_gap_ext_adv_params_tx() =  0
ext_bleprph_advertise -> ble_gap_ext_adv_configure() : 0
I (18672) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (18684) NimBLE: Reattempt advertising; reason: 0x3e, status = 0
I (18686) NimBLE: GAP procedure initiated: stop extended advertising.

ble_gap_ext_adv_configure -> (instance >= BLE_ADV_INSTANCES) :0
ble_gap_ext_adv_configure -> (!ble_hs_is_enabled()) : 0 
ble_gap_ext_adv_params_validate -> !params : 0
ble_gap_ext_adv_params_validate -> params->own_addr_type > BLE_HCI_ADV_OWN_ADDR_MAX : 0     
ble_gap_ext_adv_params_validate -> params->itvl_min : 48
ble_gap_ext_adv_params_validate -> (params->itvl_max && params->itvl_max > 0xFFFFFF) : 0    
ble_gap_ext_adv_params_validate -> (params->connectable && !ble_hs_conn_can_alloc()) : 0
ble_gap_ext_adv_params_validate -> (params->legacy_pdu) : 0
ble_gap_ext_adv_params_validate -> (params->directed) : 0
ble_gap_ext_adv_params_validate -> (!params->legacy_pdu) : 1  , (params->connectable && params->scannable) : 0  , (params->high_duty_directed) : 0
ble_gap_ext_adv_configure -> ble_gap_adv_active_instance() : 0
ble_gap_ext_adv_params_tx -> ble_hs_hci_cmd_tx() : 0
[... 2320 lines cut ...]
I (24072) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (24077) NimBLE: 46:62:1e:cc:75:f1
I (24081) NimBLE:  conn_itvl=21 conn_latency=0 supervision_timeout=72 encrypted=0 authenticated=0 bonded=0

I (24092) NimBLE:

I (24095) NimBLE: advertise complete; reason=0
ble_gap_ext_adv_configure -> (instance >= BLE_ADV_INSTANCES) :0 
ble_gap_ext_adv_configure -> (!ble_hs_is_enabled()) : 0
ble_gap_ext_adv_params_validate -> !params : 0
ble_gap_ext_adv_params_validate -> params->own_addr_type > BLE_HCI_ADV_OWN_ADDR_MAX : 0 
ble_gap_ext_adv_params_validate -> params->itvl_min : 48
ble_gap_ext_adv_params_validate -> (params->itvl_max && params->itvl_max > 0xFFFFFF) : 0
ble_gap_ext_adv_params_validate -> (params->connectable && !ble_hs_conn_can_alloc()) : 0
ble_gap_ext_adv_params_validate -> (params->legacy_pdu) : 0
ble_gap_ext_adv_params_validate -> (params->directed) : 0
ble_gap_ext_adv_params_validate -> (!params->legacy_pdu) : 1  , (params->connectable && params->scannable) : 0  , (params->high_duty_directed) : 0
ble_gap_ext_adv_configure -> ble_gap_adv_active_instance() : 0
ble_gap_ext_adv_params_tx -> ble_hs_hci_cmd_tx() : 0
ble_gap_ext_adv_configure -> rc = ble_gap_ext_adv_params_tx() =  0
ext_bleprph_advertise -> ble_gap_ext_adv_configure() : 0
I (24189) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (24205) NimBLE: mtu update event; conn_handle=0 cid=4 mtu=256
</details>

Shreeyash-17, 2024-11-28:
@Tnze I am adding a patch file that includes a tentative fix for this issue. While I still need to work further on fully resolving the issue, this should help unblock you for now.

[power_save.txt](https://github.com/user-attachments/files/17946592/power_save.txt)

Shreeyash-17, 2025-02-04:
[https://developer.apple.com/forums/thread/742497](url)  As we can see here this issue is specific to some of the iPhone models of 
specific ios version.Since this issue does not seem on iPhone 16(ios version 18.0.1)

rahult-github, 2025-03-12:
Closing issue. Feel free to reopen in case of any updates.

closed 2025-03-12 by rahult-github