espressif/esp-idf #14876: esp_ble_gap_ext_adv_params_t参数中type影响蓝牙带宽的问题

beautifulpeer, 2024-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. ESP-IDF版本 5.3.1
2. 芯片ESP32-S3 PICO
3. BLE5.0

esp_ble_gap_ext_adv_params_t ext_adv_params_2M = {
    .type           = ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE,  
    .interval_min   = 0x20,                                      // 32 * 0.625 = 20ms
    .interval_max   = 0x20,                                      // 32 * 0.625 = 20ms
    .channel_map    = ADV_CHNL_ALL,
    .filter_policy  = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy    = ESP_BLE_GAP_PHY_1M,
    .max_skip       = 0,
    .secondary_phy  = ESP_BLE_GAP_PHY_2M,
    .sid            = 0,
    .scan_req_notif = false,
    .own_addr_type  = BLE_ADDR_TYPE_PUBLIC,
    .tx_power       = EXT_ADV_TX_PWR_NO_PREFERENCE,
};
1. 您好，esp_ble_gap_ext_adv_params_t 中type使用的ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE，带宽可以达到1.3Mbit/s；
2. 但是若将type修改成ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_IND这个类型，其它地方都不改，带宽只可以达到0.7Mbit/s；
3. 由于需要适配之前的设备，只有使用ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_IND类型，之前的设备才会扫描到ESP32-S3的设备，但使用此类型会影响蓝牙带宽，这是为什么？

beautifulpeer, 2024-11-13:
@esp-zhp

esp-zhp, 2024-11-13:
ESP_BLE_GAP_SET_EXT_ADV_PROP_LEGACY_IND 不支持 2M phy，ESP_BLE_GAP_PHY_2M 是 BLE50 引入的 feature。

esp-zhp, 2024-11-13:
BLE4.2 只能使用 1M phy，所以最大的带宽也不可能超过 1M ，不然就是不遵循 BLE 协议。

beautifulpeer, 2024-11-13:
哦哦，这样，我之前旧的设备是nordic的，用的也是BLE5.0，但是使用ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE就扫描不到，不知道是为什么

esp-zhp, 2024-11-13:
广播的 primary phy 要和 scan 的 phy 对应上，要不然没有办法扫到。

beautifulpeer, 2024-11-15:
好的，感谢您的回复

closed 2024-11-15 by beautifulpeer