espressif/esp-idf #14880: Wrong static int initialization

dizcza, 2024-11-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.5-143-gdb1c67632a

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO-T8

### Power Supply used.

USB

### What is the expected behavior?

The static var `trial` initialized to zero.

### What is the actual behavior?

The static var `trial` initialized to one.

### Steps to reproduce.

```c
static FILE* recordbuf_open_file() {
    static int trial = 0;
    ESP_LOGW(TAG, "trial %d", trial);
    trial++;
    ESP_LOGW(TAG, "trial %d", trial);
    return NULL;
}
```

If I remove the keyword static, it works as expected. I have several functions like this but only this one works incorrectly.

### Debug Logs.

```plain
W (22:30:44.503) recordbuf: trial 1
W (22:30:44.508) recordbuf: trial 2
```

### More Information.

I build the app with the `-Os` flag. Building with `-O2` or `-Og` produces the same behavior.

ESP-Marius, 2024-11-14:
Strange, I'd expect if something like static initialization was broken we'd seen it a long time ago. 

Are you sure `recordbuf_open_file` is not being called earlier, possibly when printing is not working?

If you print with `esp_rom_printf()` instead, does that make a difference?

Are you able to provide an example that reproduce this? I coudlnt with the code you posted at least :thinking:

dizcza, 2024-11-14:
> Are you sure recordbuf_open_file is not being called earlier, possibly when printing is not working?

It's called only once I'm sure about it.

> If you print with esp_rom_printf() instead, does that make a difference?

No difference.

However, the following function

```c
static FILE* recordbuf_open_file2() {
    static int trial = 0;
    ESP_LOGW(TAG, "trial %d", trial);
    return NULL;
}
```

prints "trial 0". But 

```c
static FILE* recordbuf_open_file3() {
    static int trial = 0;
    ESP_LOGW(TAG, "trial %d", trial++);
    return NULL;
}
```

prints "trial 1".

```c
static int trial = 0;
static FILE* recordbuf_open_file4() {
    ESP_LOGW(TAG, "trial %d", trial++);
    return NULL;
}
```

prints also "trial 1".

> Are you able to provide an example that reproduce this?

I cannot come up with the simplest example. If in `app_main()` only this function is called, then it works as expected (trial 0 is always printed). As I said, I have several functions like that:

```c
FILE* sdcard_record_open_file(const char *pref, const char *suff, int trial) {
    char fpath[SDCARD_MAX_PATH_SIZE];
    if (m_sdcard_record_dir == NULL) {
        ESP_LOGW(TAG, "Failed to open %s.%s", pref, suff);
        return NULL;
    }
    snprintf(fpath, sizeof(fpath), "%s/%s-%03d.%s", m_sdcard_record_dir, pref, trial, suff);
    FILE *file = fopen(fpath, "w");
    if (file == NULL) {
        ESP_LOGW(TAG, "Failed to open %s", fpath);
    } else {
        ESP_LOGI(TAG, "Opened %s", fpath);
    }
    return file;
}

static FILE* recordbuf_open_file() {
	static int file_trial = 0;
    return sdcard_record_open_file("DLVR", "BIN", file_trial++);
}
```

with different arguments to `sdcard_record_open_file`. They all work correctly. I noticed the bug because a file with the filename "DLVR-001.BIN" is created instead of "DLVR-000.BIN".

dizcza, 2024-11-14:
Silly me, not noticing this function was called twice...

closed 2024-11-14 by dizcza