espressif/esp-idf #14883: https服务器报错， esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780

lujiex, 2024-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

应该是没有报错正常响应HTTPS请求的。而且如果同时只能响应3-6个请求，我希望能到10个。

### What is the actual behavior?

实际每个请求都会报错，
[image]

有的时候都收不到HTTPS请求过来，而且还会丢请求。甚至是重启。

### Steps to reproduce.

1.https初始化代码，
[image]

2.get请求处理函数
[image]

3. https和tls配置

[image]

[image]

4.测试证书
-----BEGIN EC PRIVATE KEY-----
MHcCAQEEIJWy/cPFkbPJHUmIzedXOgP7xDe9i/lJug87z2E3gdc1oAoGCCqGSM49
AwEHoUQDQgAEuybK45+9XKR1VSCvzlXbbZFjGCqvRYCgMVyUFi+OEkddBaLOm4QR
UJGc2lUR18+YrNfxdCcvni39BKGanXAr/Q==
-----END EC PRIVATE KEY-----

-----BEGIN CERTIFICATE-----
MIICGzCCAcKgAwIBAgIUD3HmLcOmu7t+dr/P40Oq3zaVLt0wCQYHKoZIzj0EATBk
MQswCQYDVQQGEwJVUzETMBEGA1UECAwKQ2FsaWZvcm5pYTEWMBQGA1UEBwwNU2Fu
IEZyYW5jaXNjbzEUMBIGA1UECgwLRXhhbXBsZSBJbmMxEjAQBgNVBAMMCWxvY2Fs
[... 335 lines cut ...]
I (1241) wifi:ap channel adjust o:1,1 n:6,1
I (1241) wifi:new:<6,1>, old:<1,1>, ap:<6,1>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
I (1241) wifi:state: init -> auth (0xb0)
I (1251) wifi:state: auth -> assoc (0x0)
I (1281) wifi:state: assoc -> run (0x10)
I (1431) wifi:connected with idc_2.4G, aid = 16, channel 6, 40U, bssid = 24:cf:24:01:c1:dc
I (1431) wifi:security: WPA2-PSK, phy: bgn, rssi: -30
I (1431) wifi:pm start, type: 1

I (1431) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1431) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (1501) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (00:10:39.264) wifiManager: got ip:192.168.8.139
I (00:10:39.265) wifiManager: apsta mode,set NAPT
I (00:10:39.265) esp_netif_handlers: sta ip: 192.168.8.139, mask: 255.255.255.0, gw: 192.168.8.1
W (00:10:39.266) esp_netif_lwip: napt disabled on esp_netif:0x3fcd551c
I (00:10:39.269) uartManager: Uart start...
I (00:10:39.272) Select: Adding uart fd 6
I (00:10:39.273) http_server: HTTP start...
I (00:10:39.275) http_server: Registering URI handlers
I (00:10:39.277) esp_https_server: Starting server
I (00:10:39.280) esp_https_server: Server listening on port 443
I (00:10:39.281) http_server: Registering URI handlers
[0nI (27511) wifi:<ba-add>idx:0 (ifx:0, 24:cf:24:01:c1:dc), tid:0, ssn:2277, winSize:64
```

AdityaHPatwardhan, 2024-11-22:
Hi @lujiex,

Is it possible to share the log with mbedtls debug logs enabled.
`CONFIG_MBEDTLS_DEBUG=y`
as well as log level set to `debug` in menuconfig.

Please also share the `sdkconfig` file for the project.

lujiex, 2024-11-22:
> Hi @lujiex,
> 
> Is it possible to share the log with mbedtls debug logs enabled. `CONFIG_MBEDTLS_DEBUG=y` as well as log level set to `debug` in menuconfig.
> 
> Please also share the `sdkconfig` file for the project.

--- esp-idf-monitor 1.4.0 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (00:07:53.051) mbedtls: ssl_msg.c:5782 => read

I (00:07:53.052) mbedtls: ssl_msg.c:4189 => read record

I (00:07:53.053) mbedtls: ssl_msg.c:2155 => fetch input

I (00:07:53.053) mbedtls: ssl_msg.c:2295 in_leSPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x18ac
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd30
load:0x403cb700,len:0x3470
entry 0x403c89ac
I (32) boot: ESP-IDF v5.3-dirty 2nd stage bootloader
I (32) boot: compile time Oct 29 2024 16:48:54
I (32) boot: Multicore bootloader
I (33) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (33) boot.esp32s3: SPI Flash Size : 16MB
I (33) boot: Enabling RNG early entropy source...
I (34) boot: Partition Table:
I (34) boot: ## Label            Usage          Type ST Offset   Length
I (34) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (35) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (35) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (35) boot:  3 factory          factory app      00 00 00010000 00300000
I (36) boot:  4 ota_0            OTA app          00 10 00310000 00500000
I (36) boot:  5 ota_1            OTA app          00 11 00810000 00500000
I (37) boot:  6 nvs_key          NVS keys         01 04 00d10000 00001000
I (37) boot:  7 storage          Unknown data     01 81 00d11000 00100000
I (38) boot: End of partition table
I (38) boot: Defaulting to factory image
I (38) esp_image: segment 0: paddr=00010020 vaddr=3c130020 size=febach (1043372) map
I (226) esp_image: segment 1: paddr=0010ebd4 vaddr=3fc98600 size=01444h (  5188) load
I (228) esp_image: segment 2: paddr=00110020 vaddr=42000020 size=12dca8h (1236136) map
I (450) esp_image: segment 3: paddr=0023dcd0 vaddr=3fc99a44 size=039ech ( 14828) load
I (454) esp_image: segment 4: paddr=002416c4 vaddr=40374000 size=14558h ( 83288) load
I (483) boot: Loaded app from partition at offset 0x10000
I (483) boot: Disabling RNG early entropy source...
I (484) cpu_start: Multicore app
I (494) cpu_start: Pro cpu start user code
I (494) cpu_start: cpu freq: 240000000 Hz
I (494) app_init: Application information:
I (494) app_init: Project name:     yfbao
I (494) app_init: App version:      0.0.1
I (494) app_init: Compile time:     Nov 22 2024 19:34:27
I (495) app_init: ELF file SHA256:  d2b7afd1f...
I (495) app_init: ESP-IDF:          v5.3-dirty
I (495) efuse_init: Min chip rev:     v0.0
I (495) efuse_init: Max chip rev:     v0.99 
I (495) efuse_init: Chip rev:         v0.2
I (495) heap_init: Initializing. RAM available for dynamic allocation:
I (496) heap_init: At 3FCA77A8 len 00041F68 (263 KiB): RAM
I (496) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (496) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (496) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (498) spi_flash: detected chip: generic
I (498) spi_flash: flash io: dio
I (501) sleep: Configure to isolate all GPIO pins in sleep state
I (501) sleep: Enable automatic switching of GPIO sleep configuration
I (00:08:04.274) main_task: Started on CPU0
[... 3145 lines cut ...]
I (00:08:24.344) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.9.1
I (1255) wifi:ap channel adjust o:1,1 n:6,1
I (1255) wifi:new:<6,1>, old:<1,1>, ap:<6,1>, sta:<6,1>, prof:1, snd_ch_cfg:0x0
I (1255) wifi:state: init -> auth (0xb0)
I (1265) wifi:state: auth -> assoc (0x0)
I (1345) wifi:state: assoc -> run (0x10)
I (1495) wifi:connected with idc_2.4G, aid = 18, channel 6, 40U, bssid = 24:cf:24:01:c1:dc
I (1495) wifi:security: WPA2-PSK, phy: bgn, rssi: -24
I (1495) wifi:pm start, type: 1

I (1495) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1495) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (1535) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (00:08:25.593) wifiManager: got ip:192.168.8.139
I (00:08:25.595) esp_netif_handlers: sta ip: 192.168.8.139, mask: 255.255.255.0, gw: 192.168.8.1
I (00:08:25.595) wifiManager: apsta mode,set NAPT
W (00:08:25.596) esp_netif_lwip: napt disabled on esp_netif:0x3fcd5524
I (00:08:25.598) uartManager: Uart start...
I (00:08:25.601) Select: Adding uart fd 6
I (00:08:25.602) http_server: HTTP start...
I (00:08:25.604) http_server: Registering URI handlers
I (00:08:25.606) esp_https_server: Starting server
I (00:08:25.611) esp_https_server: Server listening on port 443
I (00:08:25.612) http_server: Registering URI handlers
[0nI (22935) wifi:<ba-add>idx:0 (ifx:0, 24:cf:24:01:c1:dc), tid:0, ssn:1088, winSize:64

AdityaHPatwardhan, 2025-01-20:
Hi @lujiex Sorry for the delayed response.
Did you check the memory availability in this case?
Is it possible for you to print the memory stats ( Free heap size, Minimum free heap size) before attempting the new connection?

AdityaHPatwardhan, 2025-02-17:
Hi @lujiex, 
Apologies for my delay in communication earlier, but this issue seems to be related to memory and making sure that sufficient memory is available should fix it.
I am closing this issue for now.
Please feel to re-open if required.

closed 2025-02-17 by AdityaHPatwardhan