espressif/esp-idf #14884: Add LEDC ledc_calculate_divisor() function to the /driver/ledc.h.

IhorNehrutsa, 2024-11-14:
### Is your feature request related to a problem?

The `ledc_timer_set()` function requests `uint32_t clock_divider` parameter.
https://github.com/espressif/esp-idf/blob/7a305c0284b7af7cd8b8f12b48f72e2685d9a363/components/esp_driver_ledc/include/driver/ledc.h#L365
`clock_divider` is not a trivial.

It can be calculated with the `ledc_calculate_divisor()` function.
https://github.com/espressif/esp-idf/blob/7a305c0284b7af7cd8b8f12b48f72e2685d9a363/components/esp_driver_ledc/src/ledc.c#L451

This function is not accessible in the LEDC API driver. 

### Describe the solution you'd like.

Please add ledc_calculate_divisor() function to the `esp-idf/components/esp_driver_ledc/include/driver/ledc.h`
```
uint32_t ledc_calculate_divisor(uint32_t src_clk_freq, int freq_hz, uint32_t precision);
```

suda-morris, 2024-11-14:
AFAIK, some of the ledc driver API is redundant, why not use `ledc_timer_config` to setup the ledc timer? You only need to provide the source clock and the target timer resolution, the driver can help calculate the divisor.

IhorNehrutsa, 2024-11-14:
> some of the ledc driver API is redundant

Ok. Then please mark `ledc_timer_set()` as obsolete.

closed 2024-12-20 by espressif-bot