espressif/esp-idf #14887: Heap: Could fixed width integer types be used for multi_heap_info_t?

stkw0, 2024-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I was using `heap_caps_get_info` to monitor the info of the heap memory and send it to a remote computer. While doing that I found that, because `multi_heap_info_t` struct uses `size_t` for all the fields and both machines have a different width for size_t, it was not reading properly.

I was thinking on making a PR to change from `size_t` to `uint32_t` but I was not sure about potential side effects or if the same library was used for machines with longer width. Would it be worth it to change that struct to use fixed width integer types?

atanisoft, 2024-11-14:
on the esp32 size_t is 32-bit in size. Changing to uint32_t would potentially be a very large PR and has a risk of breaking various consumers of the types.

igrr, 2024-11-14:
At least the code which prints `multi_heap_info_t` members could get broken by such a change, since `uint32_t` and `size_t` require different format specifiers.

I am not sure how exactly you send the data between machines, but I would recommend using something like protobufs (if you need a compact encoding) or a text format like json (if not). This will take care of any differences in type sizes, endianness, etc., between platforms.

stkw0, 2024-11-14:
I was sending the raw struct, like `write(fd, &heap_info, sizeof(multi_heap_info_t))`, basically to save me from processing the struct and having to adapt the code in case a member on multi_heap_info_t is added/removed.