espressif/esp-idf #14888: TWAI ISR Interrupt on RX

s54b32julian, 2024-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In Automotive CAN bus it's mandatory often to respond/react to a received message within a specific or strict time frame.
So the User should be able to interact with the interrupt if a TWAI message is received. I see in most external can controllers there is a HW interrupt available. I see this request/issue also from other users that want to use TWAI and change because of this to a external can controller with HW interrupt. But I'm tied to the internal TWAI of esp and have to use it.

I have read the documentation and try to find online a solution, but I couldn't solve this issue.
So how can the user/I interact with the ISR? Best would be a user friendly setup but I would also happy if there is a workaround how I can access/set up this?

igrr, 2024-11-14:
Linking the previous discussion: https://github.com/espressif/esp-idf/issues/9152#issuecomment-1757026192

s54b32julian, 2024-11-15:
> Linking the previous discussion: [#9152 (comment)](https://github.com/espressif/esp-idf/issues/9152#issuecomment-1757026192)

Was not my comment/discussion, but thanks for linking the need is the same from this person in the linked comment.

s54b32julian, 2024-11-19:
Maybe @Dazza0 ?
Or has someone from the community a workaround to make this happen?

wanckl, 2024-11-20:
@s54b32julian  What you want to do in ISR?

I have a patch for twai driver `soft-filter`, which can register an RX callback, dose it meet your requiest?

[0001-feat-twai-support-software-filter-for-legacy-twai-dr.patch](https://github.com/user-attachments/files/17827088/0001-feat-twai-support-software-filter-for-legacy-twai-dr.patch)

s54b32julian, 2024-11-20:
@wanckl : thanks for this, I will study this deeply and see if it could help me.

My task is, to react on specific messages within a tight time frame. So the goal is, while the esp32 is busy with other load heavy tasks, always when a twai message is received and put into the internal RX buffer i need an interrupt where i can set up a ISR. In this ISR I would just quickly check if it's a message i have to respond/react immediately or can schedule the task for this message (response/calcutation/IOs etc) for later. 

This is in automotive CAN network absolute mandatory. Maybe a esp32 without any other task on it is with some luck able to respond in time, but with other tasks, especially with load heavy things running, nearly impossible. I currently see for this way no other possibility. And change the complete hardware to something like a MCP2515 is no option at the moment (and maybe wasted potential of the esp32)

wanckl, 2024-11-20:
@s54b32julian 

The patch contain an test `twai_software_filter`, you can search for it.

the callback provide only the msg ID of current RX msg, to let you decide if receive it or not.

no data or other infomation provided, not sure if it is OK for you,

Dazza0, 2024-12-02:
@s54b32julian my 2c:

> always when a twai message is received and put into the internal RX buffer i need an interrupt where i can set up a ISR. In this ISR I would just quickly check if it's a message i have to respond/react immediately or can schedule the task for this message (response/calcutation/IOs etc) for later.

Keep in mind that the TWAI ISR can have latency. Other same/higher priority interrupts and critical sections can cause the TWAI ISR to be late. Raising the TWAI ISR's priority can mitigate (but not prevent) latency. High priority interrupts can preempt critical sections, but have the following drawbacks:

- Must be written in assembly
- Sharing variables/primitives with other parts of C code is complicated due high priority ISR's ability to preempt critical section

Basically, you'll need to sum up the worst case run time of all the things that can prevent the TWAI ISR from running, and assess if that total latency is smaller than the max latency for the TWAI response.

Furthermore, if the TWAI peripheral already has a TX message pending, the peripheral will automatically attempt transmission at the next opportunity in accordance with the protocol (i.e., first SOF bit after inter-frame space). If the TWAI ISR is late to the point where it runs after the SOF bit, you'll need account for the transmission time of the current message on the bus into your worst case latency.

> And change the complete hardware to something like a MCP2515 with its external interrupt pin is no option at the moment (and maybe wasted potential of the esp32)

You'll face the exact same (if not more complex) issues with the MCP2515. The INT pin will be tied to some ISR (e.g., GPIO ISR) on the ESP32. That ISR will face the exact same ISR latency issues mentioned above. Moreover, you'll have added latency because you'll need to send some SPI transaction to the MCP2515 before it will start transmission. Writing to the TWAI register is always faster than an SPI transaction.