espressif/esp-idf #14889: Esp32 reset continuous

flux0-0, 2024-11-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I loaded code to esp32 by platform IO and have problem " rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46 " How can I fix it ? My source code can only work in some workspace in the platform io, even though the source code is the same.

ESP-Marius, 2024-11-15:
Do you have logging enabled? 

If the chip cannot even start booting the secondary bootloader then there might be something wrong with the flash configs, does the sdkconfig differ between workspaces?

Alvin1Zhang, 2025-01-14:
@flux0-0 Thanks for reporting, would you please help share more updates for the issue? Thanks.

ESP-Marius, 2025-01-26:
Closing this one due to a lack of information, feel free to reopen if there is any new updates.

closed 2025-01-26 by ESP-Marius