espressif/esp-idf #14890: Missing option for USB MSC drive format

Asanga-Viraj, 2024-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using idf 5.0.7 for software development to use in the module ESP32-S3. I have an application to mount file system on USB Flash drive. At some stage, I have to format the USB drive while is connected to ESP32-S3. But there is no function to do it in ESP. I checked with the idf source code and there are options for SD Card and FLASH, but not for USB MSC.

Is there any other way to format the USB stick or this option is going to add to idf in near future?

Thanks.

peter-marcisovsky, 2024-11-15:
Hi @Asanga-Viraj, thanks for the feature request.

The MSC Host driver formats the connected flash drive only during the driver initialization, by calling [msc_format_storage](https://github.com/leeebo/esp-usb/blob/master/host/class/msc/usb_host_msc/src/msc_host_vfs.c#L28C18-L28C36) function.

The function is static though. A workaround for your case right now, would be to force-trigger formatting by calling directly this function. 

Although, are planning to add a public API for formatting, as you are suggesting. The development is planned following months.
I will keep this issue open for tracking.

Asanga-Viraj, 2024-11-17:
Thanks @peter-marcisovsky .

closed 2024-12-16 by Alvin1Zhang