espressif/esp-idf #14891: 【求助】在测试蓝牙1对8时，第一次扫描8个成功，就没问题，但若第一次扫描连接了比如7个，然后报错，再扫描也连接不上最后一个了，报这种错误

beautifulpeer, 2024-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF版本 5.3.1
芯片ESP32-S3 PICO
BLE5.0

正常log：成功扫描到8个设备
2024-11-15 15:02:24 ESP-ROM:esp32s3-20210327
2024-11-15 15:02:24 Build:Mar 27 2021       
2024-11-15 15:02:24 rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
2024-11-15 15:02:24 SPIWP:0xee
2024-11-15 15:02:24 mode:DIO, clock div:1
2024-11-15 15:02:24 load:0x3fce2820,len:0x1918
2024-11-15 15:02:24 load:0x403c8700,len:0x4
2024-11-15 15:02:24 load:0x403c8704,len:0xe04
2024-11-15 15:02:24 load:0x403cb700,len:0x3014
2024-11-15 15:02:24 entry 0x403c8930
2024-11-15 15:02:24 I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
2024-11-15 15:02:24 I (27) boot: compile time Nov 14 2024 17:53:11        
2024-11-15 15:02:24 I (27) boot: Multicore bootloader
2024-11-15 15:02:24 I (30) boot: chip revision: v0.2
2024-11-15 15:02:24 I (34) qio_mode: Enabling default flash chip QIO
2024-11-15 15:02:24 I (39) boot.esp32s3: Boot SPI Speed : 80MHz
2024-11-15 15:02:24 I (44) boot.esp32s3: SPI Mode       : QIO
2024-11-15 15:02:24 I (49) boot.esp32s3: SPI Flash Size : 8MB
2024-11-15 15:02:24 I (54) boot: Enabling RNG early entropy source...
2024-11-15 15:02:24 I (59) boot: Partition Table:
2024-11-15 15:02:24 I (63) boot: ## Label            Usage          Type ST Offset   Length
2024-11-15 15:02:24 I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
2024-11-15 15:02:24 I (77) boot:  1 factory          factory app      00 00 00010000 00200000
2024-11-15 15:02:24 I (85) boot:  2 custom_nvs       WiFi data        01 02 00210000 00100000
2024-11-15 15:02:24 I (92) boot: End of partition table
2024-11-15 15:02:24 I (97) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1ec24h (125988) map
2024-11-15 15:02:24 I (124) esp_image: segment 1: paddr=0002ec4c vaddr=3fc98700 size=013cch (  5068) load
2024-11-15 15:02:24 I (125) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=6e334h (451380) map
2024-11-15 15:02:24 I (198) esp_image: segment 3: paddr=0009e35c vaddr=3fc99acc size=02d00h ( 11520) load
2024-11-15 15:02:24 I (201) esp_image: segment 4: paddr=000a1064 vaddr=40374000 size=14678h ( 83576) load
2024-11-15 15:02:24 I (228) boot: Loaded app from partition at offset 0x10000
2024-11-15 15:02:24 I (228) boot: Disabling RNG early entropy source...
2024-11-15 15:02:24 I (240) cpu_start: Multicore app
2024-11-15 15:02:24 I (249) cpu_start: Pro cpu start user code
2024-11-15 15:02:24 I (249) cpu_start: cpu freq: 240000000 Hz
2024-11-15 15:02:24 I (249) app_init: Application information:
2024-11-15 15:02:24 I (252) app_init: Project name:     tests
2024-11-15 15:02:24 I (257) app_init: App version:      35be65f-dirty
2024-11-15 15:02:24 I (262) app_init: Compile time:     Nov 14 2024 19:15:00
2024-11-15 15:02:24 I (268) app_init: ELF file SHA256:  2d124f9cf329f5be...
2024-11-15 15:02:24 I (274) app_init: ESP-IDF:          v5.3.1-dirty
2024-11-15 15:02:24 I (279) efuse_init: Min chip rev:     v0.0
2024-11-15 15:02:24 I (284) efuse_init: Max chip rev:     v0.99 
2024-11-15 15:02:24 I (289) efuse_init: Chip rev:         v0.2
2024-11-15 15:02:24 I (294) heap_init: Initializing. RAM available for dynamic allocation:
2024-11-15 15:02:24 I (301) heap_init: At 3FCA0590 len 00049180 (292 KiB): RAM
2024-11-15 15:02:24 I (307) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
2024-11-15 15:02:24 I (313) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
2024-11-15 15:02:24 I (320) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
2024-11-15 15:02:24 I (327) spi_flash: detected chip: generic
2024-11-15 15:02:24 I (330) spi_flash: flash io: qio
2024-11-15 15:02:25 I (335) sleep: Configure to isolate all GPIO pins in sleep state
2024-11-15 15:02:25 I (341) sleep: Enable automatic switching of GPIO sleep configuration
2024-11-15 15:02:25 I (348) coexist: coex firmware version: 4482466
2024-11-15 15:02:25 I (376) coexist: coexist rom version e7ae62f
2024-11-15 15:02:25 I (377) main_task: Started on CPU0
2024-11-15 15:02:25 I (387) main_task: Calling app_main()
[... 111 lines cut ...]
3.BT_HCI: btu_hcif_hdl_command_status,opcode:0x2043,status:0x3b这个警告是由于连接参数不对导致，但我正常和异常时使用的都是同一组参数，参入如下：

static esp_ble_ext_scan_params_t m_scan_params = {
    .own_addr_type  = BLE_ADDR_TYPE_PUBLIC,
    .filter_policy  = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_duplicate = BLE_SCAN_DUPLICATE_ENABLE,
    .cfg_mask       = ESP_BLE_GAP_EXT_SCAN_CFG_UNCODE_MASK,
    .uncoded_cfg    = {BLE_SCAN_TYPE_ACTIVE, 160, 160},  // 400ms 20ms : 640, 32     100ms 50ms : 160, 80  (625 us units)
    .coded_cfg      = {BLE_SCAN_TYPE_ACTIVE, 160, 160},
};

const esp_ble_gap_conn_params_t phy_1m_conn_params = {
    .scan_interval       = 0x50,  // 扫描间隔，单位：0.625ms
    .scan_window         = 0x50,  // 扫描窗口，单位：0.625ms
    .interval_min        = 0x10,   // 最小连接间隔，单位：1.25ms
    .interval_max        = 0x30,   // 最大连接间隔，单位：1.25ms
    .latency             = 0,      // 连接延迟，单位：连接事件
    .supervision_timeout = 100,    // 连接监督超时，单位：10ms
    .min_ce_len          = 0,      // 最小连接事件长度，单位：0.625ms
    .max_ce_len          = 0       // 最大连接事件长度，单位：0.625ms
};

4.这种情况很不稳定，有时行，有时不行，如何才能避免此种情况？

@esp-zhp

beautifulpeer, 2024-11-15:
有时还会报W (1887) BT_L2CAP: L2CAP - LE - cannot start new connection at conn st: 1 这个警告，但是也连接成功8个设备了

esp-zhp, 2024-11-15:
@beautifulpeer 
你那面方便获取一下 HCI log 吗？

esp-zhp, 2024-11-15:
[image]

启用配置项后，可以在 `app_main` 中调用以下函数来输出 HCI 日志：

```c
extern void bt_hci_log_hci_data_show(void);
extern void bt_hci_log_hci_adv_show(void);

while (1)
{
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    bt_hci_log_hci_data_show();
    bt_hci_log_hci_adv_show();
}
``` 

该循环将每秒周期性地输出 HCI 数据和广播信息。

请给我完整的  HCI log 信息。谢谢

beautifulpeer, 2024-11-15:
您好，HCI log已开启，辛苦您帮我看看，此次连接8个异常

2024-11-15 16:28:29 ESP-ROM:esp32s3-20210327
2024-11-15 16:28:29 Build:Mar 27 2021
2024-11-15 16:28:29 rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
2024-11-15 16:28:29 SPIWP:0xee
2024-11-15 16:28:29 mode:DIO, clock div:1
2024-11-15 16:28:29 load:0x3fce2820,len:0x1918
2024-11-15 16:28:29 load:0x403c8700,len:0x4
2024-11-15 16:28:29 load:0x403c8704,len:0xe04
2024-11-15 16:28:29 load:0x403cb700,len:0x3014
2024-11-15 16:28:29 entry 0x403c8930
2024-11-15 16:28:29 I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
2024-11-15 16:28:29 I (27) boot: compile time Nov 14 2024 17:53:11        
2024-11-15 16:28:29 I (27) boot: Multicore bootloader
2024-11-15 16:28:29 I (30) boot: chip revision: v0.2
2024-11-15 16:28:29 I (34) qio_mode: Enabling default flash chip QIO
2024-11-15 16:28:29 I (39) boot.esp32s3: Boot SPI Speed : 80MHz
2024-11-15 16:28:29 I (44) boot.esp32s3: SPI Mode       : QIO  
2024-11-15 16:28:29 I (49) boot.esp32s3: SPI Flash Size : 8MB  
2024-11-15 16:28:29 I (54) boot: Enabling RNG early entropy source...
2024-11-15 16:28:29 I (59) boot: Partition Table:
2024-11-15 16:28:29 I (63) boot: ## Label            Usage          Type ST Offset   Length
2024-11-15 16:28:29 I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
2024-11-15 16:28:29 I (77) boot:  1 factory          factory app      00 00 00010000 00200000
2024-11-15 16:28:29 I (85) boot:  2 custom_nvs       WiFi data        01 02 00210000 00100000
2024-11-15 16:28:29 I (92) boot: End of partition table
2024-11-15 16:28:29 I (97) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1ec54h (126036) map
2024-11-15 16:28:29 I (124) esp_image: segment 1: paddr=0002ec7c vaddr=3fc98900 size=0139ch (  5020) load
2024-11-15 16:28:29 I (125) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=6e87ch (452732) map 
2024-11-15 16:28:30 I (199) esp_image: segment 3: paddr=0009e8a4 vaddr=3fc99c9c size=02d30h ( 11568) load
2024-11-15 16:28:30 I (201) esp_image: segment 4: paddr=000a15dc vaddr=40374000 size=14840h ( 84032) load
2024-11-15 16:28:30 I (229) boot: Loaded app from partition at offset 0x10000
2024-11-15 16:28:30 I (229) boot: Disabling RNG early entropy source...
2024-11-15 16:28:30 I (241) cpu_start: Multicore app
2024-11-15 16:28:30 I (250) cpu_start: Pro cpu start user code
2024-11-15 16:28:30 I (250) cpu_start: cpu freq: 240000000 Hz
2024-11-15 16:28:30 I (250) app_init: Application information:
2024-11-15 16:28:30 I (253) app_init: Project name:     tests
2024-11-15 16:28:30 I (257) app_init: App version:      35be65f-dirty
2024-11-15 16:28:30 I (263) app_init: Compile time:     Nov 15 2024 16:25:50
2024-11-15 16:28:30 I (269) app_init: ELF file SHA256:  fa5ed780e34f224c...
2024-11-15 16:28:30 I (275) app_init: ESP-IDF:          v5.3.1-dirty
2024-11-15 16:28:30 I (280) efuse_init: Min chip rev:     v0.0
2024-11-15 16:28:30 I (285) efuse_init: Max chip rev:     v0.99 
2024-11-15 16:28:30 I (290) efuse_init: Chip rev:         v0.2
2024-11-15 16:28:30 I (294) heap_init: Initializing. RAM available for dynamic allocation:
2024-11-15 16:28:30 I (302) heap_init: At 3FCA07E0 len 00048F30 (291 KiB): RAM
2024-11-15 16:28:30 I (308) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
2024-11-15 16:28:30 I (314) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
2024-11-15 16:28:30 I (320) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
2024-11-15 16:28:30 I (327) spi_flash: detected chip: generic
2024-11-15 16:28:30 I (331) spi_flash: flash io: qio
2024-11-15 16:28:30 I (335) sleep: Configure to isolate all GPIO pins in sleep state
2024-11-15 16:28:30 I (342) sleep: Enable automatic switching of GPIO sleep configuration
2024-11-15 16:28:30 I (349) coexist: coex firmware version: 4482466
2024-11-15 16:28:30 I (376) coexist: coexist rom version e7ae62f
2024-11-15 16:28:30 I (377) main_task: Started on CPU0
2024-11-15 16:28:30 I (387) main_task: Calling app_main()
2024-11-15 16:28:30 I (447) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (457) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (467) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (477) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (487) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (497) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (507) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (517) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (527) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-15 16:28:30 I (537) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
2024-11-15 16:28:30 E (537) test_ble_c:  - - - - - - - - - - - - - - - - - - - - - -
[... 415 lines cut ...]
2024-11-15 16:28:40 47 H:06 00 0b 00 07 00 04 00 08 2c 00 ff ff 03 28
2024-11-15 16:28:40 48 E:13 05 01 06 00 01 00
2024-11-15 16:28:40 49 D:06 20 09 00 05 00 04 00 01 08 2c 00 0a
2024-11-15 16:28:40 4a C:35 0c 05 01 06 00 01 00
[... cut ...]
2024-11-15 16:28:40 5a C:35 0c 05 01 06 00 01 00
2024-11-15 16:28:40 5b H:06 00 09 00 05 00 04 00 52 2d 00 01 00
2024-11-15 16:28:40 5c E:13 05 01 06 00 01 00
2024-11-15 16:28:40 14 39 0d 01 10 00 01 a0 76 02 a4 07 31 01 00 ff 7f a7 00 00 00 00 00 00 00 00 00 1f 1e ff 06 00 01 09 20 22 73 de 35 1e 83 a7 1d bf 64 05 42 8a 77 31 d2 d6 37 b7 48 78 0c 
71 36
2024-11-15 16:28:40 15 32 0d 01 01 00 00 a2 f5 71 33 e8 64 01 02 00 7f d8 00 00 00 00 00 00 00 00 00 18 05 03 f0 ab 06 09 02 01 06 0e 09 45 72 67 6f 4c 41 42 5f 78 78 78 78 36

esp-cjh, 2024-11-15:
@beautifulpeer 好的，我来看一下

beautifulpeer, 2024-11-16:
补充一点，
            conn_params.latency = 0;
            conn_params.min_int = 12;   // min_int = 12*1.25ms = 15ms //测试发现最小值不能低于13.75ms
            conn_params.max_int = 12;   // max_int = 12*1.25ms = 15ms
            conn_params.timeout = 100;  // timeout = 100*10ms = 1000ms
此参数会下发到每个连接的设备中，连接的设备打印的log为：
2024-11-16 11:47:48 I (353016) ble_p: update connetion params status = 0, min_int = 0, max_int = 0,conn_int = 12,latency = 0, timeout = 100
2024-11-16 11:47:49 I (353556) ble_p: ESP_GATTS_MTU_EVT, MTU 247
其中，min_int = 0, max_int = 0，这两个参数为0，我不知道为什么

beautifulpeer, 2024-11-16:
还有个问题，在我重启接收端是，开始扫描，有时会重启，不是经常复现，不知道为什么

log如下：

2024-11-14 16:56:20 ESP-ROM:esp32s3-20210327
2024-11-14 16:56:20 Build:Mar 27 2021       
2024-11-14 16:56:20 rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
2024-11-14 16:56:20 SPIWP:0xee
2024-11-14 16:56:20 mode:DIO, clock div:1
2024-11-14 16:56:20 load:0x3fce2820,len:0x1918
2024-11-14 16:56:20 load:0x403c8700,len:0x4
2024-11-14 16:56:20 load:0x403c8704,len:0xe04
2024-11-14 16:56:20 load:0x403cb700,len:0x3014
2024-11-14 16:56:20 entry 0x403c8930
2024-11-14 16:56:20 I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
2024-11-14 16:56:20 I (27) boot: compile time Nov 14 2024 16:35:29        
2024-11-14 16:56:20 I (27) boot: Multicore bootloader
2024-11-14 16:56:20 I (30) boot: chip revision: v0.2
2024-11-14 16:56:20 I (34) qio_mode: Enabling default flash chip QIO
2024-11-14 16:56:20 I (39) boot.esp32s3: Boot SPI Speed : 80MHz     
2024-11-14 16:56:20 I (44) boot.esp32s3: SPI Mode       : QIO        
2024-11-14 16:56:20 I (49) boot.esp32s3: SPI Flash Size : 8MB        
2024-11-14 16:56:20 I (54) boot: Enabling RNG early entropy source...
2024-11-14 16:56:20 I (59) boot: Partition Table:
2024-11-14 16:56:20 I (63) boot: ## Label            Usage          Type ST Offset   Length  
2024-11-14 16:56:20 I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
2024-11-14 16:56:20 I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
2024-11-14 16:56:20 I (85) boot:  2 factory          factory app      00 00 00010000 00100000
2024-11-14 16:56:20 I (92) boot: End of partition table
2024-11-14 16:56:20 I (97) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=1ebd4h (125908) map
2024-11-14 16:56:20 I (124) esp_image: segment 1: paddr=0002ebfc vaddr=3fc98700 size=0141ch (  5148) load
2024-11-14 16:56:20 I (125) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=6e300h (451328) map 
2024-11-14 16:56:20 I (198) esp_image: segment 3: paddr=0009e328 vaddr=3fc99b1c size=02cb0h ( 11440) load
2024-11-14 16:56:20 I (201) esp_image: segment 4: paddr=000a0fe0 vaddr=40374000 size=14678h ( 83576) load
2024-11-14 16:56:20 I (228) boot: Loaded app from partition at offset 0x10000
2024-11-14 16:56:20 I (228) boot: Disabling RNG early entropy source...      
2024-11-14 16:56:20 I (240) cpu_start: Multicore app
2024-11-14 16:56:20 I (249) cpu_start: Pro cpu start user code
2024-11-14 16:56:20 I (249) cpu_start: cpu freq: 240000000 Hz
2024-11-14 16:56:20 I (249) app_init: Application information:
2024-11-14 16:56:20 I (252) app_init: Project name:     tests
2024-11-14 16:56:20 I (257) app_init: App version:      ccf2136-dirty
2024-11-14 16:56:20 I (262) app_init: Compile time:     Nov 14 2024 16:35:09
2024-11-14 16:56:20 I (268) app_init: ELF file SHA256:  75b0ad98c9416f9b...
2024-11-14 16:56:20 I (274) app_init: ESP-IDF:          v5.3.1-dirty
2024-11-14 16:56:20 I (279) efuse_init: Min chip rev:     v0.0
2024-11-14 16:56:20 I (284) efuse_init: Max chip rev:     v0.99 
2024-11-14 16:56:20 I (289) efuse_init: Chip rev:         v0.2
2024-11-14 16:56:20 I (294) heap_init: Initializing. RAM available for dynamic allocation:
2024-11-14 16:56:20 I (301) heap_init: At 3FCA0590 len 00049180 (292 KiB): RAM
2024-11-14 16:56:20 I (307) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
2024-11-14 16:56:20 I (313) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
2024-11-14 16:56:20 I (320) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
2024-11-14 16:56:20 I (327) spi_flash: detected chip: generic
2024-11-14 16:56:20 I (330) spi_flash: flash io: qio
2024-11-14 16:56:20 I (335) sleep: Configure to isolate all GPIO pins in sleep state
2024-11-14 16:56:20 I (341) sleep: Enable automatic switching of GPIO sleep configuration
2024-11-14 16:56:20 I (348) coexist: coex firmware version: 4482466
2024-11-14 16:56:20 I (376) coexist: coexist rom version e7ae62f
2024-11-14 16:56:20 I (376) main_task: Started on CPU0
2024-11-14 16:56:20 I (386) main_task: Calling app_main()
2024-11-14 16:56:20 I (446) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (456) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (466) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (476) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (486) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (496) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (506) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (516) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
2024-11-14 16:56:21 I (526) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[... 45 lines cut ...]
2024-11-14 16:59:36 
2024-11-14 16:59:36 Rebooting...
2024-11-14 16:59:36 �ESP-ROM:esp32s3-20210327
2024-11-14 16:59:36 Build:Mar 27 2021
2024-11-14 16:59:36 rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
2024-11-14 16:59:36 Saved PC:0x40375a78
--- 2024-11-14 16:59:36 0x40375a78: esp_restart_noos at E:/kf/15____vscode_insiders/v5.3.1/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

[... cut ...]
2024-11-14 16:59:36 I (31) boot: compile time Nov 14 2024 16:35:29
2024-11-14 16:59:36 I (31) boot: Multicore bootloader
2024-11-14 16:59:36 I (35) boot: chip revision: v0.2
2024-11-14 16:59:36 I (39) qio_mode: Enabling default flash chip QIO
2024-11-14 16:59:36 I (44) boot.esp32s3: Boot SPI Speed : 80MHz
2024-11-14 16:59:36 I (49) boot.esp32s3: SPI Mode       : QIO
2024-11-14 16:59:36 I (53) boot.esp32s3: SPI Flash Size : 8MB
2024-11-14 16:59:36 I (58) boot: Enabling RNG early entropy source...
2024-11-14 16:59:36 I (64) boot: Partition Table:

beautifulpeer, 2024-11-16:
还有就是，在连接成功8个设备后，主动给一个设备复位，接收端这边会主动根据上次连接的地址来重连，但接收端报错还是一直报警告且连接不上，一直重连，一直是报这个警告

报警告log如下：

2024-11-16 15:12:13 I (110067) egl_ble_c: reconnect start
2024-11-16 15:12:13 I (110067) egl_ble_c: 0x3fc9cf54   3c 84 27 d8 7d d2                                 |<.'.}.|
2024-11-16 15:12:13 W (110077) BT_HCI: btu_hcif_hdl_command_status,opcode:0x2043,status:0x3b
2024-11-16 15:12:13 W (110087) BT_APPL: gattc_conn_cb: if=1 st=0 id=1537 rsn=0x100
2024-11-16 15:12:13 W (110097) BT_APPL: gattc_conn_cb: if=2 st=0 id=1538 rsn=0x100
2024-11-16 15:12:13 W (110097) BT_APPL: gattc_conn_cb: if=3 st=0 id=1539 rsn=0x100
2024-11-16 15:12:13 W (110107) BT_APPL: gattc_conn_cb: if=4 st=0 id=1540 rsn=0x100
2024-11-16 15:12:13 W (110117) BT_APPL: gattc_conn_cb: if=5 st=0 id=1541 rsn=0x100
2024-11-16 15:12:13 W (110117) BT_APPL: gattc_conn_cb: if=6 st=0 id=1542 rsn=0x100
2024-11-16 15:12:13 W (110127) BT_APPL: gattc_conn_cb: if=7 st=0 id=1543 rsn=0x100
2024-11-16 15:12:13 W (110127) BT_APPL: gattc_conn_cb: if=8 st=0 id=1544 rsn=0x100

beautifulpeer, 2024-11-16:
这是我设备端广播参数

esp_ble_gap_ext_adv_params_t ext_adv_params_2M = {
    .type           = ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE, 
    .interval_min   = 0x20,                                     
    .interval_max   = 0x20,                                     
    .channel_map    = ADV_CHNL_ALL,
    .filter_policy  = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy    = ESP_BLE_GAP_PHY_1M,
    .max_skip       = 0,
    .secondary_phy  = ESP_BLE_GAP_PHY_2M,
    .sid            = 0,
    .scan_req_notif = false,
    .own_addr_type  = BLE_ADDR_TYPE_PUBLIC,
    .tx_power       = EXT_ADV_TX_PWR_NO_PREFERENCE,
};

esp-cjh, 2024-11-18:
Hi, @beautifulpeer 你能提供复现代码吗？例如 example 之类的（可以只保留蓝牙的逻辑，其他的去掉），这样能让我们更快地复现问题并 debug。否则我们需要本地编写测试用例来复现，而且不确定能不能复现。

beautifulpeer, 2024-11-18:
您提供个邮箱，我把测试代码发您可以吗

esp-cjh, 2024-11-18:
好的，这是我的邮箱：chenjianhua@espressif.com

beautifulpeer, 2024-11-18:
收到，我整理下发您，辛苦！

beautifulpeer, 2024-11-18:
您好，测试代码已发您邮箱@esp-cjh

esp-cjh, 2024-11-19:
好的，已收到

esp-cjh, 2024-11-19:
@beautifulpeer 
**问题原因：**
多连接场景下，controller 会给每个连接分配一个最小持续时间（3.75ms），如果想要 9 个连接共存的话，那么最小连接间隔（interval）必须大于等于 33.75ms。
你的代码在创建连接时使用的 interval 是 60ms，这个值是合理的，但是后续通过更新连接参数把 interval 更新为 15ms，导致带宽被占满，后续连接无法插入，controller 上报 0x3B 错误。

**解决方案：**
针对多连接场景，如果你需要使用一个最小的 interval，那我们建议将 `esp_ble_gap_prefer_ext_connect_params_set` 的 interval 设置为 27（33.75ms），这样能保证所有连接使用最小的 interval 并且能够共存，不建议在连接后进行连接参数更新。

beautifulpeer, 2024-11-19:
收到！感谢您的回复，我明天试下！

beautifulpeer, 2024-11-20:
您好，目前测试设置为33.75确实比较稳定，但是我测试对接收端的带宽会有影响，
1. 64Hz发包频率，每包发244个字节，发10秒
2. 1对8时，最大0.9Mbit/s，之前15ms的连接参数设置的比较小时，连接8个可以达到将近1Mbit/s
3. 这个还有优化的空间吗
4. 我尝试将连接参数逐渐减小，其丢包率也逐渐降低，您说的是9个设备，我其实是连接了8个设备，其中接收端也要算在内吗，还是说我设置成25（30ms）也可以
5.您说的controller会给每个连接分配一个固定的占用时间（3.75ms），这个参数是蓝牙固定的参数吗，还是可以设置

[image]
[image]

beautifulpeer, 2024-11-20:
1. 实际测试设置为33.75也会有概率出现0x3b的警告导致连接不上
2. 尝试增加这个连接参数，发现蓝牙带宽和这个参数有关，连接参数越大，接收端带宽越小（1对8时，8个同时发送数据）
@esp-cjh

beautifulpeer, 2024-11-22:
@esp-cjh hello，这方面还可以再优化吗

esp-cjh, 2024-11-22:
@beautifulpeer 可以的，我正在 debug 这个问题，并新增 API 来支持设置每个连接的带宽。

beautifulpeer, 2024-11-22:
ok，感谢，如果有新情况请第一时间告知我！

beautifulpeer, 2024-11-22:
1. 测试还发现个问题，一对一测试最大带宽时，发现phy_1m_conn_params 这个参数中的interval_min和interval_max
2. 这个两个参数设置成38，最高带宽，1.31Mbit/s
3. 而设置成64，最高带宽可以达到1.36Mbit/s，这个参数是否有推荐，使得带宽最大
@esp-cjh 

const esp_ble_gap_conn_params_t phy_1m_conn_params = {
    .scan_interval       = 0x40,                 // 扫描间隔，单位：0.625ms
    .scan_window         = 0x40,                 // 扫描窗口，单位：0.625ms
    .interval_min        = CONNECT_INTERVAL_MS,  // 最小连接间隔，单位：1.25ms   (0x26)38*1.25 = 47.5ms  最高 1.31Mbit/s  (0x40)64*1.25 = 80ms  最高 1.36Mbit/s
    .interval_max        = CONNECT_INTERVAL_MS,  // 最大连接间隔，单位：1.25ms   (0x26)38*1.25 = 47.5ms  最高 1.31Mbit/s  (0x40)64*1.25 = 80ms  最高 1.36Mbit/s
    .latency             = 0,                    // 连接延迟，单位：连接事件
    .supervision_timeout = 100,                  // 连接监督超时，单位：10ms
    .min_ce_len          = 0,                    // 最小连接事件长度，单位：0.625ms
    .max_ce_len          = 0                     // 最大连接事件长度，单位：0.625ms
};

beautifulpeer, 2024-11-30:
@esp-cjh  hello, 上述问题是否有进展

esp-cjh, 2024-12-02:
@beautifulpeer 
1 你是用 1 个 central 连接到 8 个 peripheral 的话，一共是 8 个连接。
2 S3 controller 支持的最小 connection event length 是 3.75ms（6 slot）。
3 你说 interval 为 80ms 时吞吐率达到最大，那可以将每个连接的 event length 设置为 10ms（即 16 slot）。

你可以通过以下来设置 connection event length
- IDF 应用 patch：[0001-support-set-ce-length-using-hci-command.patch](https://github.com/user-attachments/files/17973956/0001-support-set-ce-length-using-hci-command.patch)
- 打开配置项：CONFIG_BT_CTRL_CE_LENGTH_TYPE_CE=y
- 设置 perfer connection param：min_ce_len = 16, max_ce_len = 16

beautifulpeer, 2024-12-03:
好的，我尝试下，感谢您的回复

esp-zhp, 2024-12-10:
@beautifulpeer 有什么进展吗？

beautifulpeer, 2024-12-10:
您好，测试目前连接8个还算稳定，偶尔会连接不上第8个，但第二次扫描后还可以连接成功，目前以满足需求！感谢！

esp-zhp, 2024-12-10:
好的，如果能满足需求，该问题可以先关闭。

beautifulpeer, 2024-12-10:
okok

closed 2024-12-10 by beautifulpeer