espressif/esp-idf #14892: Addressing registers when using I2C and programming the LP Core. (esp32-C6)

gentooza, 2024-11-15:
### Is your feature request related to a problem?

Hi!
I'm just started using ULP programming with my esp32-C6.
I have it working, but functions described by the API seems to be able only to read from slave, and wirte to it using opcodes.
This is ok for connecting devices as the BH1750. [Datasheet here](https://www.mouser.com/datasheet/2/348/bh1750fvi-e-186247.pdf)
But using others as an INA260, I can't! [Datasheet here](https://www.ti.com/lit/ds/symlink/ina260.pdf?ts=1731575043428&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FINA260)

### Describe the solution you'd like.

I can be wrong, and it functionality is normal, but I wish I'd be able to read from a register X, and write to a register X.

cheers!

gentooza, 2024-11-18:
It seems it's possible using the Risc-V System API for the esp32-S3, isn't it?
https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32s3/api-reference/system/ulp-risc-v.html#_CPPv439ulp_riscv_i2c_master_set_slave_reg_addr7uint8_t

Why not using the esp32-c6?

ESP-Marius, 2024-11-19:
Hi, what you are describing should be possible.

Normally to read from an address X, you'd first do a i2c write to that address, followed by a i2c read to get the data.
If this doesnt work in your case please let me know.

> It seems to be possible using the Risc-V System API for the esp32-S3, isn't it?

The low power I2C had a special register for setting the reg. addr, but on C6 the peripheral operates more like a standard I2c peripheral, i.e. with just generic write and read transactions.

gentooza, 2024-11-19:
Hi @ESP-Marius 

Yes! I managed to read to X register by writing the **pointer register** before. (thanks to a colleague's hint :-D)

I've got problems with names I think, what in the i2c example provided calls opcode, the INA260 datasheet calls pointer register. So what I had to do is to use the [write_read function](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/ulp-lp-core.html#_CPPv436lp_core_i2c_master_write_read_device10i2c_port_t8uint16_tPK7uint8_t6size_tP7uint8_t6size_t7int32_t):
```
[esp_err_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/esp_err.html#_CPPv49esp_err_t) lp_core_i2c_master_write_read_device([i2c_port_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/i2c.html#_CPPv410i2c_port_t) lp_i2c_num, uint16_t device_addr, const uint8_t *data_wr, size_t write_size, uint8_t *data_rd, size_t read_size, int32_t ticks_to_wait)
```
Where I write the register address and read its content back.

I want to say too, I like more this system API (lp_core.h) than the one with the esp32-s3 (ulp_riscv.h) I found it easier to configure and play.

thanks!

closed 2024-11-19 by gentooza