espressif/esp-idf #14893: ADC Continuous read and LittleFS bug

GitHubLionel, 2024-11-15:
Hi
I use an ESP32 Dev Module and I use IDF version: v5.1.4-497-gdc859c1e67-dirty (esp v3.0.3 in Eclipse Sloeber).
My problem is simple : when I use ADC Continuous read and I try to read/write to my LittleFS partition, I crash my program with the error : `Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed).` 
The debug backtrace doesn't help.

I read lot of things about this problem and try to add : 
```
#define CONFIG_ADC_CONTINUOUS_ISR_IRAM_SAFE 1
#define CONFIG_GDMA_ISR_IRAM_SAFE 1
```
if `sdkconfig.h` and `soc_caps.h` files but we no success.
I don't know what to try anymore.
Any ideas to try ?
Thanks
Lionel

Kainarx, 2024-11-21:
Hi @GitHubLionel 
Please use menuconfig to enable these two configs and try again.  The linker won't put the function into IRAM if we define the macro by ourselves.

GitHubLionel, 2024-11-21:
Hi Kainarx
Many thanks for your answer.
Problem, I use Eclipse/Sloeber and I don't know how to use menuconfig in this IDE. 
I will ask Sloeber to know that.
Best regards,
Lionel

closed 2025-01-02 by Alvin1Zhang