espressif/esp-idf #14895: ESP32 S3 Wroom Startup is too slow

weoiss1998, 2024-11-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1 and 5.2.3

### Espressif SoC revision.

ESP32S3 Wroom

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 S3 Wroom Dev Kit

### Power Supply used.

USB

### What is the expected behavior?

https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/performance/reduce-boot-time.html says after optimizing the startup will be around 27ms.

### What is the actual behavior?

In my tests it's at the minimum of 56ms.

### Steps to reproduce.

Make optimizations of https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/performance/reduce-boot-time.html and check when a pin goes high.

### More Information.

I tried it with 5.2.3 and 5.3.1. There is practically no difference.
Here is a picture of my logic analyzer: 
[image]

zztiger123, 2024-11-18:
Hi @weoiss1998 

May I know if you have done every optimization? Especially Disable Boot ROM Log Printing. Need to burn refuses :
[image]

weoiss1998, 2024-11-18:
Hi @zztiger123 ,

yeah i did this aswell.

[image]

Xiehanxin, 2024-11-19:
hi @weoiss1998 here is my test result. I try the example/system/deep_sleep example, and set  gpio 6 level to 1 on _**call_start_cpu0()**_  ( as this function is the entry of appmain https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/startup.html#first-stage-rom-bootloader )and the top of app main, my setting GPIO code is like this:

[image]

[image]

I don't use the genneral GPIO API beacuse this it would be faster to directlty use lower apis.

[image]
from the result you can use it takes 27.5 ms to entry the start cpu0 ,which I think is the flag to finish boot, enter application, and 40 ms to print in appmain. 

however, it seems there are still some differences between your result and mine. I want to confirm the level change timing in your logic

weoiss1998, 2024-11-19:
I will try it in the next few days.
My measurement is from Power on to app_main. Maybe you can check this by your side aswell?

weoiss1998, 2024-11-19:
My spontaneous guess is that the ESP32 S3 boots from deep sleep faster than if the power is completely disconnected.

weoiss1998, 2024-11-19:
Hi @Xiehanxin, how can I modify the cpu_start.c or where do you add void IRAM_ATTR call_start_cpu0(void)?

Xiehanxin, 2024-11-20:
you can found in compinents/esp_system/port/cpu_start.c

weoiss1998, 2024-11-20:
I found it but the Pin Output doesn't change on my side.

Xiehanxin, 2024-11-20:
may I know which pin do you use?